riffnshred/nhl-led-scoreboard #142: I doubt it's an issue. Looking for advice.

jeepdrvr71, 2020-08-20:
Woke up this morning to my scoreboard totally black. Tried finding the scoreboard on my network but it wasn't discoverable. I checked the display board on an arduino that I have and it displays correctly. I checked the sd card and there is data on it. Only other things it could be are the adafruit hat or the pi zero. Anyone got any suggestions?

jeepdrvr71, 2020-08-27:
do the software install from scratch and getting the following message

-bash : get command not found 

after I type

git clone --recursive https://github.com/riffnshred/nhl-led-scoreboard

jeepdrvr71, 2020-08-28:
Got a new card and loaded everything. When I went to test I got this

pi@raspberrypi:~/nhl-led-scoreboard/src $ sudo python3 main.py --led-gpio-mapping=adafruit-hat-pwm --led-brightness=60 --led-slowdown-gpio=2
Traceback (most recent call last):
  File "main.py", line 4, in <module>
    from renderer.main import MainRenderer
  File "/home/pi/nhl-led-scoreboard/src/renderer/main.py", line 5, in <module>
    from boards.boards import Boards
  File "/home/pi/nhl-led-scoreboard/src/boards/boards.py", line 7, in <module>
    from boards.scoreticker import Scoreticker
  File "/home/pi/nhl-led-scoreboard/src/boards/scoreticker.py", line 7, in <module>
    from renderer.scoreboard import ScoreboardRenderer
  File "/home/pi/nhl-led-scoreboard/src/renderer/scoreboard.py", line 3, in <module>
    from renderer.logos import LogoRenderer
  File "/home/pi/nhl-led-scoreboard/src/renderer/logos.py", line 3, in <module>
    from images.image_helper import ImageHelper
  File "/home/pi/nhl-led-scoreboard/src/images/image_helper.py", line 2, in <module>
    import cairosvg
ModuleNotFoundError: No module named 'cairosvg'

jeepdrvr71, 2020-08-29:
reformatted my sd card and started from scratch and finally got everything working.

jeepdrvr71, 2020-08-29:
thanks for all your help

closed 2020-08-29 by jeepdrvr71

riffnshred, 2020-08-29:
Sweet. Glad you are back on track !