riffnshred/nhl-led-scoreboard #145: Possibly "port" to Adafruit MatrixPortal M4?

cheechbg, 2020-10-25:
To my (very) untrained eye, it looks like the codebases for the Adafruit MatrixPortal and Pi (with use of a Adafruit bonnet or similar interface hat) aren't that dissimilar, I was wondering if a port could be done relatively easily.  It would certainly cut down on the hardware overhead and footprint (the single M4 vs. Zero W and Bonnet).

There are 2 mainissues I can think of with such a port:   the compatibility of the errant RGB matrices to the MatrixPortal M4, and updating/config mechanism of the JSON.

As far as the matrices go, there doesn't seem to be a complete analog to hzeller's rgbmatrix library on MatrixPortal.  For example, I can't do specific multiplexing on panels on the M4 like I can on the Pi, which narrows the field of compatible 32x64 panels to the one that Adafruit sells.

Updating the JSON might not be a terrible thing if SSH access could be allowed to the M4, or if there's a rudimentary backend GUI that could be utilized.  Hell, you even might be able to pull it off via BLE.

riffnshred, 2020-10-25:
Thats something I considered awhile back when I started but using the Feather M4. The Matrix portal and all arduino based rgb matrix board are good at one thing and one thing only, showing what ever you want it to show. Its not powerful enough to add the data parsing, state managing and refreshing everything and then build the image to show on screen. With the Rpi, one core is dedicated to rendering and except for the Pi zero, the rest is handle by the 3 other cores. You can actually see a difference in how fast the screen rotate between the two.

The equivalent of Hzeller's rgb matrix library for arduino (and also whats used in circuit python) is Protomatter, which like you said is not as developped. You can't even adjust the brightness with it. I got an issue open requesting that feature but I think it will take awhile before someone works on it. 

The only way I can see the matrix portal work is if you have an app running on your network that does the data parsing and state managing, build the images and send that image to the matrix portal to show on screen. To do so hzeller have 2 other libraries called PixelPusher and Flaschen-taschen. Thats where Id start. If id take on that project. Figure out how to send images to the portal from a server.

I let me know if you start it, I can explain mine work in more details to help port it.

cheechbg, 2020-10-26:
Very much appreciate the feedback.  I absolutely see what you're talking about with the "single-use" nature of the device.  While my background is wayyy far away from being able to tackle a port of this kind (more of a networking/packet plumber myself ;) ), I do have some feelers on the Adafruit forum to see if there's a way to get over the multiplexing issue since I feel like that's the most low-hanging fruit out there.

closed 2020-10-26 by cheechbg

riffnshred, 2020-10-26:
It would be great if they do but unless someone from the community tackle it, I doubt it will happen. Which is sad because with the rpi version, there is already a good reference to make it work because the base code is C like for arduino. I think all we need is a project so great that it will annoy someone with the right skills enough that those features are not there and decide to tackle them. xD.

falkyre, 2020-10-26:
I plan on using my matrix portal as a client board to the main code and push data to it.  Not sure quite yet the whole plan but it's an idea I'll explore after I build my external goal horn.