riffnshred/nhl-led-scoreboard #147: SSL Issue with OWM

OHS-awoodbridge, 2020-11-09:
I set everything up and it worked perfectly at home. I tried to take it my setup into work, and I'm having a certificate issue. I'm assuming it's related to being on the work network, but I don't even know where to begin to track this down.

Here is my error (I had to retype it because I can't get a SSH connection to my Pi due to network issues):
Unable to get OWM data error:HTTPSConnectionPool((host='api.openweathermap.org',port=443): Max retries exceeded with url: /data/2.5/weather?APPID=<mykey>$lang=en&lon=-122.8650075&lat=47.0451022 (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self signed certificate in certificate chain (_ssl.c:1056)')))

**Hardware you are using**
- Raspberry pi model Zero W
- The Matrix driver: Adafruit Bonnet
- Anti-flicker mod or not?  Yes

**Software**
- Which version of the software are you on? 1.4.1

falkyre, 2020-11-10:
So the error that's coming up is the certificate being returned back is showing as self signed.  Are you going through a proxy at work?

If you run the following on the command line:

sudo python3 src/main.py --led-gpio-mapping=adafruit-hat-pwm --led-slowdown-gpio=3 --updatecheck --loglevel='DEBUG'

You'll see a ton of messages coming from the different libraries being used.    Run the above for about 10 seconds or so then ctrl-C to quit.  You should see a bunch of lines.  Look for something like the following:

`2020-11-09 20:22:41 urllib3.connectionpool DEBUG https://api.openweathermap.org:443 "GET /data/2.5/weather?APPID=xxxxx5a952xxxxxxxxdfcf4cd9f&lang=en&lon=-97.3047582&lat=49.8823812 HTTP/1.1" 200 475`

That's a good call to OWM.  What are you seeing?

OHS-awoodbridge, 2020-11-10:
There is not a proxy setup on local machines, but I know there is filtering happening at the top of the network (I'm in a school). I couldn't get the --loglevel = 'DEBUG' or iterations of it to be accepted
..

OHS-awoodbridge, 2020-11-10:
I edited the config for OWM library to not check certificates and I'm good now. I know it wasn't really your problem to solve, but thinks for trying to help.

falkyre, 2020-11-10:
Yah the proxy is injecting its own certificate.  The loglevel stuff is coming in next release.

riffnshred, 2021-01-04:
@falkyre  this was fixed in 1.5.0 right ?

falkyre, 2021-01-04:
No. It's not something we can fix as it was the network injecting it's own ssl certificate.  He fixed it himself.  There's nothing that can be done at a coding level easily.

riffnshred, 2021-01-04:
misread the whole thing. *facepalm* 

Thanks

closed 2021-01-04 by riffnshred