riffnshred/nhl-led-scoreboard #153: Getting N/A on the first weather board

dmryglod, 2020-12-01:
**Describe the bug**
On the weather board, I am getting N/A for some of weather data.

**Logs**
Log excerpt:

2020-11-30 20:18:05 INFO Checking for EC weather alerts
2020-11-30 20:18:05 INFO ['11/30 08:18 PM', '\uf07b', 'Curr Cond N/A', '4.9C', '0.0C', '42%', '-7.1C']
2020-11-30 20:18:05 INFO ['29.0 kph', 'NW', '\uf05d', '59.0 kph', '1009.0 hPa', '\uf057', '24.1 km']
2020-11-30 20:18:06 INFO Last Alert: []

**Software**
V. 1.5

falkyre, 2020-12-01:
What's your location?

dmryglod, 2020-12-01:
Edmonton

falkyre, 2020-12-01:
And does it happen all the time?

falkyre, 2020-12-01:
OK, I've set my location to be Edmonton, Alberta in the config.json and have recreated the issue you are seeing.  This is happening because the way the Environment Canada python library works and what's getting sent back from EC's data warehouse.  My code is expecting that the condition is returned back with a value but when you put Edmonton, Alberta in the location in the config.json, the data is coming back empty.  This is out of the control of my code, I can only use what I get.  However, if you leave the location empty, then your location will be figured out by your IP address and give a lat/lon value to the code.  This usually is more precise but can fail sometimes depending on your provider and who registered the block of IP addresses they supply to customers.  So a better way is to get your lat/lon from Google Maps or even put in your home address and that should get a better, more precise location for the weather to look up on the EC website.

dmryglod, 2020-12-01:
And format for lay/long - degrees, minutes, seconds or decimal degrees? Is it XXX°N, YYY°W or XXX°, -YYY°?

falkyre, 2020-12-02:
43.8956,-98.0233 is an example good lat/lon.  Btw, I'm.seeing issues coming from the environment Canada datamart right now for Winnipeg as well

dmryglod, 2020-12-02:
That did it! Thanks!

closed 2020-12-02 by dmryglod