riffnshred/nhl-led-scoreboard #155: ERROR Unable to connect to EC

HkyMan92, 2020-12-21:
I am new to Pi and haven't coded since high school doing html by hand in the mid 90s, but I love this project and wanted to get back into it.  I followed all steps in the README on a 3B+ with Raspbian OS Desktop, and am now trying to test everything out.  
When I run 
sudo python3 src/main.py --led-gpio-mapping=adafruit-hat-pwm --led-brightness=60 --led-slowdown-gpio=2  from ~nhl-led-scoreboard S
I get
INFO NHL-LED-SCOREBOARD - v1.5.0 (64x32)
ERROR Unable to connect to EC, try running again in a few minutes

What are my next steps to correcting this error? Thank you in advance.

falkyre, 2020-12-22:
What are you using as a location in your config.json? Anything or is it blank?

riffnshred, 2021-03-24:
closing due to inactivity

closed 2021-03-24 by riffnshred