riffnshred/nhl-led-scoreboard #158: Won’t boot directly into scoreboard

drodgers24, 2020-12-30:
Building out a scoreboard and needing to get it to run stand alone and boot directly into the scoreboard program. I have the power button added and it works. I followed the instruction to use supervisor but when I reboot it will boot like normal and ask for username and password and won’t boot up scoreboard program. 

Am I missing a step?

Thanks!

jlm4487, 2020-12-30:
It will always boot to the username and password when it reboots but supervisor should start up the app in the background. Can you access supervisor via the web browser? If so, what does it show?

drodgers24, 2020-12-30:
It does pull up and show running when I pull up supervisor on the web. The issue is that this is going to be a gift for someone else and was hoping they would be able to just turn it on and have it run without having to login. Is there a way around the login or another option to be able to have them just start it?

riffnshred, 2020-12-30:
Just found out some issues with the last update. I'll work on that tonight and try to push a fix. Sorry about that

riffnshred, 2021-01-03:
can you confirm your issue is resolved? I believe the issue was with bad config in Supervisor right?

riffnshred, 2021-03-24:
closing do to no update from OP

closed 2021-03-24 by riffnshred