riffnshred/nhl-led-scoreboard #160: Missing json file

jeepdrvr71, 2021-01-01:
Just updated the scoreboard and got it all working yesterday This morning it stopped working and got the following error in Supervisor. I'm running the latest version.

fig file config/layout/layout_64x32.json not found!
Config file config/fonts/fonts_64x32.json not found!
Config file config/layout/layout_64x32.json not found!
Config file config/fonts/fonts_64x32.json not found!
Config file config/layout/layout_64x32.json not found!
Config file config/fonts/fonts_64x32.json not found!
Config file config/layout/layout_64x32.json not found!
Config file config/fonts/fonts_64x32.json not found!
Config file config/layout/layout_64x32.json not found!
Config file config/fonts/fonts_64x32.json not found!
Config file config/layout/layout_64x32.json not found!
Config file config/fonts/fonts_64x32.json not found!
Config file config/layout/layout_64x32.json not found!
Config file config/fonts/fonts_64x32.json not found!
Config file config/layout/layout_64x32.json not found!
Config file config/fonts/fonts_64x32.json not found!
Config file config/layout/layout_64x32.json not found!
Config file config/fonts/fonts_64x32.json not found!
Config file config/layout/layout_64x32.json not found!

falkyre, 2021-01-01:
That's a normal "error".  The board is not running because there's been a change in the NHL data feed that breaks the code.

jeepdrvr71, 2021-01-01:
Did this just happen in the last 24 hours?

falkyre, 2021-01-01:
Yup

riffnshred, 2021-01-03:
Fix with version 1.5.4

closed 2021-01-03 by riffnshred