riffnshred/nhl-led-scoreboard #161: NHL API change for conferences for 2021 causes board to crash

falkyre, 2021-01-01:
It appears that in preparation for the 2021 season, the NHL API has changed for conferences and now is causing the board to crash when the data is refreshed for the day

**Logs**
`
nhl_api/__init__.py", line 44, in standings
    standings, wildcard = nhl_api.info.standings()
  File "/home/dietpi/source/weather/src/nhl_api/info.py", line 137, in standings
    conference_name = divisions[division]['conference']['name']
KeyError: 'conference'
INFO: Validating config.json.....
INFO: config.json passes validation
Traceback (most recent call last):
  File "./src/main.py", line 139, in <module>
    run()
  File "./src/main.py", line 52, in run
    data = Data(config)
  File "/home/dietpi/source/weather/src/data/data.py", line 162, in __init__
    self.refresh_standings()
  File "/home/dietpi/source/weather/src/data/data.py", line 365, in refresh_standings
    self.standings = nhl_api.standings()
  File "/home/dietpi/source/weather/src/nhl_api/__init__.py", line 44, in standings
    standings, wildcard = nhl_api.info.standings()
  File "/home/dietpi/source/weather/src/nhl_api/info.py", line 137, in standings
    conference_name = divisions[division]['conference']['name']
KeyError: 'conference'
`

See the config.json.sample (that's pretty much mine today)

- Raspberry pi 3A+ 
- both Adafruit Bonnet and Adafruit HAT (happening on all boards I have)
- Anti-flicker mod or not? Yes, anti-flicker is on

**Software**
- 1.5.3

riffnshred, 2021-01-02:
Damn you Lord Stanley !!! You strike again. 

Fixes incoming! Testing everything else making sure the standings are the only things that changed.

falkyre, 2021-01-02:
Ok I figured I'd open this and let everyone know so you weren't blasted with issues.

riffnshred, 2021-01-02:
Good idea !! Thanks !

saxonpaul92, 2021-01-02:
DEBUG log...

pi@raspberrypi:~/nhl-led-scoreboard $ sudo python3 src/main.py --led-gpio-mapping=adafruit-hat-pwm --led-brightness=60 --led-slowdown-gpio=2 --led-rows=32 --led-cols=64 --loglevel=DEBUG
INFO: Validating config.json.....
INFO: config.json passes validation
Config file config/fonts/fonts_64x32.json not found!
Config file config/layout/layout_64x32.json not found!
Traceback (most recent call last):
  File "src/main.py", line 139, in <module>
    run()
  File "src/main.py", line 52, in run
    data = Data(config)
  File "/home/pi/nhl-led-scoreboard/src/data/data.py", line 162, in __init__
    self.refresh_standings()
  File "/home/pi/nhl-led-scoreboard/src/data/data.py", line 365, in refresh_standings
    self.standings = nhl_api.standings()
  File "/home/pi/nhl-led-scoreboard/src/nhl_api/__init__.py", line 44, in standings
    standings, wildcard = nhl_api.info.standings()
  File "/home/pi/nhl-led-scoreboard/src/nhl_api/info.py", line 137, in standings
    conference_name = divisions[division]['conference']['name']
KeyError: 'conference'

syslog....
Jan  1 20:54:38 raspberrypi supervisord[446]: 2021-01-01 20:54:38,850 INFO spawned: 'scoreboard' with pid 1271
Jan  1 20:54:39 raspberrypi supervisord[446]: 2021-01-01 20:54:39,858 INFO success: scoreboard entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
Jan  1 20:54:46 raspberrypi supervisord[446]: 2021-01-01 20:54:46,428 INFO exited: scoreboard (exit status 1; not expected)
Jan  1 20:54:47 raspberrypi supervisord[446]: 2021-01-01 20:54:47,441 INFO spawned: 'scoreboard' with pid 1275
Jan  1 20:54:48 raspberrypi supervisord[446]: 2021-01-01 20:54:48,449 INFO success: scoreboard entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
Jan  1 20:54:55 raspberrypi supervisord[446]: 2021-01-01 20:54:55,060 INFO exited: scoreboard (exit status 1; not expected)
Jan  1 20:54:56 raspberrypi supervisord[446]: 2021-01-01 20:54:56,070 INFO spawned: 'scoreboard' with pid 1279
Jan  1 20:54:57 raspberrypi supervisord[446]: 2021-01-01 20:54:57,078 INFO success: scoreboard entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
Jan  1 20:55:04 raspberrypi supervisord[446]: 2021-01-01 20:55:04,307 INFO waiting for scoreboard to stop
Jan  1 20:55:04 raspberrypi supervisord[446]: 2021-01-01 20:55:04,334 INFO stopped: scoreboard (terminated by SIGTERM)

riffnshred, 2021-01-03:
Fix with the latest update (1.5.4)

closed 2021-01-03 by riffnshred