riffnshred/nhl-led-scoreboard #162: covid19 - weather - Christmas boards not displaying on offday on pi zero w 32x64 led display

machinasdei, 2021-01-03:
covid19 - weather - Christmas boards not displaying on offday on pi zero w

noting that covid19 board did display successfully on the led display in version 1.5.1 - this capability was lost beginning in version 1.5.2 +. (i'm currently running on 1.5.4). I only see the clock and screensavers for my team, and standings.  I would see it pull covid19 info in the INFO output in version 1.5.1 when the main.py ran, but i haven't seen that friendly output in 1.5.2 +

it used to look like this in version 1.5.1
"states": {
        "off_day": [
            "covid_19",
            "scoreticker",
            "clock"

i tried to change it in the 1.5.2 + configs from covid19 to covid_19 = does not display the covid19 board.
-----------------------------------------

- for Weather - i have not seen the weather board - it is enabled in the config (see below for current running config) noting for the weather that my api is valid and confirmed working - it does pull weather in the display, but not a weather board - i used to see it pull weather info in the INFO output in version 1.5.1 when the main.py ran, but i haven't seen that friendly output in 1.5.2 +
-----------------------------------------

-as for the christmas board - i've never seen it working, but added it under the 
"states": {
        "off_day": [
            "scoreticker",
            "team_summary",
            "standings",
            "covid19",
            "christmas",
            "weather",
            "clock"

still haven't seen it.
-------------------------------
**Hardware you are using**
- Raspberry pi model - pi zero w
- The Matrix driver - Adafruit Bonnet 
- Anti-flicker mod or not? -yes 

**Software**
- Which version of the software are you on? raspbian buster 
- Linux pi-Ticker 5.4.73+ #1360 Thu Oct 29 15:48:27 GMT 2020 armv6l GNU/Linux

-----------------------------------
-My current config running in version 1.5.4;

cat nhl-led-scoreboard/config/config.json
{
    "debug": false,
    "loglevel": "INFO",
    "live_mode": false,
    "preferences": {
        "time_format": "12h",
        "end_of_day": "8:00",
        "location": "",
        "live_game_refresh_rate": 10,
        "teams": [
            "Capitals"
        ],
        "sog_display_frequency": 4,
        "goal_animations": {
            "pref_team_only": true
        }
    },
    "states": {
        "off_day": [
            "scoreticker",
            "team_summary",
            "standings",
            "covid19",
            "weather",
            "clock"
        ],
        "scheduled": [
            "team_summary",
            "standings",
            "scoreticker",
            "clock"
        ],
        "intermission": [
            "scoreticker",
            "team_summary"
        ],
        "post_game": [
            "team_summary",
            "standings",
            "scoreticker",
            "clock"
        ]
    },
    "boards": {
        "scoreticker": {
            "preferred_teams_only": false,
            "rotation_rate": 5
        },
        "seriesticker": {
            "preferred_teams_only": true,
            "rotation_rate": 5
        },
        "standings": {
            "preferred_standings_only": true,
            "standing_type": "wild_card",
            "divisions": "central",
            "conference": "eastern"
        },
        "clock": {
            "duration": 15,
            "hide_indicator": false,
            "preferred_team_colors": true,
            "clock_rgb": "",
            "date_rgb": "",
            "flash_seconds": true
        },
        "covid19": {
            "worldwide_enabled": true,
            "country_enabled": true,
            "country": [
                "USA"
            ],
            "us_state_enabled": false,
            "us_state": [
                "Virginia"
            ],
            "canada_enabled": false,
            "canada_prov": [
                "Quebec"
            ]
        },
        "weather": {
            "enabled": true,
            "view": "full",
            "units": "imperial",
            "duration": 60,
            "data_feed": "EC",
            "owm_apikey": "<my weather api key is in here>",
            "update_freq": 5,
            "show_on_clock": true,
            "forecast_enabled": true,
            "forecast_days": 3,
            "forecast_update": 1
        },
        "wxalert": {
            "alert_feed": "EC",
            "update_freq": 5,
            "show_alerts": true,
            "nws_show_expire": false,
            "alert_title": true,
            "scroll_alert": true,
            "alert_duration": 5,
            "show_on_clock": true
        }
    },
    "sbio": {
        "screensaver": {
            "enabled": true,
            "animations": true,
            "start": "22:00",
            "stop": "22:05",
            "data_updates": false,
            "motionsensor": true,
            "pin": 24,
            "delay": 30
        },
        "dimmer": {
            "enabled": false,
            "source": "software",
            "daytime": "",
            "nighttime": "",
            "offset": 90,
            "frequency": 5,
            "light_level_lux": 400,
            "mode": "always",
            "sunset_brightness": 5,
            "sunrise_brightness": 40
        },
        "pushbutton": {
            "enabled": false,
            "bonnet": false,
            "pin": 25,
            "reboot_duration": 2,
            "reboot_override_process": "",
            "display_reboot": true,
            "poweroff_duration": 10,
            "poweroff_override_process": "",
            "display_halt": true,
            "state_triggered1": "weather",
            "state_triggered1_process": ""
        }
    }
}
root@pi-Ticker:/home/me#

falkyre, 2021-01-03:
How did you create your config?

machinasdei, 2021-01-03:
Really didn't do anything different, just using the default configuration
with each version, and enabling where it makes sense and put my API key.

riffnshred, 2021-01-04:
This was resolved in the discord channel and it was due to a bad config file.

closed 2021-01-04 by riffnshred