riffnshred/nhl-led-scoreboard #165: Team Blue on clock hard to read

bridge2nowhere, 2021-01-04:
On the team colors clock for the Sabres, the blue is really hard to read,. I don't think it's the same color as the logo blue on the team summary screen. 

- Is it easy for me to change the color myself if so where is that value stored?
- Is it possible to change to the clock color to white or the lighter royal in the logo instead of the navy?

riffnshred, 2021-01-04:
for sure. if you look in the config.json under boards/clock you will see you can put the RGB value for the clock and the date or you can set the preferred team color to false and it will be the standard red.

riffnshred, 2021-01-04:
forgot to mention, if you use the nhl_setup, you will find those option there too.

riffnshred, 2021-01-04:
I just remembered the Sabres returned to their old colors. I'll do the change for the next update. in the meantime here are the colors. Primary (0,48,135), text (243,208,62).

closed 2021-01-04 by riffnshred