riffnshred/nhl-led-scoreboard #170: UI and App for V2.0

riffnshred, 2021-01-10:
**Description**

Here are some things that I think the App should have and be able to do.

- Take care of the configuration and offer a quicker and easier way to change options on the fly.
- To see the schedule of the day and pick a game to display would be cool. (Think of how the NHLGame app works)
- To reboot or shut down the raspberry pi (not sure you can turn it on using software when it's completely shut down).
- Controls for brightness, show different boards, Pause or stop cycling between boards  (let's say you just want to see one at all time).

falkyre, 2021-01-10:
Also need ability to pause and start schedulers for weather and dimmer.  See the screensaver.py in the src/sbio directory.

xxxDrewedxxx, 2021-01-11:
I’d like the ability to not only turn the boards on or off, but also turn the “pulls” or “requests” to the api(s) on and off.  I.E. turn off the hockey completely in the off season so it stops checking the api each day, but keeps the weather board going.   Or limit it to once per day.  Is my team playing today? No, ok I’ll check again tomorrow.

riffnshred, 2021-01-11:
That's what the plan is for the core on V2. check when the board turns on and on new days if there's anything going on. If so, update standings and the schedule of the day. If your team has a game that day, it will add a trigger in the scheduler to switch to the Live state and start its refresh every 15 seconds (or the refresh rate the user set in the config). There will be options for lets say you want to refresh the data of the other games regularly (at a refresh rate set by the user) so if you use the scoreticker and care about having the data updated as it goes, you can.

bazookajoey, 2021-01-12:
What kind of connection will the app and scoreboard have with each other?

sflems, 2021-01-24:
I'd definitely be interested in putting together a simple app for configuring the scoreboard. I'm working on ideas for my final project in the CS50 Web path and this would certainly fit the bill. I was thinking it could be put together with Django and run locally as a python server to connect and send commands and configs to the pi/scoreboard from a web interface similar to PiHole. Let me know if this is something we could work on!

riffnshred, 2021-01-24:
> I'd definitely be interested in putting together a simple app for configuring the scoreboard. I'm working on ideas for my final project in the CS50 Web path and this would certainly fit the bill. I was thinking it could be put together with Django and run locally as a python server to connect and send commands and configs to the pi/scoreboard from a web interface similar to PiHole. I'm just working on my hardware list to build my first board as a project for the Pops. Let me know if this is something we could work on!

For V1 that would be really helpful as some people dread the ol terminal. Let's open another issue for that more specifically.

joshkay, 2021-01-24:
@sflems Are you familiar with React or Flutter at all?  Originally I was thinking of setting this up using Flutter, but I think React would be alot easier for me.  You could implement the settings portion.  Would be a good place to start.

sflems, 2021-01-24:
I've played with React a smidge. Nothing beyond toying around with a few basic concepts. Let me see how it would mesh and be built around what I've got here.