riffnshred/nhl-led-scoreboard #171: Step by Step Directions for Newcomers

jwallner0710, 2021-01-13:
After a few long nights of reading through these pages, I was finally able to get this bad boy up and running. I will admit, the HowChoo website is just the tip of the iceberg for this project. It requires a bunch of reading. You guys have done a lot of establish this group. I almost wish there was a step by step instruction manual for the software aspect of this project. For the average person, I was bouncing back and forth from forum to Howchoo to github, all of which I was completely unfamiliar with going into this project. However, I have survived. I'm good to go, but for any person in the future trying to complete this project, a more centralized step by step instruction manual might be beneficial. Maybe I'm not knowledgeable enough to understand that may not be possible with all of the constant changes, updates, and fixes. However, I'm politely reflecting on my experience over the last few days, and wishing I had something like that. In conclusion, thank you all for your efforts to help out and inform to make this whole project even possible. Thank you!!! Let's Go Blues!!

richmyers, 2021-01-14:
Average Joe here too.  I successfully have the board running....except I don't have the weather boards displaying correctly.  I am not sure how to install the weather api's correctly.  I have basic weather and humidity on the clock.....so something is right.  Any tips?  And.....I Bleed Blue!

falkyre, 2021-01-14:
Do you have weather as a board to display in the off_day, scheduled, intermission or post_game states in your config?  How did you create the config file?  Using the nhl_setup app and doing a simple setup or full setup?

richmyers, 2021-01-16:
weather board is configured in all "states" except intermission.  I used the nhl_setup app....but have also manually edited the config.json.  I have tried both the default OWM key and the one I registered for.  I am sure I am missing something.  Thanks for the support.

example of my config file
  "states": {
        "off_day": [
            "scoreticker",
            "team_summary",
            "standings",
            "clock",
            "weather"
        ],

"weather": {
            "enabled": true,
            "view": "full",
            "units": "imperial",
            "duration": 30,
            "data_feed": "OWM",
            "owm_apikey": "700423d40a49cb0611213c77bc3c59ff",
            "update_freq": 5,
            "show_on_clock": true,
            "forecast_enabled": true,
            "forecast_days": 3,
            "forecast_update": 1

riffnshred, 2021-01-16:
@richmyers open another issue about the weather issues, has this is unrelated.