riffnshred/nhl-led-scoreboard #172: Clock showing the wrong time even if I got the right timezone configured.

riffnshred, 2021-01-13:
**Describe the bug**
The Raspberry pi don't sync time with the internet on boot. Been having that issue Since the new Raspberry pi OS came out. 

Found the solution here : https://raspberrypi.stackexchange.com/questions/103458/cant-use-sudo-systemctl-start-myservice

the solution is simple. 

Run `sudo nano /boot/cmdline.txt` and add `init=/bin/systemd` at the end of the whole line. 

DO NOT USE THE LINE BELOW, USE IT AS AN EXEMPLE!!!!!

It should look like something like this: `console=serial0,115200 console=tty1 root=PARTUUID=e070d3d2-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait init=/bin/systemd`

CTRL+X and then ENTER. now reboot . `sudo reboot -n`

To confirm the time is synced, run this command. it will tell you the current time. `timedatectl`

brucebrophy, 2021-01-30:
Hey @riffnshred thank you for this amazing project! 

I just want to let you know that this command isn't working for me. 

Here's my OS information:
```
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

And the contents of `/boot/cmdline.txt`

```
console=serial0,115200 console=tty1 root=PARTUUID=71e02c9c-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait init=/bin/systemd
```

brucebrophy, 2021-01-30:
Just wanted to follow up since I got something else working for me and I hope it helps someone else.

`timedatectl set-timezone America/New_York` Change this to your timezone 
`timedatectl set-ntp true`
`sudo reboot -n`

You can check if you have your system clock synchronized by running `timedatectl status`.

@riffnshred let me know if you want to me add anything to the readme for this project.

riffnshred, 2021-03-24:
Your method was the one that failed to work for me for some reason. It might be something related to how I had my raspberry pi configured. 

Thanks for the offer. Since this is something that should not happen if you install and configure Raspberry pi OS with the new Raspberry Pi Imager, It's not necessary to add it to the read me. this Issue will be good enough documentation incase someone else has the same issues.

Cheers

closed 2021-03-24 by riffnshred