riffnshred/nhl-led-scoreboard #173: not loading setup

ceegmac, 2021-01-13:
Hi, just trying to get scoreboard to load after install however i keep getting setup error
 ________________  _____
  / __/ __/_  __/ / / / _ \
 _\ \/ _/  / / / /_/ / ___/
/___/___/ /_/  \____/_/

1.5.0
Traceback (most recent call last):
  File "nhl_setup.py", line 1605, in <module>
  File "nhl_setup.py", line 1518, in main
PermissionError: [Errno 13] Permission denied: 'config/.default/firstrun'
[1130] Failed to execute script nhl_setup
pi@raspberrypi:~/nhl-led-scoreboard $

not sure what i am doing wrong. worked before update but not after. i can get the test screen to work however

riffnshred, 2021-01-13:
that is because the firstrun file is missing.

you can create one by running this from the nhl-led-scoreboard directory

`touch config/.default/firstrun`

ceegmac, 2021-01-13:
it says permission denied

riffnshred, 2021-01-13:
Also heads up. an update is pending for tomorrow morning after testing with the first live game tonight. Standings, seriesticker and stanley_cup_champions boards are not working at the moment.

riffnshred, 2021-01-13:
what is you do `sudo touch config/.default/firstrun`

ceegmac, 2021-01-13:
got this now

Traceback (most recent call last):
  File "nhl_setup.py", line 1605, in <module>
  File "nhl_setup.py", line 1518, in main
PermissionError: [Errno 13] Permission denied: 'config/.default/firstrun'
[1602] Failed to execute script nhl_setup

riffnshred, 2021-01-13:
you'll have to run the nhl_setup with sudo (which is not ideal, we are figuring those issues out on our end)

so run `sudo ./nhl_setup`

That should work

ceegmac, 2021-01-13:
yes worked thnx

closed 2021-01-13 by ceegmac