riffnshred/nhl-led-scoreboard #176: Permission denied while using the nhl_setup

riffnshred, 2021-01-14:
This Error comes up :

`Traceback (most recent call last):
  File "nhl_setup.py", line 1605, in <module>
  File "nhl_setup.py", line 1518, in main
PermissionError: [Errno 13] Permission denied: 'config/.default/firstrun'
[14118] Failed to execute script nhl_setup`

If so, please remove the nhl-led-scoreboard from the raspberry pi, like so:

`cd`
`rm -r nhl-led-scoreboard`

Then reinstall the software by starting over from "Installing the NHL scoreboard software" in the documentation. Make sure you do not use `sudo` for the git commands otherwise you will need to start over again.

falkyre, 2021-01-14:
Did some more validation and if you run the install using sudo, the 3 git commands in the install.sh will change some critical git files to be owned by the root user.  This causes the Permission Denied error on the .git/FETCH_HEAD when trying to do a git pull.  The fix for an existing install is to do `sudo chown -R [username] .git/` where [username] is whatever user you are using to log into your pi.  For most users, this wiull be **pi** (on dietpi it's dietpi).  To not have this issue, the install.sh script needs to be run as the user you are logged in as.

So the steps in the documentation:

`sudo chmod +x scripts/install`
 `sudo ./scripts/install.sh`

are now 

`chmod +x scripts/install.sh`
 `./scripts/install.sh`

This will remove any issues with permissions that are happening.  All of the sudo steps we need for initial install are handled in the actual install.sh

riffnshred, 2021-01-14:
Thank you for that !!! will update the documentation

riffnshred, 2021-01-17:
As pointed out by @falkyre on discord, an alternative to removing and reinstalling the software is to change the ownership of the software. If you have pi as username (default user) you can run this:

`sudo chown -R pi:pi /home/pi/nhl-led-scoreboard.`

sflems, 2021-02-09:
With regards to the past permission issues, would changing the base permissions on the install.sh in the main repo have any effect (or be of concern) going forward? It could possibly save a step for the installees.

`git update-index --chmod=+x scripts/install.sh` on the repos _should_ remove the need for users to chmod the install.sh script altogether.

falkyre, 2021-02-09:
> With regards to the past permission issues, would changing the base permissions on the install.sh in the main repo have any effect (or be of concern) going forward? It could possibly save a step for [...]

That would totally help out by removing a step in the documentation which does get breezed over quite often.  Part of the permission problems did come from the update checking code in the first version was updating the git repository when running as root.  The newest updatecheck code no longer does that.  As well, sudo is no longer required to run the install.sh, but again we have some users who have been with us since the beginning that still do their updates by running sudo scripts/install.sh which will cause issues with the 3 git commands.  I'm working on a new update script that should help with a lot of the upgrading and permission stuff.  We are working our way towards having our own way of doing stuff and this project matures.  @riffnshred based his initial code on the MLB scoreboard and I believe the initial install.sh was based on what they did.

riffnshred, 2021-02-09:
> @riffnshred based his initial code on the MLB scoreboard and I believe the initial install.sh was based on what they did

Thats exactly right. Im currently rewriting the documentation as @falkyre is working on making things less prone too errors on the user side of things.

sflems, 2021-02-09:
Right on 😎

closed 2021-02-19 by riffnshred