riffnshred/nhl-led-scoreboard #177: KeyError: 'abbreviation'

bigbadbrad1, 2021-01-14:
Hi I've been getting this KeyError: 'abbreviation' error.
I had the board working fine during the offseason but so far for 2021 no luck. I've tried running the install.sh a few times to see if it would fix the issue. I apologize if this is an easy fix, I'm not good with coding.

Here's the log file:
`INFO (16:01:24): NHL-LED-SCOREBOARD - v1.4.1 (128x64)
INFO (16:01:24): Now validating config.json.....
INFO (16:01:25): config.json passes validation
INFO (16:01:25): location is: Toronto, ON [43.6534817, -79.3839347]
Traceback (most recent call last):
  File "src/main.py", line 117, in <module>
    run()
  File "src/main.py", line 53, in run
    data = Data(config)
  File "/home/pi/nhl-led-scoreboard/src/data/data.py", line 111, in __init__
    self.teams = self.get_teams()
  File "/home/pi/nhl-led-scoreboard/src/data/data.py", line 221, in get_teams
    teams = nhl_api.teams()
  File "/home/pi/nhl-led-scoreboard/src/nhl_api/__init__.py", line 23, in teams
    return [nhl_api.info.Info(x) for x in nhl_api.info.team_info()]
  File "/home/pi/nhl-led-scoreboard/src/nhl_api/info.py", line 24, in team_info
    division_abbrev = team['division']['abbreviation']
KeyError: 'abbreviation'`

riffnshred, 2021-01-14:
You will need to update your scoreboard. its because the Standings have changed in the API. This have been fixed in the new release.

Run the commend in the documentation [here ](https://github.com/riffnshred/nhl-led-scoreboard#updating-your-software) to update your scoreboard. If you have issues with Permissions, delete and reinstall the software as indicated in issue #176. make sure to read the commend of that issue. 

Cheers

bigbadbrad1, 2021-01-14:
Thanks a lot.
I've got it working now!

closed 2021-01-14 by bigbadbrad1