riffnshred/nhl-led-scoreboard #178: Weather Boards not displaying

richmyers, 2021-01-16:
weather board is added to all "states" but does not display on board.  Temp and humidity is displayed on bottom of clock

**Your config file**
A screen shot of your config.json file can be really useful to help you. 

**Hardware you are using**
- Raspberry pi model - Pi Zero W
- The Matrix driver (Adafruit Bonnet or Adafruit HAT) - Adafruit Bonnet
- Anti-flicker mod or not?  - anit flicker installed

**Software**
- Which version of the software are you on? - Version 1.5.4m 

Config File excerpt

pi@PiScoreBrd:~/nhl-led-scoreboard/config $ more config.json
{
    "debug": false,
    "loglevel": "INFO",
    "live_mode": true,
    "preferences": {
        "time_format": "12h",
        "end_of_day": "00:00",
        "location": "64012",
        "live_game_refresh_rate": 10,
        "teams": [
            "Blues"
        ],
        "sog_display_frequency": 4,
        "goal_animations": {
            "pref_team_only": true
        }
    },
    "states": {
        "off_day": [
            "scoreticker",
            "team_summary",
            "standings",
            "clock",
            "weather"
        ],
        "scheduled": [
            "team_summary",
            "standings",
            "scoreticker",
            "clock",
            "weather"
        ],
        "intermission": [
            "scoreticker",
            "team_summary"
        ],
        "post_game": [
            "team_summary",
            "standings",
            "scoreticker",
            "clock",
            "weather"
        ]
    },

        "weather": {
            "enabled": true,
            "view": "full",
            "units": "imperial",
            "duration": 30,
            "data_feed": "OWM",
            "owm_apikey": "700423d40axxxxxxxxxxxx",
            "update_freq": 5,
            "show_on_clock": true,
            "forecast_enabled": true,
            "forecast_days": 3,
            "forecast_update": 1
        },

richmyers, 2021-01-16:
tail log from Supervisor Status on restart

2021-01-16 09:43:36 INFO Checking for new day
2021-01-16 09:43:36 INFO It is not a new day
Config file config/fonts/fonts_64x32.json not found!
Config file config/layout/layout_64x32.json not found!
2021-01-16 09:45:39 INFO NHL-LED-SCOREBOARD - v1.5.4 (64x32)
2021-01-16 09:45:47 INFO Refreshing OWM current observations weather
2021-01-16 09:45:48 INFO ['01/16 09:45 AM', '\uf002', 'broken clouds', '46.4F', '42.4F', '76%', '41.1F']
2021-01-16 09:45:48 INFO ['3.0mph', 'WNW', '\uf05d', '0mph', '1028 MB', '\uf07b', '6.2 mi']
2021-01-16 09:45:48 INFO Checking NWS weather alerts
Previous request failed with code 400. Retrying...
Previous Response: {
    "correlationId": "988366e",
    "title": "Invalid Parameter",
    "type": "https://api.weather.gov/problems/InvalidParameter",
    "status": 400,
    "detail": "Parameter \"point\" is invalid: out of bounds",
    "instance": "https://api.weather.gov/requests/988366e"
}
Previous request failed with code 400. Retrying...
Previous Response: {
    "correlationId": "14060818",
    "title": "Invalid Parameter",
    "type": "https://api.weather.gov/problems/InvalidParameter",
    "status": 400,
    "detail": "Parameter \"point\" is invalid: out of bounds",
    "instance": "https://api.weather.gov/requests/14060818"
}
Previous request failed with code 400. Retrying...
Previous Response: {
    "correlationId": "14062673",
    "title": "Invalid Parameter",
    "type": "https://api.weather.gov/problems/InvalidParameter",
    "status": 400,
    "detail": "Parameter \"point\" is invalid: out of bounds",
    "instance": "https://api.weather.gov/requests/14062673"
}
Previous request failed with code 400. Retrying...
Previous Response: {
    "correlationId": "9888fe0",
    "title": "Invalid Parameter",
    "type": "https://api.weather.gov/problems/InvalidParameter",
    "status": 400,
    "detail": "Parameter \"point\" is invalid: out of bounds",
    "instance": "https://api.weather.gov/requests/9888fe0"
}
Previous request failed with code 400. Retrying...
Previous Response: {
    "correlationId": "5d0c73d",
    "title": "Invalid Parameter",
    "type": "https://api.weather.gov/problems/InvalidParameter",
    "status": 400,
    "detail": "Parameter \"point\" is invalid: out of bounds",
    "instance": "https://api.weather.gov/requests/5d0c73d"
}
2021-01-16 09:46:49 ERROR Maximum retries exceeded. Response object dump: <Response [400]>
2021-01-16 09:46:50 INFO Updating weather forecast every 1 hour(s) starting @ 10:00
2021-01-16 09:46:50 INFO Refreshing OWM daily weather forecast
2021-01-16 09:46:51 INFO New forecast: [['Sun 01/17', 'moderate rain', '\uf008', '49F', '44F'], ['Mon 01/18', 'broken clouds', '\uf002', '50F', '38F'], ['Tue 01/19', 'clear sky', '\uf002', '59F', '41F']]
2021-01-16 09:46:51 INFO Forecast has changed and been updated....
2021-01-16 09:46:51 INFO [['Sun 01/17', 'moderate rain', '\uf008', '49F', '44F'], ['Mon 01/18', 'broken clouds', '\uf002', '50F', '38F'], ['Tue 01/19', 'clear sky', '\uf002', '59F', '41F']]
2021-01-16 09:46:51 INFO Weather forecast next update @ 10:00
2021-01-16 09:46:51 INFO Using software to determine dimming
2021-01-16 09:46:52 INFO Checking for dimmer
2021-01-16 09:46:52 INFO Using software to determine dimming
2021-01-16 09:46:52 INFO Using location for dimmer:  Sunset is @ 2021-01-16 12:54:11.561725  Sunrise is @ 2021-01-17 02:28:16.678322  Offset by 90 minutes forward
2021-01-16 09:46:52 INFO It is day time
2021-01-16 09:46:52 INFO Screen saver will start @ 00:00 and end @ 08:00
2021-01-16 09:46:52 INFO Rendering...
2021-01-16 09:46:52 INFO Your preferred teams are Off today

falkyre, 2021-01-16:
Take out the zip code from your location and leave blank. Or out city, state in. The zip code is causing the nws alerts to crash.

richmyers, 2021-01-26:
replaced with city,state   No change in displaying board.

falkyre, 2021-01-26:
Are you using the weather alerts and using NWS as a provider?  If so, the NWS api server is currently down and the app gets stuck in a loop with it on.

falkyre, 2021-01-26:
If you update the noaa-sdk library it does not give up after 5 tries.  To upgrade it, `sudo pip3 install --upgrade noaa-sdk `

riffnshred, 2021-03-24:
closing this issue as the OP didn give any update.

closed 2021-03-24 by riffnshred