riffnshred/nhl-led-scoreboard #180: Scoreboard doesn't update to latest version

Q492V395P029, 2021-01-16:
Description:  When I try to update the NHL scoreboard it says it is up to date but it doesn’t update to the latest version.

I tried to update my system from 1.3.0 to 1.5.1 a while back and it only ever updated to 1.5.0 and then would say it is up to date despite 1.5.1 being out. Ever since then I have had this problem where it doesn’t update to the latest version. Now that 1.5.4+ is out my system says it is completely up to date with 1.5.1

I have included a screenshot that shows 
- The end of the update process completing and it showing “already up to date”
- The setup process on screen that shows the version number
- An error I am getting now

When I recently updated to 1.5.1 it made me delete my config, I couldn’t complete updating or installing without deleting 2 config files. Before that my config file was up to date, I remade it for 1.5.0 update. 

I don’t know if this helps but I have had the sudo problem for a while as well, I have to use sudo to update the system.

If you need more information about the issue let me know. I have a full disk image backup of the system with 1.3.0 installed, I would consider using that but I originally updated because the clock stopped working and I had the '32x64 config not found' error I saw someone else posted about as well.

Any help is appreciated. I don't really know how to post pictures either, I hope the link works but I don't know if it will.

[image]

If the picture doesn't post the error is
PermissionError: [Errno 13] Permission denied: 'config/.default/firstrun'
[3073] Failed to execute script nhl_setup

Q492V395P029, 2021-01-16:
[image]

Q492V395P029, 2021-01-16:
[image]

riffnshred, 2021-01-16:
This is related to issue #176 . Follow the steps there, especially the comment.

closed 2021-01-16 by Q492V395P029