riffnshred/nhl-led-scoreboard #182: LED Refresh Rate Limiter broke

twfarley, 2021-01-16:
Traceback (most recent call last):
  File "src/main.py", line 139, in <module>
    run()
  File "src/main.py", line 43, in run
    matrixOptions = led_matrix_options(commandArgs)
  File "/home/pi/nhl-led-scoreboard/src/utils.py", line 128, in led_matrix_options
    options.limit_refresh_rate_hz = args.led_limit_refresh
AttributeError: 'rgbmatrix.core.RGBMatrixOptions' object has no attribute 'limit_refresh_rate_hz'

riffnshred, 2021-01-16:
Are you on v1.5.5?

twfarley, 2021-01-16:
Yep.

riffnshred, 2021-01-16:
it does work on my end. can you show me the command you use to run your board ?

twfarley, 2021-01-16:
`sudo python3 src/main.py --led-gpio-mapping=adafruit-hat-pwm`

riffnshred, 2021-01-16:
check the status of the submodule
`git submodule status`. copy what it gives you here

twfarley, 2021-01-16:
```
$ git submodule status
 84e1465e9ea5ed000011d05369c5287eaa361ad7 submodules/matrix (remotes/origin/pixel-mapper-206-g84e1465)
```

riffnshred, 2021-01-16:
thats a weird one. you are on the right version of the submodule. You updated from which version do you know ?

riffnshred, 2021-01-16:
If shows you that error on startup no matter what? If so I would delete the repo and reclone and reconfig.

falkyre, 2021-01-16:
You have to reinstall the rgb matrix library.  Run through the whole make and compile and o
Pip install. I've done it for two if my boards that had same issue and the rgb matrix install steps fixed it

twfarley, 2021-01-17:
Sure. You should probably make a note of that in the commit.

closed 2021-01-17 by twfarley