riffnshred/nhl-led-scoreboard #184: App Constantly Restarting, Never Displays

patsieber, 2021-01-17:
When I run the main function, nothing ever displays, and if I look at the log, I see the following when Debug is enabled. I can't make sense of why it's looping forever and never displaying. 

`sudo python3 src/main.py --led-gpio-mapping=adafruit-hat-pwm --led-brightness=60 --led-slowdown-gpio=2
INFO: Validating config.json.....
INFO: config.json passes validation
No data for 20202021 Playoff
No default round for 20202021 Playoff.
21-01-17 18:01:43 INFO Scheduler started
21-01-17 18:01:44 DEBUG Looking for jobs to run
21-01-17 18:01:44 DEBUG No jobs; waiting until a job is added
21-01-17 18:01:44 DEBUG Starting new HTTPS connection (1): dd.weather.gc.ca:443
21-01-17 18:01:45 DEBUG https://dd.weather.gc.ca:443 "GET /citypage_weather/docs/site_list_en.csv HTTP/1.1" 200 32894
21-01-17 18:02:41 DEBUG Starting new HTTPS connection (1): dd.weather.gc.ca:443
21-01-17 18:02:43 DEBUG https://dd.weather.gc.ca:443 "GET /citypage_weather/xml/ON/s0000169_e.xml HTTP/1.1" 200 2961
21-01-17 18:02:44 DEBUG Starting new HTTPS connection (1): dd.weather.gc.ca:443
21-01-17 18:02:45 DEBUG https://dd.weather.gc.ca:443 "GET /air_quality/doc/AQHI_XML_File_List.xml HTTP/1.1" 200 6306
21-01-17 18:02:59 DEBUG Starting new HTTPS connection (1): dd.weather.gc.ca:443
21-01-17 18:03:00 DEBUG https://dd.weather.gc.ca:443 "GET /air_quality/aqhi/ont/observation/realtime/xml/AQ_OBS_FEVJR_CURRENT.xml HTTP/1.1" 200 409
21-01-17 18:03:00 DEBUG Starting new HTTPS connection (1): dd.weather.gc.ca:443
21-01-17 18:03:03 DEBUG https://dd.weather.gc.ca:443 "GET /air_quality/aqhi/ont/forecast/realtime/xml/AQ_FCST_FEVJR_CURRENT.xml HTTP/1.1" 200 619
21-01-17 18:03:14 INFO Added job "ecWxAlerts.getAlerts" to job store "default"
21-01-17 18:03:14 DEBUG Looking for jobs to run
21-01-17 18:03:14 DEBUG Next wakeup is due at 2021-01-17 18:07:31.037589-05:00 (in 256.706873 seconds)
21-01-17 18:03:14 DEBUG Starting new HTTPS connection (1): dd.weather.gc.ca:443
21-01-17 18:03:16 DEBUG https://dd.weather.gc.ca:443 "GET /citypage_weather/xml/ON/s0000169_e.xml HTTP/1.1" 200 2967
21-01-17 18:03:16 DEBUG Starting new HTTPS connection (1): dd.weather.gc.ca:443
21-01-17 18:03:19 DEBUG https://dd.weather.gc.ca:443 "GET /air_quality/aqhi/ont/observation/realtime/xml/AQ_OBS_FEVJR_CURRENT.xml HTTP/1.1" 200 409
21-01-17 18:03:19 DEBUG Starting new HTTPS connection (1): dd.weather.gc.ca:443
21-01-17 18:03:21 DEBUG https://dd.weather.gc.ca:443 "GET /air_quality/aqhi/ont/forecast/realtime/xml/AQ_FCST_FEVJR_CURRENT.xml HTTP/1.1" 200 619
21-01-17 18:03:23 INFO Added job "screenSaver.runSaver" to job store "default"
21-01-17 18:03:23 DEBUG Looking for jobs to run
21-01-17 18:03:23 DEBUG Next wakeup is due at 2021-01-17 18:07:31.037589-05:00 (in 247.907746 seconds)
21-01-17 18:03:23 INFO Added job "screenSaver.stopSaver" to job store "default"
21-01-17 18:03:23 DEBUG Looking for jobs to run
21-01-17 18:03:23 DEBUG Next wakeup is due at 2021-01-17 18:07:31.037589-05:00 (in 247.827356 seconds)
21-01-17 18:03:23 DEBUG Starting new HTTP connection (1): statsapi.web.nhl.com:80
21-01-17 18:03:23 DEBUG http://statsapi.web.nhl.com:80 "GET /api/v1/schedule?date=2021-1-17&expand=schedule.linescore HTTP/1.1" 200 None
21-01-17 18:03:46 DEBUG STREAM b'IHDR' 16 13
21-01-17 18:03:46 DEBUG STREAM b'IDAT' 41 2404
21-01-17 18:07:31 DEBUG Looking for jobs to run
21-01-17 18:07:31 DEBUG Next wakeup is due at 2021-01-17 18:13:34.826644-05:00 (in 363.747877 seconds)
21-01-17 18:07:31 INFO Running job "ecWxAlerts.getAlerts (trigger: interval[0:05:00], next run at: 2021-01-17 18:13:34 EST)" (scheduled at 2021-01-17 18:07:31.037589-05:00)
21-01-17 18:07:31 DEBUG Starting new HTTPS connection (1): dd.weather.gc.ca:443
21-01-17 18:07:33 DEBUG https://dd.weather.gc.ca:443 "GET /citypage_weather/xml/ON/s0000169_e.xml HTTP/1.1" 200 2967
21-01-17 18:07:34 DEBUG Starting new HTTPS connection (1): dd.weather.gc.ca:443
21-01-17 18:07:38 DEBUG https://dd.weather.gc.ca:443 "GET /air_quality/aqhi/ont/observation/realtime/xml/AQ_OBS_FEVJR_CURRENT.xml HTTP/1.1" 200 409
21-01-17 18:07:39 DEBUG Starting new HTTPS connection (1): dd.weather.gc.ca:443
21-01-17 18:07:41 DEBUG https://dd.weather.gc.ca:443 "GET /air_quality/aqhi/ont/forecast/realtime/xml/AQ_FCST_FEVJR_CURRENT.xml HTTP/1.1" 200 619
21-01-17 18:07:42 INFO Job "ecWxAlerts.getAlerts (trigger: interval[0:05:00], next run at: 2021-01-17 18:13:34 EST)" executed successfully
21-01-17 18:08:39 DEBUG STREAM b'IHDR' 16 13
21-01-17 18:08:39 DEBUG STREAM b'bKGD' 41 6
21-01-17 18:08:39 DEBUG b'bKGD' 41 6 (unknown)
21-01-17 18:08:39 DEBUG STREAM b'IDAT' 59 8192
21-01-17 18:08:40 DEBUG STREAM b'IHDR' 16 13
21-01-17 18:08:40 DEBUG STREAM b'tEXt' 41 25
21-01-17 18:08:40 DEBUG STREAM b'iTXt' 78 883
21-01-17 18:08:40 DEBUG STREAM b'IDAT' 973 291
`

patsieber, 2021-01-17:
I used a default config setup to start, because I was getting the same error when I did a full configuration, and wanted to eliminate potential factors.

falkyre, 2021-01-18:
Does your screen display with the demo code? If so, can you run with info level instead of debug and see if it crashes. The debugs aren't showing any errors, it all looks ok.

patsieber, 2021-01-18:
Yes, I can get the Hello World sample to display, if that's what you're asking. My config file shows the following, so I would have assumed info logs should be showing up: 

"debug": true,
"loglevel": "INFO",

falkyre, 2021-01-18:
Set debug to false.

patsieber, 2021-01-18:
Got it. Here's my last run, which I stopped after about 10 minutes without anything appearing on the screen. 

sudo python3 src/main.py --led-gpio-mapping=adafruit-hat-pwm --led-brightness=60 --led-slowdown-gpio=2
INFO: Validating config.json.....
INFO: config.json passes validation
No data for 20202021 Playoff
No default round for 20202021 Playoff.
2021-01-18 17:34:43 INFO NHL-LED-SCOREBOARD - v1.5.5 (64x32)
2021-01-18 17:35:37 INFO Checking for EC weather alerts
2021-01-18 17:35:41 INFO Last Alert: []
2021-01-18 17:35:41 INFO No active EC weather alerts in your area
2021-01-18 17:35:41 INFO Screen saver will start @ 22:00 and end @ 22:05
2021-01-18 17:35:41 INFO Rendering...
2021-01-18 17:35:41 INFO Live mode is off. Going through the boards
2021-01-18 17:35:41 INFO PING !!! Render off day
2021-01-18 17:35:41 INFO Checking for new day
2021-01-18 17:35:42 INFO It is not a new day
2021-01-18 17:40:50 INFO Checking for EC weather alerts
2021-01-18 17:40:55 INFO Last Alert: []
2021-01-18 17:40:55 INFO No active EC weather alerts in your area
2021-01-18 17:44:26 INFO Checking for EC weather alerts
2021-01-18 17:44:30 INFO Last Alert: []
2021-01-18 17:44:30 INFO No active EC weather alerts in your area

falkyre, 2021-01-18:
Ok. You aren't in live mode. So it will always be in off_day.  What board are in the odf_day state?

patsieber, 2021-01-18:
I've tried with both. Now all my other settings are just the default "simple" version from the nhl_config utility. Offday has scoreticker, team_summary, standings, and clock.

falkyre, 2021-01-19:
Take all other boards out aside from the clock.  Does it display?  Set everything in the config for weather, weather alerts, forecast, dimmer, screensaver, motion sensor (in the screen saver) from enabled to disabled.  Make sure the location is set to blank.  Does the board display?

patsieber, 2021-01-19:
I ran the configuration on default mode, then only changed offday to show only clock, and the clock displayed correctly.

patsieber, 2021-01-19:
I tried different combinations of boards. Clock and team summary worked, as did clock and standings. Clock and score ticker did not.   However, changing Live mode to true, and then just running everything else as-is, I see the upcoming game screen, then the team summary, then the standings, but then it seems to stay on the bottom of the standings forever, which was strange. Here's the log for that:
2021-01-19 11:27:08 INFO NHL-LED-SCOREBOARD - v1.5.5 (64x32)
2021-01-19 11:28:03 INFO Checking for EC weather alerts
2021-01-19 11:28:07 INFO Last Alert: []
2021-01-19 11:28:08 INFO No active EC weather alerts in your area
2021-01-19 11:28:08 INFO Screen saver will start @ 22:00 and end @ 22:05
2021-01-19 11:28:08 INFO Rendering...
2021-01-19 11:28:08 INFO Game Day Wooooo
2021-01-19 11:28:08 INFO Showing Game
2021-01-19 11:28:09 INFO Checking for new day
2021-01-19 11:28:09 INFO It is not a new day
2021-01-19 11:28:09 INFO Game is Scheduled
2021-01-19 11:28:09 INFO Showing Pre-Game

Thank you for your help, sorry this is a pain.

falkyre, 2021-01-19:
There's on going issues with the NHL API that is causing some issues that are out of our control.  What you are seeing is likely a result of that.  Can you close the issue now that you are working like everyone else?

patsieber, 2021-01-19:
Ok, I will keep my eye out for updates. I'm not sure mine is "working like everyone else", but if the API is a bigger issue, I will wait for that to be addressed from their end.

closed 2021-01-19 by patsieber