riffnshred/nhl-led-scoreboard #185: Show live update for multiple favorite teams

Pending31, 2021-01-18:
I've got multiple favorite teams set (Favorite team from where I'm from, and a second one for my current home team).  Everything works fine if they're playing at the same time, in that it defaults to favorite #1, and then if they're not playing, favorite #2.  On days where they're playing at different times, it works if favorite #1 plays first, however; if favorite #2 has an afternoon game and favorite #1 has an evening game, it stays in the pre-game mode (which I have set as the league scoreticker).

I was wondering if it would be possible to add a logic loop to look at the start times on a day where multiple favorites are playing, and then play a lower favorite until favorite #1 starts?

riffnshred, 2021-01-18:
I got a couple of features coming by the end of this week, I'll see if I can add that to the bunch. 

I'll keep this issue open as it's something that some asked before on discord.

riffnshred, 2021-01-22:
good news, It will be in the next update.

riffnshred, 2021-03-24:
This is now a feature since v1.6.0

closed 2021-03-24 by riffnshred