riffnshred/nhl-led-scoreboard #186: Team Stats not updating - Standings not showing

FyLapS, 2021-01-18:
Thanks so much for this scoreboard! I've had it up and running since the start of the season and I'm loving it - I appreciate all the hard work you've put into this!

Everything seems to be working well for me (live scores, score recaps, etc). However, my team stats is not updating. For example, its still showing me that the Maple Leafs (my favorite team - don't judge me) are 1-0 and their next game is vs OTT on Jan 15th.

In addition, I've added the standings board but it never shows up. I've tried displaying conference (which probably won't work since conferences don't seem to be listed on NHL?) with no avail. I've also tried showing by division but the division names have changed, so I tried inputting "North" and "Scotia North" but neither seems to work.

Any help on these would be great! Cheers

desigi, 2021-01-19:
I've noticed the same thing regarding the stats not updating. If I restart the board, the stats get updated correctly, until the next game is played (then they are out of date again).

So what I've noted is the standings and team summary boards don't update automatically after a game is played.

FyLapS, 2021-01-20:
The team stats just updated to the previous game (it says next game is Jan 18th, even tho its Jan 20). Prior to this, it was still showing Jan 15th, so it looks like it may be updating with a few days delay?

Still not getting any standings. Anyone have the same issue? (I've used the default settings listed on the main page and tried modifying slightly to include different possibilities with division names with no change)

riffnshred, 2021-01-20:
the standing currently are `north` `east` `west` `central`. I think the default one in the basic config is `atlantic` and that's why its not working. try that out and let me know.

Regarding the team summary and stats in general, it will be fix in the upcoming update. It probably updated because you Supervisor restarted your scoreboard after a crash or something

FyLapS, 2021-01-20:
Its been set like this:

"standings": {
            "preferred_standings_only": false,
            "standing_type": "division",
            "divisions": "north",
            "conference": "eastern"
        },

standings is in my offdays, scheduled, and post game sections.

riffnshred, 2021-01-20:
which version are you running ? You can see it with in the first few lines when the scoreboard start up. it should be 1.5.5

There was a couple of update in the last month and I think you might be on the previous one (1.5.4) , which had issues with standings.

FyLapS, 2021-01-20:
I'm 1.5.4 - drat I thought I had the latest update (I just installed it 2 weeks ago). Will update now :)

riffnshred, 2021-01-20:
Sounds good. FYI I got a new update coming probably next week with new features and some minor fixes.

riffnshred, 2021-03-24:
the data not updating properly has been fixed in the latest update (v1.6.2)

closed 2021-03-24 by riffnshred