riffnshred/nhl-led-scoreboard #187: Board goes blank erroring on CAR logo PNG

mgrayaz, 2021-01-22:
**Describe the bug**
The scoreboard runs as expected for a few minutes then goes blank and throws an error. It looks as though the error is on the Carolina Logo PNG so it may be dependent on days they have a game to recreate.

**Logs**
Traceback (most recent call last):
  File "src/main.py", line 139, in <module>
    run()
  File "src/main.py", line 134, in run
    MainRenderer(matrix, data, sleepEvent).render()
  File "/home/pi/nhl-led-scoreboard/src/renderer/main.py", line 66, in render
    self.__render_offday()
  File "/home/pi/nhl-led-scoreboard/src/renderer/main.py", line 94, in __render_offday
    self.boards._off_day(self.data, self.matrix,self.sleepEvent)
  File "/home/pi/nhl-led-scoreboard/src/boards/boards.py", line 81, in _off_day
    board(data, matrix,sleepEvent)
  File "/home/pi/nhl-led-scoreboard/src/boards/boards.py", line 222, in scoreticker
    Scoreticker(data, matrix, sleepEvent).render()
  File "/home/pi/nhl-led-scoreboard/src/boards/scoreticker.py", line 31, in render
    ScoreboardRenderer(self.data, self.matrix, Scoreboard(self.games[self.index], self.data)).render()
  File "/home/pi/nhl-led-scoreboard/src/renderer/scoreboard.py", line 24, in __init__
    'home'
  File "/home/pi/nhl-led-scoreboard/src/renderer/logos.py", line 24, in __init__
    self.load(team_abbrev)
  File "/home/pi/nhl-led-scoreboard/src/renderer/logos.py", line 42, in load
    self.logo = Image.open(filename)
  File "/usr/lib/python3/dist-packages/PIL/Image.py", line 2687, in open
    % (filename if filename else fp))
OSError: cannot identify image file '/home/pi/nhl-led-scoreboard/assets/logos/CAR/light/64x32.png'

**Your config file**
pi@scoreboard:~ $ cat nhl-led-scoreboard/config/config.json
{
    "debug": false,
    "loglevel": "INFO",
    "live_mode": false,
    "preferences": {
        "time_format": "12h",
        "end_of_day": "8:00",
        "location": "",
        "live_game_refresh_rate": 10,
        "teams": [
            "Avalanche"
        ],
        "sog_display_frequency": 4,
        "goal_animations": {
            "pref_team_only": true
        }
    },
    "states": {
        "off_day": [
            "scoreticker",
            "team_summary",
            "standings",
            "clock"
        ],
        "scheduled": [
            "team_summary",
            "standings",
            "scoreticker",
            "clock"
        ],
        "intermission": [
            "scoreticker",
            "team_summary"
        ],
        "post_game": [
            "team_summary",
            "standings",
            "scoreticker",
            "clock"
        ]
    },
    "boards": {
        "scoreticker": {
            "preferred_teams_only": false,
            "rotation_rate": 5
        },
        "seriesticker": {
            "preferred_teams_only": true,
            "rotation_rate": 5
        },
        "standings": {
            "preferred_standings_only": true,
            "standing_type": "wild_card",
            "divisions": "central",
            "conference": "eastern"
        },
        "clock": {
            "duration": 15,
            "hide_indicator": false,
            "preferred_team_colors": true,
            "clock_rgb": "",
            "date_rgb": "",
            "flash_seconds": true
        },
        "covid19": {
            "worldwide_enabled": false,
            "country_enabled": false,
            "country": [
                "USA"
            ],
            "us_state_enabled": false,
            "us_state": [
                "New York"
            ],
            "canada_enabled": false,
            "canada_prov": [
                "Quebec"
            ]
        },
        "weather": {
            "enabled": false,
            "view": "full",
            "units": "metric",
            "duration": 60,
            "data_feed": "EC",
            "owm_apikey": "",
            "update_freq": 5,
            "show_on_clock": true,
            "forecast_enabled": true,
            "forecast_days": 3,
            "forecast_update": 1
        },
        "wxalert": {
            "alert_feed": "EC",
            "update_freq": 5,
            "show_alerts": true,
            "nws_show_expire": false,
            "alert_title": true,
            "scroll_alert": true,
            "alert_duration": 5,
            "show_on_clock": true
        }
    },
    "sbio": {
        "screensaver": {
            "enabled": true,
            "animations": true,
            "start": "22:00",
            "stop": "22:05",
            "data_updates": false,
            "motionsensor": true,
            "pin": 24,
            "delay": 30
        },
        "dimmer": {
            "enabled": false,
            "source": "software",
            "daytime": "",
            "nighttime": "",
            "offset": 90,
            "frequency": 5,
            "light_level_lux": 400,
            "mode": "always",
            "sunset_brightness": 5,
            "sunrise_brightness": 40
        },
        "pushbutton": {
            "enabled": false,
            "bonnet": false,
            "pin": 25,
            "reboot_duration": 2,
            "reboot_override_process": "",
            "display_reboot": true,
            "poweroff_duration": 10,
            "poweroff_override_process": "",
            "display_halt": true,
            "state_triggered1": "weather",
            "state_triggered1_process": ""
        }
    }
}pi@scoreboard:~ $

**Hardware you are using**
- Raspberry pi model = zero wh
- The Matrix driver (Adafruit Bonnet or Adafruit HAT) = Bonnet
- Anti-flicker mod or not = anti-flicker, however I had the same issue prior installing the anti-flicker mod so it appears to not be dependent on that

**Software**
- Which version of the software are you on =1.5.5

riffnshred, 2021-01-23:
Found the issue. It was due to the generated logos being owned by root instead of the user. This will be fix for next update

mgrayaz, 2021-01-28:
This occured again today on version 1.5.6. I run a daily cron to reboot at 1:05am and startup the board on reboot, to mitigate any no update issues. Seems to only happen on CAR logo, it has been running smoothly for about 5 days. Here are todays cron logs:

pi@scoreboard:~ $ cat ~/cron.log
INFO: Validating config.json.....
INFO: config.json passes validation
No data for 20202021 Playoff
No default round for 20202021 Playoff.
2021-01-28 01:06:14 INFO NHL-LED-SCOREBOARD - v1.5.6 (64x32)
2021-01-28 01:06:22 INFO Refreshing OWM current observations weather
2021-01-28 01:06:23 INFO ['01/28 01:06 AM', '\uf002', 'clear sky', '40.1F', '31.7F', '80%', '36.0F']
2021-01-28 01:06:23 INFO ['9.2mph', 'NE', '\uf05e', '0mph', '1033 MB', '\uf07b', '6.2 mi']
2021-01-28 01:06:23 INFO Checking NWS weather alerts
2021-01-28 01:06:27 INFO Number of alerts is 0
2021-01-28 01:06:27 INFO No active NWS alerts in your area
2021-01-28 01:06:27 INFO Updating weather forecast every 1 hour(s) starting @ 02:00
2021-01-28 01:06:27 INFO Refreshing OWM daily weather forecast
2021-01-28 01:06:29 INFO New forecast: [['Fri 01/29', 'clear sky', '\uf002', '62F', '42F'], ['Sat 01/30', 'overcast clouds', '\uf002', '74F', '58F'], ['Sun 01/31', 'clear sky', '\uf002', '63F', '50F']]
2021-01-28 01:06:29 INFO Forecast has changed and been updated....
2021-01-28 01:06:29 INFO [['Fri 01/29', 'clear sky', '\uf002', '62F', '42F'], ['Sat 01/30', 'overcast clouds', '\uf002', '74F', '58F'], ['Sun 01/31', 'clear sky', '\uf002', '63F', '50F']]
2021-01-28 01:06:29 INFO Weather forecast next update @ 02:00
2021-01-28 01:06:29 INFO Screen saver will start @ 22:00 and end @ 22:05
2021-01-28 01:06:30 INFO Rendering...
2021-01-28 01:06:30 INFO Your preferred teams are Off today
2021-01-28 01:06:30 INFO PING !!! Render off day
2021-01-28 01:06:30 INFO Checking for new day
2021-01-28 01:06:30 INFO It is not a new day
2021-01-28 01:09:55 INFO PING !!! Render off day
2021-01-28 01:09:55 INFO Checking for new day
2021-01-28 01:09:55 INFO It is not a new day
2021-01-28 01:11:36 INFO Refreshing OWM current observations weather
2021-01-28 01:11:38 INFO ['01/28 01:11 AM', '\uf002', 'clear sky', '40.0F', '31.6F', '80%', '35.9F']
2021-01-28 01:11:38 INFO ['9.2mph', 'NE', '\uf05e', '0mph', '1033 MB', '\uf07b', '6.2 mi']
2021-01-28 01:12:46 INFO Checking NWS weather alerts
2021-01-28 01:12:49 INFO Number of alerts is 0
2021-01-28 01:12:49 INFO No active NWS alerts in your area
2021-01-28 01:13:20 INFO PING !!! Render off day
2021-01-28 01:13:20 INFO Checking for new day
2021-01-28 01:13:20 INFO It is not a new day
2021-01-28 01:16:47 INFO PING !!! Render off day
2021-01-28 01:16:47 INFO Checking for new day
2021-01-28 01:16:47 INFO It is not a new day
2021-01-28 01:16:55 INFO Refreshing OWM current observations weather
2021-01-28 01:16:59 INFO ['01/28 01:16 AM', '\uf002', 'clear sky', '40.0F', '31.6F', '80%', '35.9F']
2021-01-28 01:16:59 INFO ['9.2mph', 'NE', '\uf05e', '0mph', '1033 MB', '\uf07b', '6.2 mi']
2021-01-28 01:18:58 INFO Checking NWS weather alerts
2021-01-28 01:19:02 INFO Number of alerts is 0
2021-01-28 01:19:02 INFO No active NWS alerts in your area
2021-01-28 01:20:15 INFO PING !!! Render off day
2021-01-28 01:20:15 INFO Checking for new day
2021-01-28 01:20:15 INFO It is not a new day
2021-01-28 01:22:50 INFO Refreshing OWM current observations weather
2021-01-28 01:22:52 INFO ['01/28 01:22 AM', '\uf002', 'clear sky', '39.7F', '31.3F', '80%', '35.6F']
2021-01-28 01:22:52 INFO ['9.2mph', 'NE', '\uf05e', '0mph', '1033 MB', '\uf07b', '6.2 mi']
2021-01-28 01:23:40 INFO PING !!! Render off day
2021-01-28 01:23:40 INFO Checking for new day
2021-01-28 01:23:40 INFO It is not a new day
2021-01-28 01:24:44 INFO Checking NWS weather alerts
2021-01-28 01:24:50 INFO Number of alerts is 0
2021-01-28 01:24:50 INFO No active NWS alerts in your area
2021-01-28 01:27:07 INFO PING !!! Render off day
2021-01-28 01:27:07 INFO Checking for new day
2021-01-28 01:27:08 INFO It is not a new day
2021-01-28 01:28:38 INFO Refreshing OWM current observations weather
2021-01-28 01:28:40 INFO ['01/28 01:28 AM', '\uf002', 'clear sky', '39.7F', '31.3F', '80%', '35.6F']
2021-01-28 01:28:40 INFO ['9.2mph', 'NE', '\uf05e', '0mph', '1033 MB', '\uf07b', '6.2 mi']
2021-01-28 01:29:52 INFO Checking NWS weather alerts
2021-01-28 01:29:56 INFO Number of alerts is 0
2021-01-28 01:29:56 INFO No active NWS alerts in your area
2021-01-28 01:30:32 INFO PING !!! Render off day
2021-01-28 01:30:32 INFO Checking for new day
[... 1251 lines cut ...]
    run()
  File "src/main.py", line 134, in run
    MainRenderer(matrix, data, sleepEvent).render()
  File "/home/pi/nhl-led-scoreboard/src/renderer/main.py", line 74, in render
    self.__render_game_day()
  File "/home/pi/nhl-led-scoreboard/src/renderer/main.py", line 193, in __render_game_day
    self.boards._scheduled(self.data, self.matrix,self.sleepEvent)
  File "/home/pi/nhl-led-scoreboard/src/boards/boards.py", line 123, in _scheduled
    board(data, matrix,sleepEvent)
  File "/home/pi/nhl-led-scoreboard/src/boards/boards.py", line 222, in scoreticker
    Scoreticker(data, matrix, sleepEvent).render()
  File "/home/pi/nhl-led-scoreboard/src/boards/scoreticker.py", line 31, in render
    ScoreboardRenderer(self.data, self.matrix, Scoreboard(self.games[self.index], self.data)).render()
  File "/home/pi/nhl-led-scoreboard/src/renderer/scoreboard.py", line 24, in __init__
    'home'
  File "/home/pi/nhl-led-scoreboard/src/renderer/logos.py", line 29, in __init__
    self.load(team_abbrev)
  File "/home/pi/nhl-led-scoreboard/src/renderer/logos.py", line 47, in load
    self.logo = Image.open(filename)
  File "/usr/lib/python3/dist-packages/PIL/Image.py", line 2687, in open
    % (filename if filename else fp))
OSError: cannot identify image file '/home/pi/nhl-led-scoreboard/assets/logos/CAR/light/64x32.png'

pi@scoreboard:~ $ cat nhl-led-scoreboard/VERSION
1.5.6

riffnshred, 2021-01-28:
Try deleting and reinstalling the software. It's probably because the Carolina logo is still owned by root on the raspberry pi.

mgrayaz, 2021-02-01:
Reinstalled and ran the whole day with no issues, even with Carolina playing. This issue appears to be resolved in latest version. Thank you!

closed 2021-02-19 by riffnshred