riffnshred/nhl-led-scoreboard #188: Rendering Error on startup

rhazzed, 2021-01-23:
**Describe the bug**
A clear and concise description of what the bug is.

When starting the board (manually or through supervisor) there is a rendering error that display which prevents any 
of the boards (except the clock) from displaying,

**Logs**
INFO: Validating config.json.....
INFO: config.json passes validation
No data for 20202021 Playoff
No default round for 20202021 Playoff.
2021-01-23 01:02:06 INFO NHL-LED-SCOREBOARD - v1.5.5 (128x64)
2021-01-23 01:02:06 INFO Rendering...
2021-01-23 01:02:06 INFO Game Day Wooooo
2021-01-23 01:02:06 INFO Showing Game
2021-01-23 01:02:06 INFO Checking for new day
2021-01-23 01:02:06 INFO It is not a new day
2021-01-23 01:02:06 INFO Game is Scheduled
2021-01-23 01:02:06 INFO Showing Pre-Game
2021-01-23 01:02:07 INFO ERROR WHILE RENDERING: 'JSONData' object has no attribute 'scoreboard'
2021-01-23 01:02:07 INFO Refreshing data in a minute
(it keep cycling over and over here.. scoreboard never displays)

**Hardware you are using**
- Raspberry pi model:   3B+
- The Matrix driver (Adafruit Bonnet or Adafruit HAT):   Bonnet
- Anti-flicker mod or not? :     yes

**Software**
- Which version of the software are you on?   1.5.5

riffnshred, 2021-01-23:
Can you post your config bud? also set `debug` to true and loglevel to ERROR and show me what it spits out.

rhazzed, 2021-01-23:
{

    "debug": true,

    "loglevel": "INFO",

    "live_mode": true,

    "preferences": {

        "time_format": "12h",

        "end_of_day": "00:01",

        "location": "Overland, MO",

        "live_game_refresh_rate": 10,

        "teams": [

            "Blues"

        ],

        "sog_display_frequency": 4,

        "goal_animations": {

            "pref_team_only": true

        }

    },

    "states": {

        "off_day": [

            "clock",

            "scoreticker",

            "seriesticker",

            "standings",

            "team_summary",

            "covid_19"

        ],

        "scheduled": [

            "clock",

            "scoreticker",

            "seriesticker",

            "standings",

            "team_summary"

        ],

        "intermission": [

            "scoreticker",

            "seriesticker",

            "standings",

            "team_summary"

        ],

        "post_game": [

            "clock",

            "scoreticker",

            "seriesticker",

            "standings",

            "team_summary"

        ]

    },

    "boards": {

        "scoreticker": {

            "preferred_teams_only": false,

            "rotation_rate": 5

        },

        "seriesticker": {

            "preferred_teams_only": false,

            "rotation_rate": 5

        },

        "standings": {

            "preferred_standings_only": false,

            "standing_type": "division",

            "divisions": "west",

            "conference": "western"

        },

        "clock": {

            "duration": 5,

            "hide_indicator": false,

            "preferred_team_colors": true,

            "clock_rgb": "",

            "date_rgb": "",

            "flash_seconds": true

        },

        "covid19": {

            "worldwide_enabled": false,

            "country_enabled": true,

            "country": [

                "USA"

            ],

            "us_state_enabled": true,

            "us_state": [

                "Missouri"

            ],

            "canada_enabled": false,

            "canada_prov": [

                "Quebec"

            ]

        },

        "weather": {

            "enabled": false,

            "view": "full",

            "units": "imperial",

            "duration": 30,

            "data_feed": "OWM",

            "owm_apikey": "790de1ce734eb1f784672adb7c832da3",

            "update_freq": 5,

            "show_on_clock": false,

            "forecast_enabled": false,

            "forecast_days": 1,

            "forecast_update": 1

        },

        "wxalert": {

            "alert_feed": "NWS",

            "update_freq": 5,

            "show_alerts": false,

            "nws_show_expire": true,

            "alert_title": false,

            "scroll_alert": false,

            "alert_duration": 5,

            "show_on_clock": false

        }

    },

    "sbio": {

        "screensaver": {

            "enabled": false,

            "animations": false,

            "start": "22:00",

            "stop": "22:05",

            "data_updates": false,

            "motionsensor": false,

            "pin": 24,

            "delay": 30

        },

        "dimmer": {

            "enabled": false,

            "source": "software",

            "daytime": "",

            "nighttime": "",

            "offset": 90,

            "frequency": 5,

            "light_level_lux": 400,

            "mode": "always",

            "sunset_brightness": 5,

            "sunrise_brightness": 40

        },

        "pushbutton": {

            "enabled": false,

            "bonnet": false,

            "pin": 25,

            "reboot_duration": 2,

            "reboot_override_process": "",

            "display_reboot": true,

            "poweroff_duration": 10,

            "poweroff_override_process": "",

            "display_halt": true,

            "state_triggered1": "weather",

            "state_triggered1_process": ""

        }

    }

}

rhazzed, 2021-01-23:
With debug on, it just spits this out…  never gets past my clock.

 

 

INFO: Validating config.json.....

INFO: config.json passes validation

No data for 20202021 Playoff

No default round for 20202021 Playoff.

21-01-23 07:31:49 INFO Scheduler started

21-01-23 07:31:49 DEBUG Looking for jobs to run

21-01-23 07:31:49 DEBUG Starting new HTTP connection (1): statsapi.web.nhl.com:80

21-01-23 07:31:49 DEBUG No jobs; waiting until a job is added

21-01-23 07:31:49 DEBUG http://statsapi.web.nhl.com:80 "GET /api/v1/game/2020020078/feed/live?site=en_nhl HTTP/1.1" 200 None

21-01-23 07:31:56 DEBUG Starting new HTTP connection (1): statsapi.web.nhl.com:80

21-01-23 07:31:56 DEBUG http://statsapi.web.nhl.com:80 "GET /api/v1/schedule?date=2021-1-23&expand=schedule.linescore HTTP/1.1" 200 None

21-01-23 07:31:56 DEBUG Starting new HTTP connection (1): statsapi.web.nhl.com:80

21-01-23 07:31:56 DEBUG http://statsapi.web.nhl.com:80 "GET /api/v1/game/2020020078/feed/live?site=en_nhl HTTP/1.1" 200 None

21-01-23 07:32:02 DEBUG Starting new HTTP connection (1): statsapi.web.nhl.com:80

21-01-23 07:32:02 DEBUG http://statsapi.web.nhl.com:80 "GET /api/v1/schedule?date=2021-1-23&expand=schedule.linescore HTTP/1.1" 200 None

21-01-23 07:32:02 DEBUG Starting new HTTP connection (1): statsapi.web.nhl.com:80

21-01-23 07:32:02 DEBUG http://statsapi.web.nhl.com:80 "GET /api/v1/game/2020020078/feed/live?site=en_nhl HTTP/1.1" 200 None

21-01-23 07:32:09 DEBUG Starting new HTTP connection (1): statsapi.web.nhl.com:80

21-01-23 07:32:09 DEBUG http://statsapi.web.nhl.com:80 "GET /api/v1/schedule?date=2021-1-23&expand=schedule.linescore HTTP/1.1" 200 None

21-01-23 07:32:09 DEBUG Starting new HTTP connection (1): statsapi.web.nhl.com:80

21-01-23 07:32:09 DEBUG http://statsapi.web.nhl.com:80 "GET /api/v1/game/2020020078/feed/live?site=en_nhl HTTP/1.1" 200 None

21-01-23 07:32:15 DEBUG Starting new HTTP connection (1): statsapi.web.nhl.com:80

21-01-23 07:32:15 DEBUG http://statsapi.web.nhl.com:80 "GET /api/v1/schedule?date=2021-1-23&expand=schedule.linescore HTTP/1.1" 200 None

21-01-23 07:32:15 DEBUG Starting new HTTP connection (1): statsapi.web.nhl.com:80

21-01-23 07:32:15 DEBUG http://statsapi.web.nhl.com:80 "GET /api/v1/game/2020020078/feed/live?site=en_nhl HTTP/1.1" 200 None

21-01-23 07:32:22 DEBUG Starting new HTTP connection (1): statsapi.web.nhl.com:80

21-01-23 07:32:22 DEBUG http://statsapi.web.nhl.com:80 "GET /api/v1/schedule?date=2021-1-23&expand=schedule.linescore HTTP/1.1" 200 None

21-01-23 07:32:22 DEBUG Starting new HTTP connection (1): statsapi.web.nhl.com:80

21-01-23 07:32:22 DEBUG http://statsapi.web.nhl.com:80 "GET /api/v1/game/2020020078/feed/live?site=en_nhl HTTP/1.1" 200 None

21-01-23 07:32:28 DEBUG Starting new HTTP connection (1): statsapi.web.nhl.com:80

21-01-23 07:32:28 DEBUG http://statsapi.web.nhl.com:80 "GET /api/v1/schedule?date=2021-1-23&expand=schedule.linescore HTTP/1.1" 200 None

21-01-23 07:32:28 DEBUG Starting new HTTP connection (1): statsapi.web.nhl.com:80

21-01-23 07:32:28 DEBUG http://statsapi.web.nhl.com:80 "GET /api/v1/game/2020020078/feed/live?site=en_nhl HTTP/1.1" 200 None

21-01-23 07:32:35 DEBUG Starting new HTTP connection (1): statsapi.web.nhl.com:80

21-01-23 07:32:35 DEBUG http://statsapi.web.nhl.com:80 "GET /api/v1/schedule?date=2021-1-23&expand=schedule.linescore HTTP/1.1" 200 None

21-01-23 07:32:35 DEBUG Starting new HTTP connection (1): statsapi.web.nhl.com:80

21-01-23 07:32:35 DEBUG http://statsapi.web.nhl.com:80 "GET /api/v1/game/2020020078/feed/live?site=en_nhl HTTP/1.1" 200 None

21-01-23 07:32:41 DEBUG Starting new HTTP connection (1): statsapi.web.nhl.com:80

21-01-23 07:32:41 DEBUG http://statsapi.web.nhl.com:80 "GET /api/v1/schedule?date=2021-1-23&expand=schedule.linescore HTTP/1.1" 200 None

21-01-23 07:32:41 DEBUG Starting new HTTP connection (1): statsapi.web.nhl.com:80

21-01-23 07:32:41 DEBUG http://statsapi.web.nhl.com:80 "GET /api/v1/game/2020020078/feed/live?site=en_nhl HTTP/1.1" 200 None

21-01-23 07:32:48 DEBUG Starting new HTTP connection (1): statsapi.web.nhl.com:80

21-01-23 07:32:48 DEBUG http://statsapi.web.nhl.com:80 "GET /api/v1/schedule?date=2021-1-23&expand=schedule.linescore HTTP/1.1" 200 1503

21-01-23 07:32:48 DEBUG Starting new HTTP connection (1): statsapi.web.nhl.com:80

21-01-23 07:32:48 DEBUG http://statsapi.web.nhl.com:80 "GET /api/v1/game/2020020078/feed/live?site=en_nhl HTTP/1.1" 200 None

21-01-23 07:32:54 DEBUG Starting new HTTP connection (1): statsapi.web.nhl.com:80

21-01-23 07:32:54 DEBUG http://statsapi.web.nhl.com:80 "GET /api/v1/schedule?date=2021-1-23&expand=schedule.linescore HTTP/1.1" 200 1503

21-01-23 07:32:54 DEBUG Starting new HTTP connection (1): statsapi.web.nhl.com:80

21-01-23 07:32:54 DEBUG http://statsapi.web.nhl.com:80 "GET /api/v1/game/2020020078/feed/live?site=en_nhl HTTP/1.1" 200 None

rhazzed, 2021-01-23:
INFO: Validating config.json.....

INFO: config.json passes validation

No data for 20202021 Playoff

No default round for 20202021 Playoff.

21-01-23 09:13:13 INFO Scheduler started

21-01-23 09:13:13 DEBUG Looking for jobs to run

21-01-23 09:13:13 DEBUG Starting new HTTP connection (1): statsapi.web.nhl.com:80

21-01-23 09:13:13 DEBUG No jobs; waiting until a job is added

21-01-23 09:13:13 DEBUG http://statsapi.web.nhl.com:80 "GET /api/v1/game/2020020078/feed/live?site=en_nhl HTTP/1.1" 200 1346

Traceback (most recent call last):

  File "src/main.py", line 139, in <module>

    run()

  File "src/main.py", line 134, in run

    MainRenderer(matrix, data, sleepEvent).render()

  File "/home/pi/nhl-led-scoreboard/src/renderer/main.py", line 75, in render

    self.__render_game_day()

  File "/home/pi/nhl-led-scoreboard/src/renderer/main.py", line 186, in __render_game_day

    self.__render_pregame(self.scoreboard)

  File "/home/pi/nhl-led-scoreboard/src/renderer/main.py", line 212, in __render_pregame

    ScoreboardRenderer(self.data, self.matrix, scoreboard).render()

  File "/home/pi/nhl-led-scoreboard/src/renderer/scoreboard.py", line 24, in __init__

    'home'

  File "/home/pi/nhl-led-scoreboard/src/renderer/logos.py", line 19, in __init__

    gameLocation

  File "/home/pi/nhl-led-scoreboard/src/config/files/layout.py", line 46, in get_scoreboard_logo

    logo = self.logo_config.data.scoreboard.logos._default.__copy__()

AttributeError: 'JSONData' object has no attribute 'scoreboard'

riffnshred, 2021-01-23:
As discussed on discord, this is related to logos not being accessible because they are owned as root after they are generated. Which is the same issue as issue #187 . This will be fix next update (v1.5.6)

closed 2021-03-24 by riffnshred