riffnshred/nhl-led-scoreboard #189: V1 configuration web app

riffnshred, 2021-01-24:
**Description**
On issue #170 someone proposed to make a web app to take care of the configuration. 

Here are some key notes I currently have in mind about that. 

- This project evolves on a regular basis still, as it's in development. So the configurations are subject to change every now and then. This means this app needs to be able to check out the configuration of the current version installed. Luckily for our current nhl_setup app, we have config/config.schema.json that contains all the info and details of each option. 

- I'd prefer it does not interfere with the current core of the scoreboard software as I'm still working on it and some things move around to allow some features to be added. If you do plan to do so, please let me know, and let's discuss what's possible.

sflems, 2021-01-26:
Right now, I'm looking at a main dashboard that delivers today's games from NHLs stats API. Users will eventually be able to select a game to display from here. 

The config is imported as a JSONField in a Settings model. Currently working on displaying this JSON in a user-friendly way, such as a form, which can be submitted, validated, and then saved to the Pi.

My biggest issue at the moment is ensure proper (de)serialization of the models. Working on it.

riffnshred, 2021-01-26:
> Right now, I'm looking at a main dashboard that delivers today's games from NHLs stats API. Users will eventually be able to select a game to display from here.
> 

Thats where the app wont work for V1. If I remember correctly, that was the intension @joshkay had last year, but doing this require a major refactor of the code, hence the need for a V2. I recommand chatting with him about it, he tested a lot of things related to that. 

What Im looking is an easier way for users to config or reconfig their scoreboard. Some find the nhl_setup difficult to use because they dont like working in terminals. What I need is something they can run to reconfig, than stop  when they are done, and then they can run the scoreboard the usual way. All that to keep the PiZero from getting slower ( its already slow enough xD)

sflems, 2021-01-26:
The game selection is on the back burner, but my idea was that it would
simply update the config appropriately. I've got a pretty good idea as to
how it could possibly work on V1, but at the least I just wanted a simple
rudimentary games dashboard that we could build on.

As for the config its coming together. The config will have its own
Settings Page. Here we'll be able to edit the default config, and/or add
new profiles on a form dynamically  generated by the current config schema.
I've got that working, now it's just on to handling the config CRUD. Once
that's done, we'll simply be able to pick a valid profile from a list and
presto... Bob's your auntie's live lover.

joshkay, 2021-01-26:
Is this done using vanilla JS?  I had already had something setup for the client and it would be great to get some help.  Could combine our efforts hopefully.  

This is why I was asking if you were familiar with React.

https://github.com/joshkay/nhl-led-scoreboard-client

Keep in mind this hasn't been updated for quite some time.

sflems, 2021-01-26:
Not really, the config data is saved in a Django model as raw JSON and then built from there using jsonforms. The views.py will likely handle the calls themselves via POST form submissions(no need to AJAX them). This way the form JSON can be validated on the front and back end quite easily.

riffnshred, 2021-01-27:
Hey bud, I just want to clear up some things about this project and where I'm heading with it so you know what you are getting into.

At the moment V1 is heading into its last few feature updates and I won't allow anymore to be added unless it's an obvious need that helps to fix some issues, like your app which helps to configure the scoreboard in a more user-friendly way. That is because of the way I originally coded the project. It was never meant to become what it is and it is now difficult to properly add features without affecting a whole bunch of other stuff. I knew that right away when I officially release V1 a year ago. We did a lot of testing and figured that adding a complex app would just complicate the project even more.

Keep in mind that what you are making right now, there is a good chance it won't be ported to V2 because I already committed to working with @joshkay on that one and he knows the project and my plans, and how I work.

That being said, my plan is not to stop supporting V1 at all. I do love that it is possible to run this project on a Pi Zero, and I would love to optimize it to run more smoothly, at the cost of removing some features that will be on V2.

The app for V2 is currently only a plan and nothing surrounding it is solid, but Josh is looking for people to join him and give him a hand when the time to code the app comes. Feel free to get in touch if you are interested. 

Feel free to get in touch if you have any question. 

Cheers bud

sflems, 2021-01-27:
Heyo,

Thanks for the update. I'm feeling confident now that it will be able to modify and update the config in an easy manner. This should work for any version, and with a few small tweaks, as a standalone JSON config editor for similar apps. There _shouldn't_ be a need to directly port it in the end. 

Once I've got a rudimentary GUI working, I'll turn over to digging into the client @joshkay has built up. I still have to wrap my head around React a bit more it seems. 

Cheers!

sflems, 2021-02-10:
I'd like to finally submit a candidate as a possible solution. If anyone would like to take the interface for a spin you can do so here:

[NHL LED Scoreboard WebGUI](https://github.com/sflems/nhl-led-scoreboard-webgui)

Your feedback and questions are more than welcomed!