riffnshred/nhl-led-scoreboard #192: NWS Currently Down/Bad Gateway Error (502) and Scoreboard just keeps trying to connect on start

necromancyr, 2021-01-25:
**Describe the bug**
Upon update and restarting the scoreboard it's stopped from starting by the weather.gov site being down when a NWS alert check is done.  A 502 error is returned, but it keeps checking over and over instead of failing gracefully and moving to the next step.

Note - the weather.gov api is currently down/presenting a bad gateway error to everyone, not just a me thing.

**Logs**
INFO: Validating config.json.....
INFO: config.json passes validation
No data for 20202021 Playoff
No default round for 20202021 Playoff.
21-01-24 20:11:39 INFO Scheduler started
21-01-24 20:11:39 DEBUG Looking for jobs to run
21-01-24 20:11:39 DEBUG No jobs; waiting until a job is added
21-01-24 20:11:40 INFO Added job "owmWxWorker.getWeather" to job store "default"
21-01-24 20:11:40 DEBUG Looking for jobs to run
21-01-24 20:11:40 DEBUG Next wakeup is due at 2021-01-24 20:15:56.098778-05:00 (in 256.016133 seconds)
21-01-24 20:11:40 DEBUG Starting new HTTPS connection (1): api.openweathermap.org:443
21-01-24 20:11:40 DEBUG https://api.openweathermap.org:443 "GET /data/2.5/weather?APPID=[REMOVEDAPIKEY]&lang=en&                            lon=[REMOVED]&lat=[REMOVED] HTTP/1.1" 200 478
21-01-24 20:11:40 INFO Added job "nwsWxAlerts.getAlerts" to job store "default"
21-01-24 20:11:40 DEBUG Looking for jobs to run
21-01-24 20:11:40 DEBUG Next wakeup is due at 2021-01-24 20:15:56.098778-05:00 (in 255.758208 seconds)
21-01-24 20:11:40 DEBUG Starting new HTTPS connection (1): api.weather.gov:443
21-01-24 20:11:40 DEBUG https://api.weather.gov:443 "GET //alerts?active=1&point=[REMOVED LOCATION] HTTP/1.1" 502 15                            0
21-01-24 20:11:42 DEBUG Starting new HTTPS connection (1): api.weather.gov:443
21-01-24 20:11:42 DEBUG https://api.weather.gov:443 "GET //alerts?active=1&point=[REMOVED LOCATION] HTTP/1.1" 502 15                            0
21-01-24 20:11:45 DEBUG Starting new HTTPS connection (1): api.weather.gov:443
21-01-24 20:11:46 DEBUG https://api.weather.gov:443 "GET //alerts?active=1&point=[REMOVED LOCATION] HTTP/1.1" 502 15                            0
21-01-24 20:11:51 DEBUG Starting new HTTPS connection (1): api.weather.gov:443
21-01-24 20:11:51 DEBUG https://api.weather.gov:443 "GET //alerts?active=1&point=[REMOVED LOCATION] HTTP/1.1" 502 15                            0
21-01-24 20:11:59 DEBUG Starting new HTTPS connection (1): api.weather.gov:443
21-01-24 20:12:00 DEBUG https://api.weather.gov:443 "GET //alerts?active=1&point=[REMOVED LOCATION] HTTP/1.1" 502 15                            0
21-01-24 20:12:13 DEBUG Starting new HTTPS connection (1): api.weather.gov:443
21-01-24 20:12:13 DEBUG https://api.weather.gov:443 "GET //alerts?active=1&point={REMOVED LOCATION] HTTP/1.1" 502 150
21-01-24 20:12:35 DEBUG Starting new HTTPS connection (1): api.weather.gov:443
^C Exiting NHL-LED-SCOREBOARD

falkyre, 2021-01-25:
I've reached out to the developer of the python library I am using to see if he can point out how to force a maximum number of tries before having his code stop.  I don't see a straight forward way of doing it because the error handling code is wrapped under 4 or 5 layers.

falkyre, 2021-01-26:
There's a newer version of the noaa-sdk that was released in November 2020 that adds messages on the retries.  Still don't have a work on controlling the maximum number of retries so it doesn't get stuck in a loop.  However, if you want to, you can install the latest version of the library by doing `sudo pip3 install --upgrade noaa-sdk`

falkyre, 2022-06-25:
@riffnshred you can close this too. The python library I use doesn't handle this error so if it happens again, it'll just cause a crash.  I might replace this library with the open weather one once the weather alerts portion is officially released.

riffnshred, 2022-06-25:
roger that !! Thx !!

closed 2022-06-25 by riffnshred