riffnshred/nhl-led-scoreboard #198: NHL-scoreboard does not start up

saxonpaul92, 2021-02-03:
**Describe the bug**
I upgraded last night and now getting a file not found error....  All was working fine before upgrade...  Tried to upgrade again, with no change

**Logs**
pi@raspberrypi:~ $ sudo python3 /home/pi/nhl-led-scoreboard/src/main.py --led-gpio-mapping=adafruit-hat-pwm --led-brightness=40 --led-slowdown-gpio=0 --led-rows=32 --led-cols=64
Traceback (most recent call last):
  File "/home/pi/nhl-led-scoreboard/src/main.py", line 4, in <module>
    from renderer.main import MainRenderer
  File "/home/pi/nhl-led-scoreboard/src/renderer/main.py", line 5, in <module>
    from boards.boards import Boards
  File "/home/pi/nhl-led-scoreboard/src/boards/boards.py", line 7, in <module>
    from boards.scoreticker import Scoreticker
  File "/home/pi/nhl-led-scoreboard/src/boards/scoreticker.py", line 7, in <module>
    from renderer.scoreboard import ScoreboardRenderer
  File "/home/pi/nhl-led-scoreboard/src/renderer/scoreboard.py", line 3, in <module>
    from renderer.logos import LogoRenderer
  File "/home/pi/nhl-led-scoreboard/src/renderer/logos.py", line 10, in <module>
    uid = int(os.stat("./VERSION").st_uid)
FileNotFoundError: [Errno 2] No such file or directory: './VERSION'

**Your config file**
{
    "debug": true,
    "loglevel": "INFO",
    "live_mode": true,
    "preferences": {
        "time_format": "12h",
        "end_of_day": "02:00",
        "location": "Anna, Texas",
        "live_game_refresh_rate": 10,
        "teams": [
            "Stars"
        ],
        "sog_display_frequency": 4,
        "goal_animations": {}
    },
    "states": {
        "off_day": [
            "weather",
            "wxalert",
             "wxforecast",
            "scoreticker",
            "standings",
            "team_summary"
        ],
        "scheduled": [
            "weather",
            "wxalert",
            "wxforecast",
            "scoreticker",
            "standings",
            "team_summary"
        ],
        "intermission": [
            "scoreticker"
        ],
        "post_game": [
            "scoreticker",
            "standings",
            "team_summary",
            "weather",
            "wxalert",
            "wxforecast"
        ]
    },
    "boards": {
        "scoreticker": {
            "preferred_teams_only": false,
            "rotation_rate": 5
        },
        "seriesticker": {
            "preferred_teams_only": true,
            "rotation_rate": 5
        },
        "standings": {
            "preferred_standings_only": true,
            "standing_type": "division",
            "divisions": "central",
            "conference": "western"
        },
        "clock": {
            "duration": 15,
            "hide_indicator": false,
            "preferred_team_colors": true,
            "clock_rgb": "",
            "date_rgb": "",
            "flash_seconds": true
        },
        "covid19": {
            "worldwide_enabled": false,
            "country_enabled": false,
            "country": [
                "USA"
            ],
            "us_state_enabled": false,
            "us_state": [
                "New York"
            ],
            "canada_enabled": false,
            "canada_prov": [
            "canada_prov": [
                "Quebec"
            ]
        },
        "weather": {
            "enabled": true,
            "view": "full",
            "units": "imperial",
            "duration": 30,
            "data_feed": "OWM",
            "owm_apikey": "1661ce0d759474413cab54a3be5ac47b",
            "update_freq": 5,
            "show_on_clock": true,
            "forecast_enabled": true,
            "forecast_days": 1,
            "forecast_update": 1
        },
        "wxalert": {
            "alert_feed": "NWS",
            "update_freq": 5,
            "show_alerts": true,
            "nws_show_expire": true,
            "alert_title": true,
            "scroll_alert": true,
            "alert_duration": 5,
            "show_on_clock": true
        }
    },
    "sbio": {
        "screensaver": {
            "enabled": true,
            "animations": true,
            "start": "03:00",
            "stop": "09:00",
            "data_updates": false,
            "motionsensor": false,
            "pin": 24,
            "delay": 30
        },
        "dimmer": {
            "enabled": false,
            "source": "software",
            "daytime": "",
            "nighttime": "",
            "offset": 90,
            "frequency": 5,
            "light_level_lux": 400,
            "mode": "always",
            "sunset_brightness": 5,
            "sunrise_brightness": 40
        },
        "pushbutton": {
            "enabled": false,
            "bonnet": false,
            "pin": 25,
            "reboot_duration": 2,
            "reboot_override_process": "",
            "display_reboot": true,
            "poweroff_duration": 10,
            "poweroff_override_process": "",
            "display_halt": true,
            "state_triggered1": "weather",
            "state_triggered1_process": ""
        }
    }
}

**Hardware you are using**
- Raspberry pi 2
- The Matrix driver (Adafruit Bonnet)
- Anti-flicker mod = yes

**Software**
- Latest build as of last night (2/1/2021)

riffnshred, 2021-02-03:
Hey bud. It sounds like you updated from a few versions ago, is
that correct?. There as been many changes since last summer. Try the update
process one more time and if that doesn't work, delete the repo and
reinstall it. That will definitely work.

saxonpaul92, 2021-02-03:
Actually I updated a little over a week ago and had the newer team logo's (very nice) and to pick up new features.

I upgraded last night just to make sure I got all the latest changes.

I will blow out the system and re-install from scratch.  Thought there might have been an easy fix.

Thanks for the info.

saxonpaul92, 2021-02-03:
I blew out the installation and reinstalled from scratch and go the same issue.

Any ideas?

saxonpaul92, 2021-02-03:
I found the problem....

You MUST execute the main.py script from within the /home/pi/nhl-led-scoreboard directory.

You CANNOT execute from /home/pi or it will not find the directory callouts in the files.

The instructions do have to enter the nhl-led-scoreboard directory but, you may want to add a note...

Thanks!!

closed 2021-02-03 by saxonpaul92

riffnshred, 2021-02-03:
Thanks, Im in the process of rewriting the whole documentation I will keep that in mind. Cheers