riffnshred/nhl-led-scoreboard #203: Daisy chained 32x64 led panels?

rgrech35, 2021-02-13:
**Describe the bug**
I'm not really sure if this is a bug or how the code works. I have two adafruit 64x32 panels daisy chained together. Running in default mode, the scoreboard live update is mirrored on each panel. I've tried setting cols and rows to 64 and played around with the panel chain setting as well, but it doesn't seem like the scoreboard will render in a "full screen" mode - which is fine - i just wasn't sure if I'm doing something wrong or not.

**Hardware you are using**
- Pi 3
- Adafruit HAT
- no

**Software**
- Latest - just installed

falkyre, 2021-02-13:
This works for 2 panels to give you a 128x32 display
sudo python3 ./src/main.py --led-gpio-mapping=adafruit-hat-pwm --led-slowdown-gpio=2 --led-rows=32 --led-cols=64 --led-chain=2

rgrech35, 2021-02-13:
Thank you - what I was trying to do was to chain to a 64x64 layout but
after noodling around with all the options it doesn't seem possible? Unless
I am missing something.

riffnshred, 2021-02-13:
verify what type of panels you got. 64x64 panels and bigger panels, in general, have different pin layout or configuration and the most common one I see is that they have an extra E line that you need to solder. 

If you have a bonnet, look underneath it and you will see 3 small pads in a row with labels 16,E,8 . Solder the center to whichever pin is your E line connected to. If you have the HAT, those pads are on the top. 

If you do that soldering, make sure you set the dimmer stuff to false as it uses the same pin.

rgrech35, 2021-02-13:
Thanks for the reply Joel. I think you are misunderstanding the issue. To
clarify, I do not have a single 64x64 panel.
I have two adafruit 62x32 panels.  I am trying to daisy-chain the panels
into a 64x64 display, but when I start any of the scripts using
--led-rows=32 --led-cols=64 --led-chain=2 it recognizes that as a 128x32
panel. I'm not having a display issue, the actual issue is
rpi-rgb-led-matrix recognizing as a 128x32 and not a 64x64, so I am trying
to figure out if i'm missing a parameter somewhere

riffnshred, 2021-02-13:
OH !!! Sorry, I didn't finish my coffee yet hah. check that section in the rgb matrix library documentation.

https://github.com/hzeller/rpi-rgb-led-matrix#panel-arrangement. 

Keep in mind that I don't support that layout yet and I'm not sure I will support 64x64 yet. the plan is 128x64 and 128x32 for now.

closed 2021-02-19 by riffnshred