riffnshred/nhl-led-scoreboard #207: Concantenate Error   -  BETA branch

rhazzed, 2021-02-17:
tail /var/log/nhl/nhl.err.log 
    size = self.get_size()
  File "/dev/shm/nhl-led-scoreboard/src/renderer/logos.py", line 37, in get_size
    int(round_normal(self.matrix.width * self.layout.zoom)), 
  File "/dev/shm/nhl-led-scoreboard/src/utils.py", line 202, in round_normal
    value = math.floor(n * multiplier + 0.5) / multiplier
TypeError: can only concatenate str (not "float") to str
INFO: Validating config.json.....
INFO: config.json passes validation
No data for 20202021 Playoff
No default round for 20202021 Playoff.

The board will only display clock and shows 1 game scheduled (despite there are 5 games today)
It shows Panthers vs Hurricanes and then it cycles back to the clock.   I have not made ANY changes and 
I only have Blues as favorite.  Perhaps it having an issue rendering one of the missing teams teams logos?

it's missing Senators/Maple Leafs, Blackhawks/Red Wings, Canucks/Flames, and the Jets/Oilers.

riffnshred, 2021-02-18:
Have you changed the position of your logos recently?

rhazzed, 2021-02-18:
perhaps a month ago.  do you think may have an invalid value?

~ Eric

riffnshred, 2021-02-18:
can you send me what you have for the layout of your logos using https://pastebin.pl/ ?

rhazzed, 2021-02-18:
Well... silly me.    TOR  was missing the "%" and its working now.

~ Eric

riffnshred, 2021-02-18:
bingoooooo hahah

rhazzed, 2021-02-18:
It's working now.   missing a % for the zoom level for TOR

~ Eric

riffnshred, 2021-02-18:
Sweet haha, I though it was somehting like that. Cheers bud

closed 2021-02-19 by riffnshred