riffnshred/nhl-led-scoreboard #209: OSError: cannot identify image file '/home/pi/nhl-led-scoreboard/assets/logos/BUF/light/64x32.png'

seanscal, 2021-02-20:
I think there is something corrupted about the BUF image. it is preventing my scoreboard from starting up at all. 

```bash
 sudo python3 src/main.py --led-gpio-mapping=adafruit-hat --led-brightness=60 --led-slowdown-gpio=2
INFO: Validating config.json.....
INFO: config.json passes validation
No data for 20202021 Playoff
No default round for 20202021 Playoff.
2021-02-20 12:56:50 INFO NHL-LED-SCOREBOARD - v1.5.7 (64x32)
2021-02-20 12:56:50 INFO Screen saver will start @ 22:00 and end @ 22:05
2021-02-20 12:56:50 INFO Rendering...
2021-02-20 12:56:50 INFO Live mode is off. Going through the boards
2021-02-20 12:56:50 INFO PING !!! Render off day
2021-02-20 12:56:50 INFO Checking for new day
2021-02-20 12:56:50 INFO It is not a new day
Traceback (most recent call last):
  File "src/main.py", line 139, in <module>
    run()
  File "src/main.py", line 134, in run
    MainRenderer(matrix, data, sleepEvent).render()
  File "/home/pi/nhl-led-scoreboard/src/renderer/main.py", line 65, in render
    self.__render_offday()
  File "/home/pi/nhl-led-scoreboard/src/renderer/main.py", line 92, in __render_offday
    self.boards._off_day(self.data, self.matrix,self.sleepEvent)
  File "/home/pi/nhl-led-scoreboard/src/boards/boards.py", line 81, in _off_day
    board(data, matrix,sleepEvent)
  File "/home/pi/nhl-led-scoreboard/src/boards/boards.py", line 222, in scoreticker
    Scoreticker(data, matrix, sleepEvent).render()
  File "/home/pi/nhl-led-scoreboard/src/boards/scoreticker.py", line 31, in render
    ScoreboardRenderer(self.data, self.matrix, Scoreboard(self.games[self.index], self.data)).render()
  File "/home/pi/nhl-led-scoreboard/src/renderer/scoreboard.py", line 32, in __init__
    'away'
  File "/home/pi/nhl-led-scoreboard/src/renderer/logos.py", line 29, in __init__
    self.load(team_abbrev)
  File "/home/pi/nhl-led-scoreboard/src/renderer/logos.py", line 47, in load
    self.logo = Image.open(filename)
  File "/usr/lib/python3/dist-packages/PIL/Image.py", line 2687, in open
    % (filename if filename else fp))
OSError: cannot identify image file '/home/pi/nhl-led-scoreboard/assets/logos/BUF/light/64x32.png'

```

seanscal, 2021-02-20:
i deleted + reinstalled everything and that fixed it. I don't know what happened here, I want to give this board as a gift though so just normal running/restarting the script without something like this happening would be nice

riffnshred, 2021-02-20:
You might have used sudo somewhere in the installation ?  Just a heads up, this is not the type of project I recommend to give to someone as its something that will need maintenance overtime. Be prepared to give support and troubleshoot the scoreboard for them once in awhile. I made this project for educational purposes only.

closed 2021-02-20 by riffnshred