riffnshred/nhl-led-scoreboard #214: ERROR WHILE RENDERING: 'Division' object has no attribute 'pacific'

ravedoglv, 2021-02-26:
**Describe the bug**
When running tail, I get an error about my division

**Logs**

2021-02-25 18:39:04 INFO Refreshing data in a minute
2021-02-25 18:39:24 INFO Rendering...
2021-02-25 18:39:24 INFO Game Day Wooooo
2021-02-25 18:39:24 INFO Showing Game
2021-02-25 18:39:25 INFO Checking for new day
2021-02-25 18:39:25 INFO It is not a new day
2021-02-25 18:39:25 INFO Game is irregular
2021-02-25 18:39:25 INFO Showing Irregular
2021-02-25 18:40:33 INFO ERROR WHILE RENDERING: 'Division' object has no attribute 'pacific'
2021-02-25 18:40:33 INFO Refreshing data in a minute
2021-02-25 18:40:54 INFO Rendering...
2021-02-25 18:40:55 INFO Game Day Wooooo
2021-02-25 18:40:55 INFO Showing Game
2021-02-25 18:40:55 INFO Checking for new day
2021-02-25 18:40:55 INFO It is not a new day
2021-02-25 18:40:55 INFO Game is irregular
2021-02-25 18:40:55 INFO Showing Irregular
2021-02-25 18:42:53 INFO NHL-LED-SCOREBOARD - v1.5.7 (64x32)

**Your config file**

{
    "debug": false,
    "loglevel": "INFO",
    "live_mode": true,
    "preferences": {
        "time_format": "12h",
        "end_of_day": "12:00",
        "location": " xxxx",
        "live_game_refresh_rate": 10,
        "teams": [
            "Golden Knights"
        ],
        "sog_display_frequency": 4,
        "goal_animations": {
            "pref_team_only": true
        }
    },
    "states": {
        "off_day": [
            "scoreticker",
            "team_summary",
            "standings",
            "clock",
            "covid_19",
            "weather"
        ],
        "scheduled": [
            "team_summary",
            "standings",
            "scoreticker",
            "clock",
            "covid_19",
            "weather"
        ],
        "intermission": [
            "scoreticker",
            "team_summary"
        ],
        "post_game": [
            "team_summary",
            "standings",
            "scoreticker",
            "clock"
        ]
    },
    "boards": {
        "scoreticker": {
            "preferred_teams_only": true,
            "rotation_rate": 5
        },
        "seriesticker": {
            "preferred_teams_only": true,
            "rotation_rate": 5
        },
        "standings": {
            "preferred_standings_only": true,
            "standing_type": "wild_card",
            "divisions": "pacific",
            "conference": "western"
        },
        "clock": {
            "duration": 15,
            "hide_indicator": false,
            "preferred_team_colors": true,
            "clock_rgb": "",
            "date_rgb": "",
            "flash_seconds": true
        },
        "covid19": {
            "worldwide_enabled": true,
            "country_enabled": true,
            "country": [
                "USA"
            ],
            "us_state_enabled": true,
            "us_state": [
                "Nevada"
            ],
            "canada_enabled": false,
            "canada_prov": [
                "Quebec"
            ]
        },
        "weather": {
            "enabled": true,
            "view": "full",
            "units": "imperial",
            "duration": 60,
            "data_feed": "OWM",
            "owm_apikey": "xxxxx",
            "update_freq": 5,
            "show_on_clock": true,
            "forecast_enabled": true,
            "forecast_days": 3,
            "forecast_update": 1
        },
        "wxalert": {
            "alert_feed": "NWS",
            "update_freq": 5,
            "show_alerts": true,
            "nws_show_expire": false,
            "alert_title": true,
            "scroll_alert": true,
            "alert_duration": 5,
            "show_on_clock": true
        }
    },
    "sbio": {
        "screensaver": {
            "enabled": true,
            "animations": true,
            "start": "22:00",
            "stop": "22:05",
            "data_updates": false,
            "motionsensor": true,
            "pin": 24,
            "delay": 30
        },
        "dimmer": {
            "enabled": true,
            "source": "software",
            "daytime": "",
            "nighttime": "",
            "offset": 90,
            "frequency": 5,
            "light_level_lux": 400,
            "mode": "always",
            "sunset_brightness": 40,
            "sunrise_brightness": 80
        },
        "pushbutton": {
            "enabled": false,
            "bonnet": false,
            "pin": 25,
            "reboot_duration": 2,
            "reboot_override_process": "",
            "display_reboot": true,
            "poweroff_duration": 10,
            "poweroff_override_process": "",
            "display_halt": true,
            "state_triggered1": "weather",
            "state_triggered1_process": ""
        }
    }
}

**Hardware you are using**
- raspberry pi zero
- The Matrix driver (Adafruit Bonnet
- Anti-flicker mod or not? Yes

**Software**
1.5.7

ravedoglv, 2021-02-26:
If I change it from Pacific to West it works but shouldn’t it accept “Pacific?”

falkyre, 2021-02-26:
Not for this season.  Divisions are north, east, west and central.

ravedoglv, 2021-02-26:
And there it is! Thank you so much!

closed 2021-02-26 by ravedoglv