riffnshred/nhl-led-scoreboard #215: Weather icons missing

ravedoglv, 2021-02-26:
**Describe the bug**
I’m not getting any icons in wx conditions

**Hardware you are using**
- Raspberry pi zero w
- The Matrix driver Adafruit Bonnet
- Anti-flicker mod or not? Yes

**Software**
1.5.7

[image]

falkyre, 2021-02-26:
Looks like the fonts are corrupted. In the asset/fonts directory there are two weathericons files.  What are the sizes for them?

ravedoglv, 2021-02-26:
.ttf 100k
.otf 54k

falkyre, 2021-02-26:
Sizes are correct.

In your log file, you should see something like the following:

`2021-02-26 14:31:03 INFO ['02/26 02:31 PM', '\uf01b', 'Light Snow', '-0.8C', '-3.3C', '91%', '-2.1C']
2021-02-26 14:31:03 INFO ['7.0 kph', 'WNW', '\uf05d', '0.0 kph', '999.0 hPa', '\uf057', '1.0 km']
`
The first line, the second field after the time that's the icon code.  What are you seeing?  And is it showing the 3 squares everytime? or just sometimes?

ravedoglv, 2021-02-28:
Sorry for delay... the unit is at my dads. 

2021-02-27 19:14:39 INFO ['02/27 07:14 PM', '�\x80\x82', 'clear sky', '50.0F', '41.8F', '12%', '12.8F']
2021-02-27 19:14:39 INFO ['3.4mph', 'S', '\uf05c', '0mph', '1016 MB', '\uf07b', '6.2 mi']

That’s the output. And it’s consistent. 

So question!: when setting up the pi, in raspi-config, does the localization have anything to do with this?

falkyre, 2021-02-28:
Yah localization needs to be utf8 so an english utf8 would be needed for proper display.

ravedoglv, 2021-03-01:
Woohoo. That was it! Bad localization. Thank you so much for this!

closed 2021-03-01 by ravedoglv