riffnshred/nhl-led-scoreboard #224: after a week of working . now all i get is the clock

may2700, 2021-03-03:
**Describe the bug**
board was working fine then all of a sudden, the only thing that displays is the clock. I tried restarting the board in supervisor but same outcome

**Logs**
2021-03-03 14:21:29 INFO Refreshing data in a minute
2021-03-03 14:21:39 INFO Rendering...
2021-03-03 14:21:39 INFO Your preferred teams are Off today
2021-03-03 14:21:39 INFO PING !!! Render off day
2021-03-03 14:21:39 INFO Checking for new day
2021-03-03 14:21:39 INFO It is not a new day
2021-03-03 14:21:39 INFO ERROR WHILE RENDERING: 'Object' object has no attribute 'intermissionInfo'

**Your config file**
A screen shot of your config.json file can be really useful to help you. 

**Hardware you are using**
- Raspberry pi 3+
- The Matrix driver - Adafruit Bonnet
- Anti-flicker mod - yes

**Software**
- 1.5.7

tsmooth3, 2021-03-03:
seeing the same starting around 1:50PM EST
Thought it might have something to do with SSL certs so performed an apt-get update/upgrade (then noticed the api calls use http not https)
Thought it might be issues with talking to the API but am able to query via curl (in screenshot)
[image]

tsmooth3, 2021-03-03:
issue seems to be with the 'linescore' node of the live data feed and issue was intermittent for about 2 hours .. seems to be stable again as of 3:45p EST.    

using the following to pull scheduled game stats for today's Caps v Bruins game:
`curl --location --request GET 'http://statsapi.web.nhl.com/api/v1/game/2020020359/feed/live?site=en_nhl'`

when it wasn't working linescore didn't have an "intermissionInfo" child node:
[image]

when it was working "intermissionInfo" is present:
[image]

riffnshred, 2021-03-04:
this is also related to #225. Like I said overthere, and like you point out, the api had part of the information missing. I will handle that error in the next update to prevent this in the future.

closed 2021-03-05 by riffnshred