riffnshred/nhl-led-scoreboard #225: Scoreboard stuck in loop - flashing "loading" image v1.6.0

ramsaytc, 2021-03-03:
**Describe the bug**
Scoreboard is stuck in a loop, flashing the "loading" image on the LED board.  After rebooting the Pi, the issue persists.

**Logs**
pastebin logs: [https://pastebin.com/Lttjzs7M](https://pastebin.com/Lttjzs7M)

**Your config file**
pastebin of config: [https://pastebin.com/tyAJSTCx](https://pastebin.com/tyAJSTCx)

**Hardware you are using**
- Pi model 3B
- Adafruit HAT
- No anti-flicker mod 

**Software**
- latest v1.6.0, using NHL LED Scoreboard Raspberry Pi Image by @falkyre

cipriani14, 2021-03-03:
Seeing the same exact thing.

— Jason Cipriani
Freelance Writer
@MrCippy
JasonCipriani.com
Signal: 719.679.2770

<https://github.com/riffnshred/nhl-led-scoreboard/issues/225>, or
unsubscribe

mrenigma03, 2021-03-03:
Also seeing the same problem.

Tried running it from console (rather than via supervisor or alternative) and after a couple minutes it errors out with:

Traceback (most recent call last):
  File "src/main.py", line 157, in <module>
    run()
  File "src/main.py", line 152, in run
    MainRenderer(matrix, data, sleepEvent).render()
  File "/home/pi/nhl-led-scoreboard/src/renderer/main.py", line 70, in render
    self.__render_offday()
  File "/home/pi/nhl-led-scoreboard/src/renderer/main.py", line 86, in __render_offday
    self.boards._off_day(self.data, self.matrix,self.sleepEvent)
  File "/home/pi/nhl-led-scoreboard/src/boards/boards.py", line 81, in _off_day
    board(data, matrix,sleepEvent)
  File "/home/pi/nhl-led-scoreboard/src/boards/boards.py", line 236, in team_summary
    TeamSummary(data, matrix, sleepEvent).render()
  File "/home/pi/nhl-led-scoreboard/src/boards/team_summary.py", line 74, in render
    next_game_scoreboard = Scoreboard(nhl_api.overview(next_game_id), self.data)
  File "/home/pi/nhl-led-scoreboard/src/data/scoreboard.py", line 158, in __init__
    self.periods = Periods(overview)
  File "/home/pi/nhl-led-scoreboard/src/data/periods.py", line 10, in __init__
    intermission_info = period_info.intermissionInfo
AttributeError: 'Object' object has no attribute 'intermissionInfo'

ramsaytc, 2021-03-03:
Issue with API has corrected itself.  This issue can be closed.

closed 2021-03-03 by ramsaytc

riffnshred, 2021-03-03:
I will handle that error for the future in case it comes back. thanks for reporting 8)