riffnshred/nhl-led-scoreboard #226: Object Has No Attribute

kempdawg20, 2021-03-03:
Ran the update and now I'm getting the following message on the terminal:
Traceback (most recent call last):
  File "src/main.py", line 157, in <module>
    run()
  File "src/main.py", line 152, in run
    MainRenderer(matrix, data, sleepEvent).render()
  File "/home/pi/nhl-led-scoreboard/src/renderer/main.py", line 76, in render
    self.__render_game_day()
  File "/home/pi/nhl-led-scoreboard/src/renderer/main.py", line 92, in __render_game_day
    self.scoreboard = Scoreboard(self.data.overview, self.data)
  File "/home/pi/nhl-led-scoreboard/src/data/scoreboard.py", line 158, in __init__
    self.periods = Periods(overview)
  File "/home/pi/nhl-led-scoreboard/src/data/periods.py", line 10, in __init__
    intermission_info = period_info.intermissionInfo
AttributeError: 'Object' object has no attribute 'intermissionInfo'

ramsaytc, 2021-03-03:
this is a dupe of #225

riffnshred, 2021-03-03:
closing because its a duplicate. thanks bud !

closed 2021-03-03 by riffnshred