riffnshred/nhl-led-scoreboard #227: Empty Location in config causing loop

sflems, 2021-03-03:
**Describe the bug**
When a location, or string, is missing preferences.location in config.json, board enters a restart loop.

**Logs**
```
INFO: Validating config.json.....
INFO: config.json passes validation
Status code 429 from http://ipinfo.io/json: ERROR - 429 Client Error: Too Many Requests for url: http://ipinfo.io/json
Traceback (most recent call last):
  File "./src/main.py", line 157, in <module>
    run()
  File "./src/main.py", line 61, in run
    data = Data(config)
  File "/home/pi/nhl-led-scoreboard/src/data/data.py", line 87, in __init__
    self.latlng, self.latlng_msg = get_lat_lng(config.location)
  File "/home/pi/nhl-led-scoreboard/src/utils.py", line 40, in get_lat_lng
    message = "location is: " + g.city + ","+ g.country + " " + str(g.latlng)
TypeError: can only concatenate str (not "NoneType") to str
```

**Your config file**
```
{
    "debug": false,
    "loglevel": "INFO",
    "live_mode": true,
    "preferences": {
        "time_format": "12h",
        "end_of_day": "00:00",
        "location": "",
        "live_game_refresh_rate": 10,
        "teams": [
            "Maple Leafs",
            "Golden Knights",
            "Canucks"
        ],
        "sog_display_frequency": 4,
(Contd.)
``` 

**Hardware you are using**
- Raspberry Pi 4b 2GB
- Adafruit Bonnet)
- Anti-flicker mod? YES

**Software**
- v1.6.0

Did we max out the IP requests with the new update? Hehe

sflems, 2021-03-03:
[image]

falkyre, 2021-03-03:
No, it is blocking you from doing the look up based on your IP address, not everbody's IP address.  The location being empty has been there since before V1, but if you were having the same issue with everyone else and your board being restarted every time because of the nhl api, that's why you are being blocked.  The code I have currently doesn't handle an error code coming back from the IP location lookup and because it's a 429 error, it's not an issue with the code or the fact the location is empty, it's an issue that everytime the code restarts, the location is sending a request.  I'll add that in to gracefully handle an error or figure out a way to cache a good lookup and not do a lookup every start.  @riffnshred I'll look into this over the next few days.  It's not a show stopper but a side effect f the interns at NHL Stats website day drinking again

sflems, 2021-03-03:
When ISNT it those damn interns?

 Thanks for the specifics on the issue, Sean

>

riffnshred, 2021-03-03:
I guess they also had a party for this projects anniversary xD

falkyre, 2021-03-08:
#240 Submitted for limiting calls to ipinfo.io by creating a cache file on first successful lookup for either IP address (ie location is blank) or if you enter something in the location.  The cache file will get refreshed if older than 7 days old (on a restart of the scoreboard), there's no automated refresh

jrakestraw, 2021-03-11:
I think this goes along with this. After the update, if there is already a location.json in nhl-led-scoreboard/config, then when you try to run the program you get there error below. As soon as I remove location.json, it runs smooth.

pi@raspberrypi:~/nhl-led-scoreboard $ sudo python3 src/main.py --led-gpio-mapping=adafruit-hat --led-brightness=60 --led-slowdown-gpio=2
INFO: Validating config.json.....
INFO: config.json passes validation
Traceback (most recent call last):
  File "src/main.py", line 157, in <module>
    run()
  File "src/main.py", line 61, in run
    data = Data(config)
  File "/home/pi/nhl-led-scoreboard/src/data/data.py", line 86, in __init__
    self.latlng, self.latlng_msg = get_lat_lng(config.location)
  File "/home/pi/nhl-led-scoreboard/src/utils.py", line 50, in get_lat_lng
    message = "location loaded from cache (saved {} days ago): ".format(filetime.days) + j["city"] + ", "+ j["country"] + " " + str(latlng)
KeyError: 'city'

falkyre, 2021-03-12:
What's you location.json?

jrakestraw, 2021-03-12:
> What's you location.json?

This is what's in my location.json. 

pi@raspberrypi:~/nhl-led-scoreboard/config $ more location.json
{
    "accuracy": 0.6112048267421166,
    "address": "Hilliard, Norwich Township, Franklin County, Ohio, 43026, United States",
    "bbox": {
        "northeast": [
            40.0691255,
            -83.0971951
        ],
        "southwest": [
            39.9995883,
            -83.2060639
        ]
    },
    "confidence": 4,
    "country": "United States",
    "country_code": "us",
    "county": "Franklin County",
    "icon": "https://nominatim.openstreetmap.org/ui/mapicons//poi_boundary_administrative.p.20.png",
    "importance": 0.6112048267421166,
    "lat": 40.033814,
    "lng": -83.1596108,
    "municipality": "Norwich Township",
    "ok": true,
    "osm_id": 182503,
    "osm_type": "relation",
    "place_id": 257862088,
    "place_rank": 16,
    "postal": "43026",
    "quality": "administrative",
    "raw": {
        "place_id": 257862088,
        "licence": "Data \u00a9 OpenStreetMap contributors, ODbL 1.0. https://osm.org/copyright",
        "osm_type": "relation",
        "osm_id": 182503,
        "boundingbox": [
            "39.9995883",
            "40.0691255",
            "-83.2060639",
            "-83.0971951"
        ],
        "lat": "40.033814",
        "lon": "-83.1596108",
        "display_name": "Hilliard, Norwich Township, Franklin County, Ohio, 43026, United States",
        "place_rank": 16,
        "category": "boundary",
        "type": "administrative",
        "importance": 0.6112048267421166,
        "icon": "https://nominatim.openstreetmap.org/ui/mapicons//poi_boundary_administrative.p.20.png",
        "address": {
            "town": "Hilliard",
            "municipality": "Norwich Township",
            "county": "Franklin County",
            "state": "Ohio",
            "postcode": "43026",
            "country": "United States",
            "country_code": "us"
        }
    },
    "region": "Ohio",
    "state": "Ohio",
    "status": "OK",
    "town": "Hilliard",
    "type": "administrative"
}

falkyre, 2021-03-12:
Do you have something in the location field in config.json?

xTrackstar04x, 2021-03-12:
Im running into the same issue

```pi@raspberrypi:~/nhl-led-scoreboard $ python3 src/main.py --led-gpio-mapping=adafruit-hat --led-brightness=40 --led-slowdown-gpio=2
FYI: not running as root which means we can't properly control timing unless this is a real-time kernel. Expect color degradation. Consider running as root with sudo.
Can't set realtime thread priority=99: Operation not permitted.
	You are probably not running as root ?
	This will seriously mess with color stability and flicker
	of the matrix. Please run as `root` (e.g. by invoking this
	program with `sudo`), or setting the capability on this
	binary by calling
	sudo setcap 'cap_sys_nice=eip' /usr/bin/python3.7
INFO: Validating config.json.....
INFO: config.json passes validation
Traceback (most recent call last):
  File "src/main.py", line 157, in <module>
    run()
  File "src/main.py", line 61, in run
    data = Data(config)
  File "/home/pi/nhl-led-scoreboard/src/data/data.py", line 86, in __init__
    self.latlng, self.latlng_msg = get_lat_lng(config.location)
  File "/home/pi/nhl-led-scoreboard/src/utils.py", line 50, in get_lat_lng
    message = "location loaded from cache (saved {} days ago): ".format(filetime.days) + j["city"] + ", "+ j["country"] + " " + str(latlng)
KeyError: 'city'
```

Beginning of Config: 
```{
    "debug": true,
    "loglevel": "INFO",
    "live_mode": true,
    "preferences": {
        "time_format": "12h",
        "end_of_day": "8:00",
        "location": "Chicago, Illinois",
        "live_game_refresh_rate": 10,
        "teams": [
            "Golden Knights"
        ],
        "sog_display_frequency": 4,
        "goal_animations": {
            "pref_team_only": true
        }
    },
    "states": {
        "off_day": [
            "scoreticker"
        ],
        "scheduled": [
            "scoreticker"
        ],
        "intermission": [
            "scoreticker",
            "seriesticker"
        ],
        "post_game": [
            "scoreticker",
            "seriesticker"
        ]
    },

```

falkyre, 2021-03-12:
Do you have a location set in your config.json?

falkyre, 2021-03-12:
Remove your location from the config.  Also delete the location.json from the config folder.  It's owned by root so you'll have to run sudo to delete the file.

xTrackstar04x, 2021-03-12:
That worked!

jrakestraw, 2021-03-12:
> Do you have something in the location field in config.json?

I do. I quickly found the issue. I don't know if for others if you want to put in a quick rm or mv of the location.json in the run script for anyone who may have something in the location field of the config.json. I only started noticing the issue after 1.16.1 came out.

falkyre, 2021-03-12:
Yah, I put code in that would cache the location due to some people being blocked on the IP address lookup so we wouldn't hammer the service that was providing that.  The issue is whatever you put in your location entry in the config.json comes back from Open Street Maps as something other than a city (likely you are being shown as a town) so the code was looking for city.  I have a fix in place and will be issuing another PR to fix this.

riffnshred, 2021-03-15:
I belive this was fixed in the latest update v1.6.2.

Closing

closed 2021-03-15 by riffnshred