riffnshred/nhl-led-scoreboard #228: Not overwriting files when trying to manually update

xTrackstar04x, 2021-03-04:
When I am trying to update the scoreboard manually by using 

```git reset --hard
git checkout master
git pull
chmod +x scripts/install.sh
./scripts/install.sh
```

After I use ```git pull``` I get the following error 
```Updating 194a12f..60eb1eb
error: The following untracked working tree files would be overwritten by merge:
	apt-requirements
	assets/animations/goal/goal_light_animation.gif
	assets/animations/penalty/general/.gitignore
	assets/animations/penalty/opposing/.gitignore
	assets/animations/penalty/penalty_animation.gif
	assets/animations/penalty/preferred/.gitignore
	assets/fonts/VGA.ttf
	assets/images/128x64_scoreboard_center_gradient.png
	assets/images/raspberrypi_OS_lite.png
	assets/loading/loading.png
Please move or remove them before you merge.
Aborting
```

**Hardware you are using**
- Raspberry pi model: 3B+
-Adafruit HAT
- Anti-flicker mod: No

**Software**
- Between 1.5.7 and 1.6

xTrackstar04x, 2021-03-04:
I deleted NHL-led-scoreboard and reinstalled and it looks like everything works now. Still going to keep it open just incase there is something minor that can be done in order to overwrite the files instead of uninstall/reinstall

falkyre, 2021-03-04:
So I started with a fresh brand new install of v1.5.7.  I did all of your steps above and no issues.  Were you doing anything in the directory (playing with animations, gradients etc), tried out beta prior to trying to upgrade?

xTrackstar04x, 2021-03-04:
Nope. Only thing I have changed was the Config.json.

sflems, 2021-03-04:
> Nope. Only thing I have changed was the Config.json.
> 
> I added that .sh file to the root, so that should not effect it. I also have the Webgui set up. idk if that messes with them at all

Just a side note, it doesn't look like anything there is related to the GUI, it only updates the config.json and creates the customprofile.json.bak files if backed up.

Could a missing `git reset --hard`  before the pull command be the culprit? This could possibly confuse gis working tree during the pull if I'm not mistaken.

xTrackstar04x, 2021-03-04:
Oops. I was running ```git reset --hard``` in the beginning as well. Must not have copied over when I created this.

riffnshred, 2021-03-04:
I wonder what message it might have thrown out when you did the git reset --hard, that would have solved the mystery. I'll keep that open for a bit in case someone else get this issue. 

the git reset --hard should have cleared any files that were different.

xTrackstar04x, 2021-03-11:
```git reset --hard
git checkout master
git pull
chmod +x scripts/install.sh
./scripts/install.sh
```

Was run for update 1.6.1 and it worked without a problem. Im pretty sure I used ```git reset --hard``` for 1.6.0 but I cant be 100% sure. Idk if you want to keep this open or not

riffnshred, 2021-03-11:
Sweet!!! thanks for letting me know. Cheers bud!

closed 2021-03-11 by riffnshred