riffnshred/nhl-led-scoreboard #237: My config.json seems to revert back to the initial config after I unplug my Pi

alpacula, 2021-03-06:
**Describe the bug**
After I unplug/restart my scoreboard my configuration reverts back to the initial setup config. I have to re-enter my config.json info every time I restart.

**Logs**
Copy and past any error or messages that you can get. 

**Your config file**
{
    "debug": false,
    "loglevel": "INFO",
    "live_mode": true,
    "preferences": {
        "time_format": "12h",
        "end_of_day": "8:00",
        "location": "",
        "live_game_refresh_rate": 10,
        "teams": [
            "Blues"
        ],
        "sog_display_frequency": 4,
        "goal_animations": {
            "pref_team_only": true
        }
    },
    "states": {
        "off_day": [
            "team_summary",
            "clock"
        ],
        "scheduled": [
            "team_summary",
            "scoreticker",
            "clock"
        ],
        "intermission": [
            "scoreticker",
            "team_summary"
        ],
        "post_game": [
            "team_summary",
            "scoreticker",
            "clock"
        ]
    },
    "boards": {
        "scoreticker": {
            "preferred_teams_only": false,
            "rotation_rate": 5
        },
        "seriesticker": {
            "preferred_teams_only": true,
            "rotation_rate": 5
        },
        "standings": {
            "preferred_standings_only": true,
            "standing_type": "wild_card",
            "divisions": "north",
            "conference": "eastern"
        },
        "clock": {
            "duration": 15,
            "hide_indicator": false,
            "preferred_team_colors": true,
            "clock_rgb": "",
            "date_rgb": "",
            "flash_seconds": true
        },
        "covid19": {
            "worldwide_enabled": false,
            "country_enabled": false,
            "country": [
                "USA"
            ],
            "us_state_enabled": false,
            "us_state": [
                "New York"
            ],
            "canada_enabled": false,
            "canada_prov": [
                "Quebec"
            ]
        },
        "weather": {
            "enabled": false,
            "view": "full",
            "units": "metric",
            "duration": 60,
            "data_feed": "EC",
            "owm_apikey": "",
            "update_freq": 5,
            "show_on_clock": true,
            "forecast_enabled": false,
            "forecast_days": 3,
            "forecast_update": 1
        },
        "wxalert": {
            "alert_feed": "EC",
            "update_freq": 5,
            "show_alerts": false,
            "nws_show_expire": false,
            "alert_title": true,
            "scroll_alert": true,
            "alert_duration": 5,
            "show_on_clock": true
        }
    },
    "sbio": {
        "screensaver": {
            "enabled": false,
            "animations": true,
            "start": "22:00",
            "stop": "08:00",
            "data_updates": false,
            "motionsensor": false,
            "pin": 7,
            "delay": 30
        },
        "dimmer": {
            "enabled": false,
            "source": "software",
            "daytime": "",
            "nighttime": "",
            "offset": 0,
            "frequency": 5,
            "light_level_lux": 400,
            "mode": "always",
            "sunset_brightness": 5,
            "sunrise_brightness": 40
        },
        "pushbutton": {
            "enabled": false,
            "bonnet": false,
            "pin": 25,
            "reboot_duration": 2,
            "reboot_override_process": "",
            "display_reboot": true,
            "poweroff_duration": 10,
            "poweroff_override_process": "",
            "display_halt": true,
            "state_triggered1": "weather",
            "state_triggered1_process": ""
        }
    }

**Hardware you are using**
- Raspberry pi B+
- The Matrix driver Adafruit HAT
- Anti-flicker mod or not? No flicker mod.

**Software**
- the latest update
Thanks for the help.

alpacula, 2021-03-06:
I just had this issue... re-entered my config info in nhl_setup... stopped the scoreboard... restarted the scoreboard and got the flashing loading screen. I went back to nhl_setup and got this error:

1.5.2
Now validating config......
Your config.json fails validation: error: [data.preferences.goal_animations must contain ['pref_team_only'] properties]
pi@scoreboard-875f: ~/nhl-led-scoreboard (master)$ 

Am I the only one having this issue?

pjockey, 2021-03-06:
have you tried to edit the /home/pi/nhl-led-scoreboard/config/config,json with nano? That may get the file to be saved.

Perhaps update to version 1.5.7. not sure if that will change anything, but it can't hurt.

alpacula, 2021-03-06:
is there a 1.5.7? I thought it updated automatically

pjockey, 2021-03-06:
Mine is version 1.5.7. I did do an update via git yesterday or the day before and I have 1.5.7 now.

alpacula, 2021-03-06:
just saved the config.json in nano and am updating. Fingers crossed

alpacula, 2021-03-06:
just restarted the scoreboard and got the flashing load screen again, tried to run nhl_setup and got this error
1.5.2
Directory /home/pi/config does not exist.  Are you running in the right directory?
pi@scoreboard-875f: ~ $

pjockey, 2021-03-06:
home/pi/nhl-led-scoreboard/ is where the nhl_setup is located abd should be run from. The config  directory is also located in there.

riffnshred, 2021-03-06:
We just release V1.6.0. If you do the update sequence

`git reset --hard
git checkout master
git pull
chmod +x scripts/install.sh
./scripts/install.sh
`

And then do the nhl_setup. When you run the scoreboard which version does it show ?

alpacula, 2021-03-06:
I still have 1.5.2 on the nhl_setup, I do get this though:

ERROR: Cannot uninstall 'rgbmatrix'. It is a distutils installed project and thus we cannot accurately determine which files belong to it which would lead to only a partial uninstall.
HEAD is now at 60eb1eb fix(config) added the firstrun file
Already up to date.
make: 'submodules/matrix/lib/librgbmatrix.a' is up to date.
If you didn't see any errors above, everything should be installed!
Installation complete! Play around with the examples in nhl-led-scoreboard/submodules/matrix/bindings/python/samples to make sure your matrix is working.

riffnshred, 2021-03-06:
That error can be ignored, its not a bug or anything. Did you just update or have you started from scratch using the nhl-led-scoreboard-img? Just want to make sure

alpacula, 2021-03-06:
I started from scratch and this happens on every attempt

riffnshred, 2021-03-06:
You are using the image ?

alpacula, 2021-03-06:
yes

riffnshred, 2021-03-06:
Thanks Ill check with @falkyre if it could be something from the new tool that reset or delete the config.

riffnshred, 2021-03-06:
Are you using the GUI app from @sflems ?

alpacula, 2021-03-06:
im not sure what that is.

riffnshred, 2021-03-06:
https://github.com/sflems/nhl-led-scoreboard-webgui

Just want to make sure nothing else could cause this

alpacula, 2021-03-06:
no, I just went off the instructions on the page with the downloadable Pi image.

riffnshred, 2021-03-06:
Ok cool, ill try on my side later to see if I can replicate it and find something

alpacula, 2021-03-06:
sweet, thanks

falkyre, 2021-03-06:
How are you restarting your raspberry pi?

alpacula, 2021-03-06:
just pulling the power out of the rib hat. If I restart in supervisor things seem to be fine but if I kill power the config.json vanishes.

riffnshred, 2021-03-06:
Bingo !. So the image runs everything off the ram and on reboot or shutdown, it syncs everything on the SD card. This is to save on read/write on the SD card which extends its life span. If you do a proper reboot after you do your config, I believe it should be properly saved on the SD card. 

Heads up, It's not recommended to pull the plug on a raspberry pi. You are playing Russian roulette when you do this. One suggestion would be to connect a button, we have a script that shuts down or reboots the raspberry pi without having to use a terminal. Check it out in the SBIO documentation, we show how it works.

alpacula, 2021-03-06:
bingo bango... that worked.
I have a power button but haven't soldered it on yet... I'm printing a case (half) as I type this.

Thanks for all the help guys.

riffnshred, 2021-03-06:
No problem bud !!! Enjoy !

closed 2021-03-06 by riffnshred