riffnshred/nhl-led-scoreboard #239: Config.json Not Valid

pjockey, 2021-03-07:
**Describe the bug**
I wiped the ngl-led-scoreboard dir off the SD card and reinstalled the software, 1.6.0, following the instrructions on github. Everything instlalled with out any errors. I had saved my original config.json file as a safety backup then copyoed it back into ~/nhl-led-scoreboard/config/ and tried to start the software, but the json file would not validate. So, I deleted it and started with the config.sample.json and ran ./nhl_setup. Answered the question and the software started. I then tried to update the json file using nano. When I reboot, the error that the json file is not valid and to re-run ./nhl_setup. I did and then I get the error message below. 

**Logs**
1.5.2
Now validating config......
Your config.json fails validation: error: [data must be one of ['wxalert', 'wxforecast', 'scoreticker', 'seriesticker', 'standings', 'team_summary', 'stanley_cup_champions', 'christmas', 'clock', 'covid19', 'weather']]
**Your config file**
A screen shot of your config.json file can be really useful to help you. 
```
{
    "debug": false,
    "loglevel": "INFO",
    "live_mode": true,
    "preferences": {
        "time_format": "24h",
        "end_of_day": "23:00",
        "location": "",
        "live_game_refresh_rate": 10,
        "teams": [
            "Flames"
        ],
        "sog_display_frequency": 4,
        "goal_animations": {
            "pref_team_only": true
        }
    },
    "states": {
        "off_day": [
            "covid_19",
            "team_summary",
            "clock"
        ],
        "scheduled": [
            "team_summary",
            "scoreticker",
            "clock"
        ],
        "intermission": [
            "scoreticker",
            "team_summary"
        ],
        "post_game": [
            "team_summary",
            "scoreticker",
            "clock"
        ]
    },
    "boards": {
        "scoreticker": {
            "preferred_teams_only": false,
            "rotation_rate": 5
        },
        "seriesticker": {
            "preferred_teams_only": true,
            "rotation_rate": 5
        },
        "standings": {
            "preferred_standings_only": true,
            "standing_type": "wild_card",
            "divisions": "north",
            "conference": "eastern"
        },
        "clock": {
            "duration": 15,
            "hide_indicator": false,
            "preferred_team_colors": true,
            "clock_rgb": "",
            "date_rgb": "",
            "flash_seconds": true
        },
        "covid19": {
            "worldwide_enabled": false,
            "country_enabled": true,
            "country": [
                "CA"
            ],
            "us_state_enabled": false,
            "us_state": [
                "New York"
            ],
            "canada_enabled": true,
            "canada_prov": [
                "BC"
            ]
        },
        "weather": {
            "enabled": true,
            "view": "full",
            "units": "metric",
            "duration": 60,
            "data_feed": "EC",
            "owm_apikey": "",
            "update_freq": 5,
            "show_on_clock": true,
            "forecast_enabled": true,
            "forecast_days": 3,
            "forecast_update": 1
        },
        "wxalert": {
            "alert_feed": "EC",
            "update_freq": 5,
            "show_alerts": true,
            "nws_show_expire": false,
            "alert_title": true,
            "scroll_alert": true,
            "alert_duration": 5,
            "show_on_clock": true
        }
    },
    "sbio": {
        "screensaver": {
            "enabled": false,
            "animations": true,
            "start": "23:59",
            "stop": "08:00",
            "data_updates": false,
            "motionsensor": false,
            "pin": 7,
            "delay": 30
        },
        "dimmer": {
            "enabled": false,
            "source": "software",
            "daytime": "",
            "nighttime": "",
            "offset": 0,
            "frequency": 5,
            "light_level_lux": 400,
            "mode": "always",
            "sunset_brightness": 5,
            "sunrise_brightness": 40
        },
        "pushbutton": {
            "enabled": false,
            "bonnet": false,
            "pin": 25,
            "reboot_duration": 2,
            "reboot_override_process": "",
            "display_reboot": true,
            "poweroff_duration": 10,
            "poweroff_override_process": "",
            "display_halt": true,
            "state_triggered1": "weather",
            "state_triggered1_process": ""
        }
    }
}
```

**Hardware you are using**
- Raspberry pi model - Zero W
- The Matrix driver (Adafruit Bonnet)
- Anti-flicker mod or not? Yes

**Software**
- 1.6.0

riffnshred, 2021-03-07:
Try taking out the covid_19 board. I think there is a mistake in the config.schema.json which is what the validator use

pjockey, 2021-03-07:
I get this now

'''hello
INFO: Validating config.json.....
WARN: config.json fails validation: error: [data.boards must contain ['scoreticker', 'standings', 'clock', 'covid19', 'weather'] properties]
WARN: Rerun the nhl_setup app to create a valid config.json
```

pjockey, 2021-03-07:
OK I copied this piece of code from my config.json.backup file back into the config.json and the validation passed. I don't know why.
```
"covid19": {
            "worldwide_enabled": false,
            "country_enabled": false,
            "country": [
                "USA"
            ],
            "us_state_enabled": false,
            "us_state": [
                "New York"
            ],
            "canada_enabled": false,
            "canada_prov": [
                "Quebec"
            ]
                },
```

riffnshred, 2021-03-07:
sorry I meant from the off_day hah. its supposed to be covid_19 but the config.schema.json uses covid19. it will be fixed on the next patch

alpacula, 2021-03-09:
this sounds like the same problem I was having. enter all your config info... save and shut down the Pi by typing 
sudo halt

see if that fixes things

pjockey, 2021-03-09:
Removing the covi19 as @riffnshred advised also worked. 
I used sudo reboot now or sudo shutdown now to manipulate the Pi.

riffnshred, 2021-03-09:
Just a heads-up, I've been thinking of removing the Covid 19 board for a while now and I think it's happening on the next update. It's just an extra thing to maintain and that very few still use. anyone who wants to still use it can fork the repo and incorporate the board back in their setup but I won't maintain that feature anymore.

riffnshred, 2021-03-10:
I would say fork the current version and use that one for now and copy and paste whats missing on the next update. That would be the easiest way.

closed 2021-03-10 by riffnshred