riffnshred/nhl-led-scoreboard #247: Solid green bar across the top row

JoeFromDenver, 2021-03-13:
I started from scratch using the pre-built .img file on both a Pi Zero WH and 3B+ with the no flicker mod. 

I don't see any errors in the logs. Any idea what might cause this? The green bar doesn't show up when loading, or on other screens, but it's there 75% of the time when the scoreboard is in use.

[image]
[image]

JoeFromDenver, 2021-03-13:
Updated to the newest version and it's working as expected. Not sure what was going on there, but it looks great now!

closed 2021-03-13 by JoeFromDenver

riffnshred, 2021-03-13:
Thats the update notification indicator. You can set it to false in the config.

JoeFromDenver, 2021-03-13:
Fantastic, thanks! That makes a lot of sense in hindsight.

Follow-up simple question. I can't find that option in config.json (or the sample) and when I searched for "update notification" in the documentation I don't see it mentioned. I'd love to set that option to false, can you point me in the right direction?

riffnshred, 2021-03-13:
Actually, Im wrong, do you have `--updatecheck` in the command you use to run the scoreboard ?

falkyre, 2021-03-14:
The image puts the update checker in automatically for your supervisor config. It's a command line option.

riffnshred, 2021-03-14:
One way to remove it then is to open the supervisor conf file

`sudo nano /etc/supervisor/conf.d/scoreboard.conf`

and you will see the command that supervisor use to run the scoreboard, find `--updatecheck` and remove it.

riffnshred, 2021-03-14:
I will add it as an option in the config on the next update so you can have it on or off, however you want it

JoeFromDenver, 2021-03-14:
Very cool, thank you!