riffnshred/nhl-led-scoreboard #250: Missing module 'cairo'

msmnr9, 2021-03-14:
**Describe the bug**
I tried to upgrade and a fresh install of the current version.  During the install script I have a [NEW] package python3-pillow but it looks as if nothing is installed during this part of the script.  When I try to run the scoreboard it errors out with 
`ModuleNotFoundError: No module named 'cairo'` 
I'll paste all of my errors below and output of the install script.  I tried a fresh install of python3 and a new clone of the nhl scoreboard github.

**Logs**
Install script output
`./scripts/install.sh
Installing required dependencies. This may take some time (10-20 minutes-ish)...
Running update
[OK] package build-essential
[OK] package git
[OK] package python3-setuptools
[OK] package python3-pip
[OK] package python3-dev
W: --force-yes is deprecated, use one of the options starting with --allow instead.
[NEW] package python3-pillow
[OK] package python3-numpy
[OK] package python3-gpiozero
[OK] package python3-cairosvg
[OK] package libatlas3-base
[OK] package libatlas-base-dev
[OK] package libraqm-dev
[OK] package jq
[OK] package pastebinit
[OK] package neofetch
[OK] package zsh
[OK] package dbus
🚨  ALL GOOD TO GO
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: env-canada==0.0.35 in /usr/local/lib/python3.7/dist-packages (from -r requirements.txt (line 1)) (0.0.35)
Requirement already satisfied: ephem==3.7.7.0 in /usr/local/lib/python3.7/dist-packages (from -r requirements.txt (line 2)) (3.7.7.0)
Requirement already satisfied: fastjsonschema>=2.14.4 in /usr/local/lib/python3.7/dist-packages (from -r requirements.txt (line 3)) (2.14.5)
Requirement already satisfied: geocoder==1.38.1 in /usr/local/lib/python3.7/dist-packages (from -r requirements.txt (line 4)) (1.38.1)
Requirement already satisfied: gpiozero==1.5.1 in /usr/local/lib/python3.7/dist-packages (from -r requirements.txt (line 5)) (1.5.1)
Requirement already satisfied: noaa-sdk>=0.1.18 in /usr/local/lib/python3.7/dist-packages (from -r requirements.txt (line 6)) (0.1.19)
Requirement already satisfied: printtools==1.2 in /usr/local/lib/python3.7/dist-packages (from -r requirements.txt (line 7)) (1.2)
Requirement already satisfied: PyInstaller==3.6 in /usr/local/lib/python3.7/dist-packages (from -r requirements.txt (line 8)) (3.6)
Requirement already satisfied: pyowm>=3.0.0 in /usr/local/lib/python3.7/dist-packages (from -r requirements.txt (line 9)) (3.2.0)
Requirement already satisfied: python-tsl2591==0.2.0 in /usr/local/lib/python3.7/dist-packages (from -r requirements.txt (line 10)) (0.2.0)
Requirement already satisfied: questionary>=1.5.2 in /usr/local/lib/python3.7/dist-packages (from -r requirements.txt (line 11)) (1.9.0)
Requirement already satisfied: regex>=2020.4.4 in /usr/local/lib/python3.7/dist-packages (from -r requirements.txt (line 12)) (2020.11.13)
Requirement already satisfied: RPi.GPIO==0.7.0 in /usr/local/lib/python3.7/dist-packages (from -r requirements.txt (line 13)) (0.7.0)
Requirement already satisfied: APScheduler>=3.6.3 in /usr/local/lib/python3.7/dist-packages (from -r requirements.txt (line 14)) (3.6.3)
Requirement already satisfied: lastversion>=1.1.6 in /usr/local/lib/python3.7/dist-packages (from -r requirements.txt (line 15)) (1.3.0)
Requirement already satisfied: nameparser==1.0.6 in /usr/local/lib/python3.7/dist-packages (from -r requirements.txt (line 16)) (1.0.6)
Requirement already satisfied: pillow==7.1.2 in /usr/local/lib/python3.7/dist-packages (from -r requirements.txt (line 17)) (7.1.2)
Requirement already satisfied: dbus-python in /usr/local/lib/python3.7/dist-packages (from -r requirements.txt (line 18)) (1.2.16)
Requirement already satisfied: iso6709 in /usr/local/lib/python3.7/dist-packages (from -r requirements.txt (line 19)) (0.1.5)
Requirement already satisfied: requests-futures>=0.9.7 in /usr/local/lib/python3.7/dist-packages (from env-canada==0.0.35->-r requirements.txt (line 1)) (1.0.0)
Requirement already satisfied: ratelimit>=2.2.1 in /usr/local/lib/python3.7/dist-packages (from env-canada==0.0.35->-r requirements.txt (line 1)) (2.2.1)
Requirement already satisfied: requests>=2.19.1 in /usr/local/lib/python3.7/dist-packages (from env-canada==0.0.35->-r requirements.txt (line 1)) (2.22.0)
Requirement already satisfied: geopy>=1.16.0 in /usr/local/lib/python3.7/dist-packages (from env-canada==0.0.35->-r requirements.txt (line 1)) (2.1.0)
Requirement already satisfied: imageio>=2.3.0 in /usr/local/lib/python3.7/dist-packages (from env-canada==0.0.35->-r requirements.txt (line 1)) (2.9.0)
Requirement already satisfied: python-dateutil>=2.8.1 in /usr/local/lib/python3.7/dist-packages (from env-canada==0.0.35->-r requirements.txt (line 1)) (2.8.1)
Requirement already satisfied: click in /usr/local/lib/python3.7/dist-packages (from geocoder==1.38.1->-r requirements.txt (line 4)) (7.1.2)
Requirement already satisfied: six in /usr/lib/python3/dist-packages (from geocoder==1.38.1->-r requirements.txt (line 4)) (1.12.0)
Requirement already satisfied: future in /usr/local/lib/python3.7/dist-packages (from geocoder==1.38.1->-r requirements.txt (line 4)) (0.18.2)
Requirement already satisfied: ratelim in /usr/local/lib/python3.7/dist-packages (from geocoder==1.38.1->-r requirements.txt (line 4)) (0.1.6)
Requirement already satisfied: colorzero in /usr/local/lib/python3.7/dist-packages (from gpiozero==1.5.1->-r requirements.txt (line 5)) (1.1)
Requirement already satisfied: pyfiglet in /usr/local/lib/python3.7/dist-packages (from printtools==1.2->-r requirements.txt (line 7)) (0.8.post1)
Requirement already satisfied: setuptools in /usr/lib/python3/dist-packages (from PyInstaller==3.6->-r requirements.txt (line 8)) (40.8.0)
Requirement already satisfied: altgraph in /usr/local/lib/python3.7/dist-packages (from PyInstaller==3.6->-r requirements.txt (line 8)) (0.17)
Requirement already satisfied: geojson<3,>=2.3.0 in /usr/local/lib/python3.7/dist-packages (from pyowm>=3.0.0->-r requirements.txt (line 9)) (2.5.0)
Requirement already satisfied: PySocks<2,>=1.7.1 in /usr/local/lib/python3.7/dist-packages (from pyowm>=3.0.0->-r requirements.txt (line 9)) (1.7.1)
Requirement already satisfied: smbus2>=0.2 in /usr/local/lib/python3.7/dist-packages (from python-tsl2591==0.2.0->-r requirements.txt (line 10)) (0.4.0)
Requirement already satisfied: prompt_toolkit<4.0,>=2.0 in /usr/local/lib/python3.7/dist-packages (from questionary>=1.5.2->-r requirements.txt (line 11)) (3.0.16)
Requirement already satisfied: tzlocal>=1.2 in /usr/local/lib/python3.7/dist-packages (from APScheduler>=3.6.3->-r requirements.txt (line 14)) (2.1)
Requirement already satisfied: pytz in /usr/local/lib/python3.7/dist-packages (from APScheduler>=3.6.3->-r requirements.txt (line 14)) (2020.5)
Requirement already satisfied: PyYAML in /usr/local/lib/python3.7/dist-packages (from lastversion>=1.1.6->-r requirements.txt (line 15)) (5.3.1)
[... 252 lines cut ...]
        }
    },
    "sbio": {
        "screensaver": {
            "enabled": false,
            "animations": true,
            "start": "22:00",
            "stop": "08:00",
            "data_updates": false,
            "motionsensor": false,
            "pin": 7,
            "delay": 30
        },
        "dimmer": {
            "enabled": false,
            "source": "software",
            "daytime": "",

**Hardware you are using**
- Raspberry pi model: Pi 4 B
- The Matrix driver: Adafruit Bonnet
- Anti-flicker mod: Yes

**Software**
- Which version of the software are you on? 1.6.2

riffnshred, 2021-03-14:
If you run 

`pip3 install pycairo`

does that fix it ?

msmnr9, 2021-03-15:
Unfortunately no.

```
pip3 install pycairo
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting pycairo
  Using cached https://www.piwheels.org/simple/pycairo/pycairo-1.20.0-cp37-cp37m-linux_armv7l.whl (260 kB)
Installing collected packages: pycairo
Successfully installed pycairo-1.20.0
```
```
sudo python3 src/main.py --led-gpio-mapping=adafruit-hat-pwm --led-brightness=60 --led-slowdown-gpio=2
    ignoring unknown attr section aeabi"
    ignoring unknown attr section aeabi"
    ignoring unknown attr section aeabi"
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/cairosvg/surface/__init__.py", line 25, in <module>
    import cairocffi as cairo
  File "/usr/lib/python3/dist-packages/cairocffi/__init__.py", line 46, in <module>
    cairo = dlopen(ffi, 'cairo', 'cairo-2')
  File "/usr/lib/python3/dist-packages/cairocffi/__init__.py", line 43, in dlopen
    raise OSError("dlopen() failed to load a library: %s" % ' / '.join(names))
OSError: dlopen() failed to load a library: cairo / cairo-2

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "src/main.py", line 5, in <module>
    from renderer.main import MainRenderer
  File "/home/pi/nhl-led-scoreboard/src/renderer/main.py", line 5, in <module>
    from boards.boards import Boards
  File "/home/pi/nhl-led-scoreboard/src/boards/boards.py", line 7, in <module>
    from boards.scoreticker import Scoreticker
  File "/home/pi/nhl-led-scoreboard/src/boards/scoreticker.py", line 7, in <module>
    from renderer.scoreboard import ScoreboardRenderer
  File "/home/pi/nhl-led-scoreboard/src/renderer/scoreboard.py", line 3, in <module>
    from renderer.logos import LogoRenderer
  File "/home/pi/nhl-led-scoreboard/src/renderer/logos.py", line 3, in <module>
    from images.image_helper import ImageHelper
  File "/home/pi/nhl-led-scoreboard/src/images/image_helper.py", line 2, in <module>
    import cairosvg
  File "/usr/lib/python3/dist-packages/cairosvg/__init__.py", line 27, in <module>
    from . import surface
  File "/usr/lib/python3/dist-packages/cairosvg/surface/__init__.py", line 30, in <module>
    import cairo  # pycairo
ModuleNotFoundError: No module named 'cairo'
```

falkyre, 2021-03-15:
The cairosvg is installed as a packaged file via apt and not through pip.  So you need to use the `apt install python3-cairosvg`.  For some reason, the pip version causes issues.

riffnshred, 2021-03-15:
When you say you did a fresh install, do you mean you just deleted the repo and reinstalled it or you formatted the sd card and installed the new raspberry pi os and then installed the software ?

msmnr9, 2021-03-15:
For the fresh install I deleted the repo and reinstalled.  I haven't had a chance to reformat the sd card.  I'm running pihole on it as well and haven't had the time to take down the network yet.

I tried installed the cairosvg package through apt and it gave me 

`E: Write error - ~LZMAFILE (28: No space left on device)`

Did some research and it looks like I have no available Inodes to write the changes needed by apt.  If I run the df command I get

```
df
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root       14679776 2533916  11516940  19% /
devtmpfs          823564       0    823564   0% /dev
tmpfs             955660    4136    951524   1% /dev/shm
tmpfs             955660   92264    863396  10% /run
tmpfs               5120       0      5120   0% /run/lock
tmpfs             955660       0    955660   0% /sys/fs/cgroup
/dev/mmcblk0p1    258095   48782    209314  19% /boot
log2ram            40960   40960         0 100% /var/log
tmpfs             191132       0    191132   0% /run/user/999
tmpfs             191132       0    191132   0% /run/user/1000
```
And when I run df -i I get

```
df -i
Filesystem     Inodes IUsed  IFree IUse% Mounted on
/dev/root      931040 63190 867850    7% /
devtmpfs       103559   394 103165    1% /dev
tmpfs          169607    12 169595    1% /dev/shm
tmpfs          169607   527 169080    1% /run
tmpfs          169607     2 169605    1% /run/lock
tmpfs          169607    15 169592    1% /sys/fs/cgroup
/dev/mmcblk0p1      0     0      0     - /boot
log2ram        169607    48 169559    1% /var/log
tmpfs          169607    10 169597    1% /run/user/999
tmpfs          169607    10 169597    1% /run/user/1000
```
Since my /boot has 0 Inoes I assume that's my problem.  And I assume reformatting the sd card is the best bet.  I haven't changed any system settings besides regular updates so I'm not sure where it could have got bungled up.

riffnshred, 2021-03-15:
Yeah the best bet is formatting the SD card and installing raspberry pi OS. Im not sure to what extend using the raspberry pi for something else like running Pihole could affect the scoreboard software.

closed 2021-03-15 by riffnshred