riffnshred/nhl-led-scoreboard #256: Custom Goal Animation

kempdawg20, 2021-03-19:
I may have misread the instructions on how to do this so any guidance would be great. I wanted to have a goal animation for preferred and a separate one for opposing. I did as instructed putting gifs in the respective folders but the general one kept firing. I removed the general gif and now I'm getting an error. Appreciate the guidance on this!

Traceback (most recent call last):
  File "src/main.py", line 157, in <module>
    run()
  File "src/main.py", line 152, in run
    MainRenderer(matrix, data, sleepEvent).render()
  File "/home/pi/nhl-led-scoreboard/src/renderer/main.py", line 76, in render
    self.__render_game_day()
  File "/home/pi/nhl-led-scoreboard/src/renderer/main.py", line 150, in __render_game_day
    self.check_new_goals()
  File "/home/pi/nhl-led-scoreboard/src/renderer/main.py", line 293, in check_new_goals
    self._draw_event_animation("goal", home_id, home_name)
  File "/home/pi/nhl-led-scoreboard/src/renderer/main.py", line 368, in _draw_event_animation
    self.play_gif(filename)
  File "/home/pi/nhl-led-scoreboard/src/renderer/main.py", line 371, in play_gif
    im = Image.open(get_file(file))
  File "/usr/local/lib/python3.7/dist-packages/PIL/Image.py", line 2843, in open
    fp = builtins.open(filename, "rb")
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/nhl-led-scoreboard/assets/animations/goal/goal_light_animation.gif'

**Hardware you are using**
- Pi Zero
- The Matrix driver: Adafruit Bonnet
- Anti-flicker mod or not?: No

**Software**
-Version 1.6.2

riffnshred, 2021-03-22:
Turns out I disabled this feature by adding the penalty animation. Thanks for finding this, I will add it back in with the other stuff I'm working on.

riffnshred, 2021-03-24:
This is now fixed on the beta branch. I will not release it until next official update but to use it do

`git checkout beta` and you will switch to the Beta branch. I will add new features and fixed to that branch in the coming month which could break the scoreboard or create new bug. I would suggest switching now and not update after that unless you are interested in testing new features.

cheers

closed 2021-03-24 by riffnshred

kempdawg20, 2021-05-06:
Riff, since the last update having the same issue. Tried the beta version and it crashes due to the Seattle team issue.

riffnshred, 2021-05-08:
Give it an other go. I just merged the last commits from the master onto the Beta branch