riffnshred/nhl-led-scoreboard #258: Tried Updating, but stilling getting V1.5.1

jwallner0710, 2021-03-24:
I've been a part of this group for a few months now. I follow the announcements on discord and try to stay up-to-date. I am trying to update to V1.6.2 and something appears to be off. I have run the following code

git reset --hard
git checkout master
git pull
chmod +x scripts/install.sh
./scripts/install.sh

,but my pi doesn't seem to responding to the first 4 lines of code. It does do it's install thing after ./scripts/install.sh is entered, but I'm stuck with 1.5.1. 

Any help is appreciated. Thank you all!

falkyre, 2021-03-24:
How are you seeing that it's version 1.5.1?  What's the file named VERSION have in it?

jwallner0710, 2021-03-24:
After I supposedly update. (I don't think I'm updating correctly), I run the ./nhl_setup and it gives me the V1.5.1 when it asks me all of the reconfigure questions, like specify team, pre-game boards, etc.

pjockey, 2021-03-25:
did you cd into the nhl-led-scoreboard directory first before running those commands?

jwallner0710, 2021-03-25:
Yes. Here is what I do. 

1. I sign in via SSH and putty. 
2. Type username and password
3. cd nhl-led-scoreboard
4. Run this code: 
git reset --hard
git checkout master
git pull
chmod +x scripts/install.sh
./scripts/install.sh

5. Reconfig with ./nhl_setup and still get V1.5.1

pjockey, 2021-03-25:
Does the feedback from git tell you that it has downloaded 1.6.2? can you copied the output from git?

jwallner0710, 2021-03-25:
Here is what I get when I run the first line of code. 

fatal: not a git repository (or any of the parent directories): .git

jwallner0710, 2021-03-25:
This may be a silly question but with is config.json?

pjockey, 2021-03-25:
you're not in pi@raspberry/nhl-led-scoreboard/config  are you?

jwallner0710, 2021-03-25:
No. Here is a look at my code. 

[image]

riffnshred, 2021-03-25:
it's the file that contains the configuration for the scoreboard. It what the scoreboard read to know how to operate.

seeing the error you get when you run the git reset, I would delete and reinstall it.

The step to remove and reinstall can be found in the step by step guide here
https://github.com/riffnshred/nhl-led-scoreboard/wiki/Step-by-step-guide.#step-4---installing-the-nhl-scoreboard-software

jwallner0710, 2021-03-25:
Thank you sir!

pjockey, 2021-03-25:
I was getting to think it maybe easier to re-install. If you can, save your config.json file. Not a big if not.

jwallner0710, 2021-03-25:
By the way, you guys have done a great job of helping out newcomers and organizing all of this! Thank you!

falkyre, 2021-03-25:
> After I supposedly update. (I don't think I'm updating correctly), I run the ./nhl_setup and it gives me the V1.5.1 when it asks me all of the reconfigure questions, like specify team, pre-game boards, etc.

That's the version number of the nhl_setup, it's different from the actual scoreboard version.  Anyways, you have an older version for sure as the new nhl_setup will now show it's version and the version of the scoreboard code.  
`nhl_setup V1.6.1
nhl led scoreboard V1.6.2
`

Plus, the scoreboard will print out what version it is in the logs when you start it.

I agree with @riffnshred that it's probably easier to do a reinstall.  Or if you want, you can try the image that we have now.   It's the latest version and everything is setup already, you just need to get it onto an SD card, power up your pi and do a few things to set it up.    See here:  https://github.com/falkyre/nhl-led-scoreboard-img

closed 2021-05-02 by jwallner0710