riffnshred/nhl-led-scoreboard #261: Config.json fails validation after not saving on intial run of nhl_setup

mij2, 2021-04-02:
I recently installed the project, but then started from scratch without a desktop interface after reading a lot of recommendations. While going through the setup I launched the nhl_setup and clicked a few options I didn't mean to. When I was prompted to save, I clicked no. Upon relaunching of nhl_setup, I get an error that says "Your config.json fails validation: error: [unable to load config/config.json or config/config.schema.json]

Ultimately I was able to resolve the problem myself by copying the config.json.sample and renaming as config.json. However, it took me a bit of time before I figured out that this would resolve the issue. nhl_setup could use an option to start a new config if the config.json file is invalid or a similar option.

Hardware
- Pi Zero W
- Adafruit RGB Matrix hat
- No Anti-Flicker Mod 

Software
- v1.6.2

riffnshred, 2021-04-02:
thanks for the heads up. its because the `firstrun` file is deleted after the first nhl_setup and after that the validator check you config and because its faulty it wont work. quickes way to fix it is either by doing what you did by doing a copy of the config.json.sample (which was the old way to do it before we had an nhl_setup app) or delete the repo and clone it again or create a new "firtrun" file in the config folder.

riffnshred, 2021-04-02:
I'll keep this one open in  case someone would want to make a fix for it.

falkyre, 2021-04-03:
I'll look into making sure the first run file only gets deleted on successful creation of the config.json