riffnshred/nhl-led-scoreboard #264: Flicker & Changing LED Brightness

dickeyad, 2021-04-10:
I used Method 1 (Using the nhl-led-scoreboard-img (Recommended)) to install with no issues, but there is still a slight flicker. I read that you can fix this by adjusting the flicker. How do I manually adjust (lower) the flicker?

pjockey, 2021-04-10:
Are these in your start up line? Brightness can be changed and you need the pwm added to the line if you did the hardware fix.

-led-gpio-mapping=adafruit-hat**-pwm** **--led-brightness=60** --led-slowdown-gpio=2

dickeyad, 2021-04-10:
Thanks for the quick reply. I did not do a hardware fix. I am using a Raspberry Pi 3 (which improved the flicker after switching from a Pi Zero).

Should I still do the soldering fix that was mentioned in previous documentation?

Since I currently didn’t do the fix, is there another line of code that can be used?

pjockey, 2021-04-10:
I would do the hardware fix. Easy to do, small piece of wire and some solder.

riffnshred, 2021-04-12:
anti flickering mod is a must and running on a Pi 3 and up is also strongly recommended. Many things will affect this but those are the fix that will reduce 80% of the flickering. After this, the rest of the occasional flickery can be fix by playing around with the different flags. Keep in mind, you still will have some flicker from time to time.

closed 2021-04-12 by riffnshred

dickeyad, 2021-04-18:
So I did the solder fix, connected, and then went to "Scoreboard Options," (sb tools) then S2 - "Supervisor Config...", then "SV1 scoreboard.conf" and then go to option 2 which is "64x32 adafruiter flcker mod", ru n the test, confirm, and still same problem. Apologies for being a "noob," but I'm just not getting it. Please help :)

riffnshred, 2021-04-18:
the flickering it the same as before? can you copy-paste the command you are using?

dickeyad, 2021-04-19:
That's what I'm confused about. I haven't entered a command, nor do I know how to update the "start-up" line (noob here). How do I update the start-up line commands?

riffnshred, 2021-04-19:
sudo nano /etc/supervisor/conf.d/scoreboard.conf

thats your supervisor config file for you scoreboard that was generated when you installed using the image.

the second line is your command.

dickeyad, 2021-04-19:
Here is how it looks as of now. I am not sure how to save it once I update :(

[program:scoreboard] command=python3 src/main.py --led-slowdown-gpio=2 --led-gpio-mapping=adafruit-hat
--led-rows=32 --led-cols=64 --led-brightness=40 --updatecheck directory=/home/pi/nhl-led-scoreboard
autostart=true autorestart=true stderr_logfile=/var/log/scoreboard.stderr.log stderr_logfile_maxbytes=1MB
stderr_logfile_backups=10 stdout_logfile=/var/log/scoreboard.stdout.log stdout_logfile_maxbytes=1MB
stdout_logfile_backups=10

riffnshred, 2021-04-19:
replace the command for this one.

`--led-gpio-mapping=adafruit-hat-pwm --led-brightness=50 --led-slowdown-gpio=2 --led-pwm-bits=10 --led-limit-refresh=120 --led-pwm-dither-bits=1`

to save do

```
Press CTRL-x
Press y
Press [enter]
```