riffnshred/nhl-led-scoreboard #266: Pi flashes loading sign and does not boot up.

ArcticShadows45, 2021-04-16:
What format does the location have to be? I think thats what is throwing the Error. 

INFO: Validating config.json.....
INFO: config.json passes validation
Status code Unknown from https://nominatim.openstreetmap.org/search: ERROR - HTTPSConnectionPool(host='nominatim.openstreetmap.org', port=443): Max retries exceeded with url: /search?q=Columbus%2COH&format=jsonv2&addressdetails=1&limit=1 (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x6ffc5770>: Failed to establish a new connection: [Errno 111] Connection refused'))
Traceback (most recent call last):
  File "src/main.py", line 157, in <module>
    run()
  File "src/main.py", line 61, in run
    data = Data(config)
  File "/home/pi/nhl-led-scoreboard/src/data/data.py", line 86, in __init__
    self.latlng, self.latlng_msg = get_lat_lng(config.location)
  File "/home/pi/nhl-led-scoreboard/src/utils.py", line 72, in get_lat_lng
    message = "Unable to find [{}] with Open Street Map, used IP address to find your location is: ".format(location) + g.city + ","+ g.country + " " + str(g.latlng)
TypeError: can only concatenate str (not "NoneType") to str

Thanks!

falkyre, 2021-04-16:
Your raspberry pi for whatever reason is being blocked to the openstreetmap server (error 111: connection refused).  The URL that it's trying to hit is:  https://nominatim.openstreetmap.org/search?q=Columbus%2COH&format=jsonv2&addressdetails=1&limit=1

I tried that and I get a valid json returned with all the data required to get the latitude and longitude for Columbus, Ohio.

So you can try that URL in a normal browser and see if you are being blocked from the server (maybe from it trying a bunch of times, your IP address is blacklisted for a period of time).

The other option is to leave the location blank in the config.json and allow the lat/lon to be found by your IP address.  This works for 98% of the people just fine.

ArcticShadows45, 2021-04-16:
Changed that, this is what the new error code is 

INFO: Validating config.json.....
INFO: config.json passes validation
Traceback (most recent call last):
  File "src/main.py", line 157, in <module>
    run()
  File "src/main.py", line 61, in run
    data = Data(config)
  File "/home/pi/nhl-led-scoreboard/src/data/data.py", line 86, in __init__
    self.latlng, self.latlng_msg = get_lat_lng(config.location)
  File "/home/pi/nhl-led-scoreboard/src/utils.py", line 80, in get_lat_lng
    g = geocoder.ip('me')
  File "/usr/local/lib/python3.7/dist-packages/geocoder/api.py", line 498, in ip
    return get(location, provider='ipinfo', **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/geocoder/api.py", line 198, in get
    return options[provider][method](location, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/geocoder/base.py", line 410, in __init__
    self._initialize()
  File "/usr/local/lib/python3.7/dist-packages/geocoder/base.py", line 457, in _initialize
    json_response = self._connect()
  File "/usr/local/lib/python3.7/dist-packages/geocoder/base.py", line 492, in _connect
    json_response = response.json()
  File "/usr/local/lib/python3.7/dist-packages/requests/models.py", line 897, in json
    return complexjson.loads(self.text, **kwargs)
  File "/usr/lib/python3.7/json/__init__.py", line 348, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3.7/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python3.7/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

falkyre, 2021-04-16:
Is there a location.json in the config directory?  If so, delete it and try again.  

Also try this on the pi:  wget https://ipinfo.io

Look at the index.html that gets downloaded.  What does it look like?

ArcticShadows45, 2021-04-16:
It stopped doing that message, heres the new one 

pi@scoreboard-e87e: ~/nhl-led-scoreboard (master)$ sudo python3 src/main.py --le                                                                                                                                                      d-gpio-mapping=adafruit-hat --led-brightness=60 --led-slowdown-gpio=2
INFO: Validating config.json.....
INFO: config.json passes validation
Status code Unknown from http://ipinfo.io/json: ERROR - HTTPConnectionPool(host=                                                                                                                                                      'ipinfo.io', port=80): Max retries exceeded with url: /json (Caused by ConnectTi                                                                                                                                                      meoutError(<urllib3.connection.HTTPConnection object at 0x6ff598d0>, 'Connection                                                                                                                                                       to ipinfo.io timed out. (connect timeout=5.0)'))
Failed to Get the list of Teams. 5 attempt remaining.
HTTPConnectionPool(host='statsapi.web.nhl.com', port=80): Max retries exceeded w                                                                                                                                                      ith url: /api/v1/teams?expand=team.roster,team.stats,team.schedule.previous,team                                                                                                                                                      .schedule.next (Caused by ConnectTimeoutError(<urllib3.connection.HTTPConnection                                                                                                                                                       object at 0x70486290>, 'Connection to statsapi.web.nhl.com timed out. (connect                                                                                                                                                       timeout=5)'))
Failed to Get the list of Teams. 4 attempt remaining.
HTTPConnectionPool(host='statsapi.web.nhl.com', port=80): Max retries exceeded with url: /api/v1/teams?expand=team.roster,team.stats,team.schedule.previous,team.schedule.next (Caused by ConnectTimeoutError(<urllib3.connection.HTTPConnection object at 0x6ff59e30>, 'Connection to statsapi.web.nhl.com timed out. (connect timeout=5)'))
Failed to Get the list of Teams. 3 attempt remaining.
HTTPConnectionPool(host='statsapi.web.nhl.com', port=80): Max retries exceeded with url: /api/v1/teams?expand=team.roster,team.stats,team.schedule.previous,team.schedule.next (Caused by ConnectTimeoutError(<urllib3.connection.HTTPConnection object at 0x6ff6a290>, 'Connection to statsapi.web.nhl.com timed out. (connect timeout=5)'))
Failed to Get the list of Teams. 2 attempt remaining.
HTTPConnectionPool(host='statsapi.web.nhl.com', port=80): Max retries exceeded with url: /api/v1/teams?expand=team.roster,team.stats,team.schedule.previous,team.schedule.next (Caused by ConnectTimeoutError(<urllib3.connection.HTTPConnection object at 0x6ff6a6d0>, 'Connection to statsapi.web.nhl.com timed out. (connect timeout=5)'))
Failed to Get the list of Teams. 1 attempt remaining.
HTTPConnectionPool(host='statsapi.web.nhl.com', port=80): Max retries exceeded with url: /api/v1/teams?expand=team.roster,team.stats,team.schedule.previous,team.schedule.next (Caused by ConnectTimeoutError(<urllib3.connection.HTTPConnection object at 0x6ff6ab10>, 'Connection to statsapi.web.nhl.com timed out. (connect timeout=5)'))
getting status
Failed to refresh the Status data. 5 attempt remaining.
HTTPConnectionPool(host='statsapi.web.nhl.com', port=80): Max retries exceeded with url: /api/v1/gameStatus (Caused by ConnectTimeoutError(<urllib3.connection.HTTPConnection object at 0x6ff6af70>, 'Connection to statsapi.web.nhl.com timed out. (connect timeout=5)'))
getting status
Failed to refresh the Status data. 4 attempt remaining.
HTTPConnectionPool(host='statsapi.web.nhl.com', port=80): Max retries exceeded with url: /api/v1/gameStatus (Caused by ConnectTimeoutError(<urllib3.connection.HTTPConnection object at 0x6ff6ae70>, 'Connection to statsapi.web.nhl.com timed out. (connect timeout=5)'))
getting status

it kept repeating that.

ArcticShadows45, 2021-04-16:
File "src/main.py", line 157, in <module>
    run()
  File "src/main.py", line 61, in run
    data = Data(config)
  File "/home/pi/nhl-led-scoreboard/src/data/data.py", line 174, in __init__
    self.refresh_playoff()
  File "/home/pi/nhl-led-scoreboard/src/data/data.py", line 483, in refresh_playoff
    self.playoffs = nhl_api.playoff(self.status.season_id)
AttributeError: 'list' object has no attribute 'season_id'

ArcticShadows45, 2021-04-16:
[image]

ArcticShadows45, 2021-04-16:
After flashing again and going through setup. same thing.

ArcticShadows45, 2021-04-16:
[image]

falkyre, 2021-04-16:
I think you have an issue with the wifi or internet because you are not even able to connect to anything (the nhl api, the ipinfo web site).  

What does `ping google.com` return back?
What does `ping ipinfo.io` show?

ArcticShadows45, 2021-04-17:
that was the problem a few ago. since i got that fixed this is the returned message 

[image]

falkyre, 2021-04-17:
Ok, don't run the application, don't even try.  Let's see if you can even get anything from ipinfo.io from the command line.

`wget ipinfo.io`

what's the index.html show?

do a `cat index.html`

Anything?

ArcticShadows45, 2021-04-17:
I finally got it to ping correctly by changing my ethernet cable. 

[image]

ArcticShadows45, 2021-04-17:
fatal: unable to access 'https://github.com/riffnshred/nhl-led-scoreboard.git/': Failed to connect to github.com port 443: Connection timed out
fatal: unable to access 'https://github.com/riffnshred/nhl-led-scoreboard.git/': Failed to connect to github.com port 443: Connection timed out

ArcticShadows45, 2021-04-17:
a/data.py", line 174, in __init__
    self.refresh_playoff()
  File "/home/pi/nhl-led-scoreboard/src/data/data.py", line 483, in refresh_playoff
    self.playoffs = nhl_api.playoff(self.status.season_id)
AttributeError: 'list' object has no attribute 'season_id'
INFO: Validating config.json.....
INFO: config.json passes validation
Status code Unknown from http://ipinfo.io/json: ERROR - HTTPConnectionPool(host='ipinfo.io', port=80): Max retries exceeded with url: /json (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x6ff49a50>: Failed to establish a new connection: [Errno 111] Connection refused'))
Failed to Get the list of Teams. 5 attempt remaining.
HTTPConnectionPool(host='statsapi.web.nhl.com', port=80): Max retries exceeded with url: /api/v1/teams?expand=team.roster,team.stats,team.schedule.previous,team.schedule.next (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x703f8270>: Failed to establish a new connection: [Errno 111] Connection refused'))
Failed to Get the list of Teams. 4 attempt remaining.
HTTPConnectionPool(host='statsapi.web.nhl.com', port=80): Max retries exceeded with url: /api/v1/teams?expand=team.roster,team.stats,team.schedule.previous,team.schedule.next (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x6ff49eb0>: Failed to establish a new connection: [Errno 111] Connection refused'))
Failed to Get the list of Teams. 3 attempt remaining.

riffnshred, 2021-04-17:
Dont use any of the playoff feature at the moment. In your config remove
the series ticker and the stanley cup champions bpards from any states.

riffnshred, 2021-04-17:
You are having connexion issues.  The pi seem to have a hard time cpmmunicating with the internet.

ArcticShadows45, 2021-04-17:
It is pinging google and everywhere except the API

Ill remove the playoff features and post after

> You are having connexion issues. The pi seem to have a hard time cpmmunicating with the internet.

ArcticShadows45, 2021-04-17:
nevermind, not pinging currently. 

removed playoff features.

riffnshred, 2021-04-17:
If you are directly connected to your router, it looks like you Pi is dying or something. Which model are you using ?

ArcticShadows45, 2021-04-17:
Raspberry Pi 3 Model B Rev 1.2

riffnshred, 2021-04-17:
Kk. Yeah if you can't ping consistently through wifi or wired up,  my bet
is on a dying network chip or something. The errors you posted are caused
by failed connexions.

ArcticShadows45, 2021-04-19:
New Raspberry Pi 3 Model B Plus Rev 1.3  

Same issue. 

json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

it keeps looping back to that same issue.

Traceback (most recent call last):
  File "src/main.py", line 157, in <module>
    run()
  File "src/main.py", line 61, in run
    data = Data(config)
  File "/home/pi/nhl-led-scoreboard/src/data/data.py", line 86, in __init__
    self.latlng, self.latlng_msg = get_lat_lng(config.location)
  File "/home/pi/nhl-led-scoreboard/src/utils.py", line 80, in get_lat_lng
    g = geocoder.ip('me')
  File "/usr/local/lib/python3.7/dist-packages/geocoder/api.py", line 498, in ip
    return get(location, provider='ipinfo', **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/geocoder/api.py", line 198, in get
    return options[provider][method](location, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/geocoder/base.py", line 410, in __init__
    self._initialize()
  File "/usr/local/lib/python3.7/dist-packages/geocoder/base.py", line 457, in _initialize
    json_response = self._connect()
  File "/usr/local/lib/python3.7/dist-packages/geocoder/base.py", line 492, in _connect
    json_response = response.json()
  File "/usr/local/lib/python3.7/dist-packages/requests/models.py", line 897, in json
    return complexjson.loads(self.text, **kwargs)
  File "/usr/lib/python3.7/json/__init__.py", line 348, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3.7/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python3.7/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

riffnshred, 2021-04-19:
you tried the fixes @falkyre gave in the previous comments? It still looks like a poor connection.

falkyre, 2021-04-19:
Are you getting an IP address for your raspberry Pi?  What does this command return back when you run it?

`ip a`

ArcticShadows45, 2021-04-19:
[image]

ArcticShadows45, 2021-04-19:
I have no idea why it wont connect to google or anything external to the network.

this is another one i had lying around. that is known to work.

falkyre, 2021-04-19:
Try downloading the official raspberry pi image and see if that works with your Ethernet.  The image for this project is more setup to get an easy wireless connection via wifi.  There "could" be something in those settings that "might" be conflicting with the Ethernet.   That being said, the image is built off the official Raspberry Pi OS and just has extensions for this project.  The Ethernet is supposed to take precedence over wifi.  Your Ethernet setting doesn't look right to me but I have no idea what's providing your DHCP, if it's a simple network or what device you are using for a router.

anjuffras, 2021-04-30:
hello I reinstalled the software and did the ./nhlsetup but I still get this error.
 Traceback (most recent call last):
  File "src/main.py", line 157, in <module>
    run()
  File "src/main.py", line 61, in run
    data = Data(config)
  File "/home/pi/nhl-led-scoreboard/src/data/data.py", line 134, in __init__
    self.teams = self.get_teams()
  File "/home/pi/nhl-led-scoreboard/src/data/data.py", line 240, in get_teams
    teams = nhl_api.teams()
  File "/home/pi/nhl-led-scoreboard/src/nhl_api/__init__.py", line 23, in teams
    return [nhl_api.info.Info(x) for x in nhl_api.info.team_info()]
  File "/home/pi/nhl-led-scoreboard/src/nhl_api/info.py", line 22, in team_info
    short_name = team['shortName']
KeyError: 'shortName'

im not sure what this error means it is different from others errors

riffnshred, 2021-04-30:
Are you trying to set the Kraken as your team?

anjuffras, 2021-04-30:
no
I set bruins and sabers

riffnshred, 2021-04-30:
Just checked mine, I'm having the same issue. there is something going on with the API. I'll if its still acting up later tonight, I'll try to find whats going on

anjuffras, 2021-04-30:
thankyou

riffnshred, 2021-05-03:
This got fixed on v1.6.3

closed 2021-05-03 by riffnshred