riffnshred/nhl-led-scoreboard #267: ValueError when running from source

craigjmidwinter, 2021-04-26:
**Describe the bug**
Running from source on the master branch on a fresh raspbian I see a value error on startup, but the examples work.

**Logs**
```
pi@raspberrypi:~/nhl-led-scoreboard $ sudo python3 src/main.py --led-rows=32 --led-cols=64 --led-gpio-mapping=adafruit-hat --led-brightness=60
Traceback (most recent call last):
  File "src/main.py", line 15, in <module>
    from api.weather.ecWeather import ecWxWorker
  File "/home/pi/nhl-led-scoreboard/src/api/weather/ecWeather.py", line 1, in <module>
    from env_canada import ECData
  File "/usr/local/lib/python3.7/dist-packages/env_canada/__init__.py", line 2, in <module>
    from .ec_radar import *
  File "/usr/local/lib/python3.7/dist-packages/env_canada/ec_radar.py", line 9, in <module>
    import imageio
  File "/usr/local/lib/python3.7/dist-packages/imageio/__init__.py", line 22, in <module>
    from .core import FormatManager, RETURN_BYTES
  File "/usr/local/lib/python3.7/dist-packages/imageio/core/__init__.py", line 10, in <module>
    from .util import Image, Array, Dict, asarray, image_as_uint, urlopen
ValueError: source code string cannot contain null bytes
```

**Your config file**
```
{
    "debug": false,
    "loglevel": "INFO",
    "live_mode": true,
    "preferences": {
        "time_format": "12h",
        "end_of_day": "8:00",
        "location": "",
        "live_game_refresh_rate": 10,
        "teams": [
            "Jets"
        ],
        "sog_display_frequency": 4,
        "goal_animations": {
            "pref_team_only": true
        }
    },
    "states": {
        "off_day": [
            "team_summary",
            "clock"
        ],
        "scheduled": [
            "team_summary",
            "scoreticker",
            "clock"
        ],
        "intermission": [
            "scoreticker",
            "team_summary"
        ],
        "post_game": [
            "team_summary",
            "scoreticker",
            "clock"
        ]
    },
    "boards": {
        "scoreticker": {
            "preferred_teams_only": false,
            "rotation_rate": 5
        },
        "seriesticker": {
            "preferred_teams_only": true,
            "rotation_rate": 5
        },
        "standings": {
            "preferred_standings_only": true,
            "standing_type": "wild_card",
            "divisions": "north",
            "conference": "eastern"
        },
        "clock": {
            "duration": 15,
            "hide_indicator": false,
            "preferred_team_colors": true,
            "clock_rgb": "",
            "date_rgb": "",
            "flash_seconds": true
        },
        "weather": {
            "enabled": false,
            "view": "full",
            "units": "metric",
            "duration": 60,
            "data_feed": "EC",
            "owm_apikey": "",
            "update_freq": 5,
            "show_on_clock": true,
            "forecast_enabled": false,
            "forecast_days": 3,
            "forecast_update": 1
        },
        "wxalert": {
            "alert_feed": "EC",
            "update_freq": 5,
            "show_alerts": false,
            "nws_show_expire": false,
            "alert_title": true,
            "scroll_alert": true,
            "alert_duration": 5,
            "show_on_clock": true
        }
    },
    "sbio": {
        "screensaver": {
            "enabled": false,
            "animations": true,
            "start": "22:00",
            "stop": "08:00",
            "data_updates": false,
            "motionsensor": false,
            "pin": 7,
            "delay": 30
        },
        "dimmer": {
            "enabled": false,
            "source": "software",
            "daytime": "",
            "nighttime": "",
            "offset": 0,
            "frequency": 5,
            "light_level_lux": 400,
            "mode": "always",
            "sunset_brightness": 5,
            "sunrise_brightness": 40
        },
        "pushbutton": {
            "enabled": false,
            "bonnet": false,
            "pin": 25,
            "reboot_duration": 2,
            "reboot_override_process": "",
            "display_reboot": true,
            "poweroff_duration": 10,
            "poweroff_override_process": "",
            "display_halt": true,
            "state_triggered1": "weather",
            "state_triggered1_process": ""
        }
    }
```
**Hardware you are using**
- Raspberry pi 2b+
- Adafruit Bonnet 
- no Anti-flicker mod

**Software**
- main branch

falkyre, 2021-04-26:
How old is your SD card?  This looks like the library is corrupt on your SD card.

craigjmidwinter, 2021-04-26:
brand new SD but the imageio package must have been corrupt, uninstall and reinstall fixed it

closed 2021-04-26 by craigjmidwinter