riffnshred/nhl-led-scoreboard #274: Tried Updating to V1.6.3, but the world came crashing down

jwallner0710, 2021-05-02:
I have Raspberry Pi Zero W and I communicate with it using Putty and SSH. 

I had been using the scoreboard with no issues until I learned there was an issue with the API just a few days ago. I was getting the Loading Screen and that was it. I have always updated using commands through SSH (with the help from others). This past update I tried to basically start from scratch using the directions on the GitHub page. I downloaded the latest V1.6.3 zip to my windows computer then tried to use Etcher to flash the files, but Etcher kept failing... I had never done it this way before. I cleared my SD card and tried to flash the rpios-scoreboard zip file, but got a failed flash. 

Any insight on this? Needless to say, I need set the scoreboard up from scratch because I was unable to flash using Etcher.

pjockey, 2021-05-02:
If you are using SSH to get to the Pi, cd into the nhl-led-scoreboard folder then follow this:

```
git reset --hard
git checkout master
git pull
chmod +x scripts/install.sh
./scripts/install.sh
```

falkyre, 2021-05-02:
If etcher is failing, your card is bad

jwallner0710, 2021-05-02:
I just tried another sd card and I got the same error message. It does fine until the validation stage, then it fails.

pjockey, 2021-05-02:
Possibly something in the configuration file that has changed? Usually, from my experience is where the trouble is if the validation stage fails.

jwallner0710, 2021-05-02:
Correct me if I'm wrong. 

I should be able to delete everything off of the sd card and install the rpios-scoreboard file using Etcher? 

Is there anything I need to with the pi before this?

jwallner0710, 2021-05-02:
UPDATE:

It must have been the SD cards. I bought a new one today and it worked with no issues. I just fired up the pi....Let's see what I can do.

falkyre, 2021-05-02:
I just flashed the image twice, no issues.  When Etcher is validating the flash, it's comparing the SD card to the image you flashed.  If it fails, it's a bad SD card or a bad reader.

@pjockey The issue had nothing to do with the code or configuration, it wasn't even getting to the point of the image being flashed to the SD card.  So no Raspberry Pi OS to start and no scoreboard to run because there's no OS.

jwallner0710, 2021-05-03:
So the SD card is flashed and inserted into the pi. I am trying to interact with the raspberry via ssh, but I'm having no luck. It keeps saying "Connection Timed Out". I'm aware this means that I'm probably using the wrong IP address. 

I've tried using Fing to look for the raspberry pi on the network, I've tried to ping it using the command line on Windows, but all with no luck. 

What am I missing here?

jwallner0710, 2021-05-03:
UPDATE: 

I've remained persistent and have succeeded! It should be up and running in no time!! I learned a few new methods along the way. The directions on GitHub are almost perfect. It took me a little bit to get the hang of them, but they were super helpful. Thanks again all!

closed 2021-05-03 by jwallner0710