riffnshred/nhl-led-scoreboard #275: Flicker and Bad Colors

jwallner0710, 2021-05-03:
I downloaded the latest version on Pi Zero with the flicker mod on, but I'm having an issue with a bad flicker and dull colors on my board? 

Any ideas? Is it a bad solder connection between the bonnet and pi?

riffnshred, 2021-05-03:
can you copy and paste the command you are using to run the scoreboard?

jwallner0710, 2021-05-03:
[image]

I'm not quite sure if this is what you need.

jwallner0710, 2021-05-03:
It is very possible I'm not doing what I need to be....

riffnshred, 2021-05-04:
all good. when you run the sb-tools to change the setup, make sure your select the right options for your setup. specifically for the anti fliker mod

jwallner0710, 2021-05-04:
I tried that a few times, but didn't find any luck. When I do the /.nhl_setup command. I am also not seeing the goal and horn. Is this an issue?

falkyre, 2021-05-04:
The goal and horn not showing up has nothing to do with the actual scoreboard running.  It's only for a nice showing on the terminal when you run the nhl_setup application which is only used to create your initial config.json

Did you do the flicker mod on the RGB Bonnet/HAT?  It requires soldering.  Then you can select the anti-flicker mod in the sb-tools setup (assuming you are using the image).  Without that hardware mod, you will get flicker.

jwallner0710, 2021-05-05:
I checked my solder connections and it did fix the color issue. However, I still have the flicker. I have anti-flicker mod on and have selected that in the sb-tools setup. Could it be the RGB Bonnet/HAT? Do those "burn up"? Or does that it get affected by touching it too many times? FYI. I have a new bonnet and raspberry pi 3+ on the way in case I can't get this figured out. 

I'm going to try and redo the anti flicker soldering job on the bonnet and see if that is having a bad connection. 

It's just weird because I didn't have these issues before the latest update.

falkyre, 2021-05-07:
Sorry can't help with the soldering, I am horrible at it.  @riffnshred has a site he has given people the link that gives some pointers for soldering (it might be this:  https://learn.adafruit.com/adafruit-guide-excellent-soldering).  It could be your HAT/bonnet depending on how you soldered and if you messed up some of the traces.  That would more than likely manifest itself as bad colors like you were seeing or nothing at all.  The problem with using a Pi Zero, it only has a single CPU core so you'll get flickering on it no matter what you do.  You can minimize it with the anitflicker mod and possible some of the command line options @riffnshred has listed before (he'd have to pipe in and let you know which ones), but you will never get rid of it completely.  Moving to anything other than a Pi Zero will reduce the flickering by quite a bit.  I still get very minor flickering on my 3A+ and 4 raspberry pis but that happens on the 3A+ usually when there's scrolling of text and on the Pi4, it happens because I'm running the full version of the Raspberry Pi OS.

jwallner0710, 2021-05-08:
@falkyre Thank you for the response. Your information is truly appreciated. I definitely need to upgrade to the 3B+. Do you recommend the 3b+? 

So grateful for github and discord. HOWCHOO sucks and should be taken down. Lol. It's where I began this whole journey.

falkyre, 2021-05-08:
The 3b+ is fine. I go with the 3a+ just for how I do my builds, it's the perfect size and I don't need usb or ethernet. It's double the size of a pi zero but 4 cores and no ethernet and a single usb A port. The 3b+ is same as 3a+ but double the form factor because it has 4 usb ports and an ethernet port.

jwallner0710, 2021-05-08:
Is this the bonnet you buy? 
[image]

jwallner0710, 2021-05-08:
OR this one?
[image]

pjockey, 2021-05-08:
The second one is the one I have. 

I have that one mounted on a Pi Zero and I have Zero flicker. But perhaps I don't notice/see it because I have a the diffusing acrylic on it.

jwallner0710, 2021-05-08:
@pjockey What is this diffusing acrylic you speak of?

pjockey, 2021-05-08:
https://www.digikey.ca/en/products/detail/adafruit-industries-llc/4749/13426657

jwallner0710, 2021-05-08:
What do you do with these? 

On another note.....I'm getting the LOADING screen in a constant loop. I have reconfig and restarted. Any ideas?

pjockey, 2021-05-08:
I first ran into the acrylic in Adabox 016. It diffuses the light making it not as intense. Now it would depend on your distance from the screen. Mine is sitting about 8 to 10 feet from me.  I gave one to my son and sent the acrylic panel off of it and he likes it a bit brighter as it is probably 20 feet from his couch. He's never told me whether or not there is flicker. I'll have to ask. him.

jwallner0710, 2021-05-08:
I'm using 5mm and sitting about 8-10 feet away. Thanks for the info.

riffnshred, 2021-05-08:
> What do you do with these?
> 
> On another note.....I'm getting the LOADING screen in a constant loop. I have reconfig and restarted. Any ideas?

have you done the latest update? there was a change in the API side that cause the scoreboard to crash. it is fix on 1.6.3

jwallner0710, 2021-05-08:
I did do the latest update and it was working okay for a few days. I changed my password on the pi, I change config. file a few times adding different boards etc. 

When I got home today it was just constant loading screen loop.

jwallner0710, 2021-05-08:
Update. I have it back up and running. Not really sure what went wrong... I'll see how it runs throughout the day tomorrow.

falkyre, 2021-05-08:
If you are running the image, there's a bunch of tools that can help out why the board might be looping in start.  If you run sb-help it will give a list.  sb-stderr will show you what errors have been spit out. sb-stdout  shows the last 50kb oh the normal log. sb-livelog will show the live log file for the scoreboard.  If you see an issue, run sb-issues and copy the Pastebin link it gives you and post it here. That way we get you config.jsin and your logs and we can investigate.

closed 2021-05-14 by jwallner0710