riffnshred/nhl-led-scoreboard #276: Switch between NHL and MLB

FRiles427, 2021-05-04:
Not really an issue, but a question.  How would one go about switching from the NHL scoreboard to the MLB scoreboard?  By the time I get this thing done the Blackhawks season will be over and my attention will be focused on the Cubs.  Is there a quick way short of a second board setup or reprogramming every time I want to change?   Thanks!

pjockey, 2021-05-04:
Use Supervisor to perform the trick

https://github.com/riffnshred/nhl-led-scoreboard/wiki/Step-by-step-guide.#step-7---now-to-run-it

xjmsp, 2021-05-12:
What I did was set up the MLB / NHL both as services via https://github.com/MLB-LED-Scoreboard/mlb-led-scoreboard/wiki/Creating-a-systemd-startup-script
In the services you'll have to decide which board you want to start up when the Pi starts up.  
So one will need Restart=always & RestartSec=2 lines commented out with a # and that would be the non-primary board.  I comment them out so I can switch seasons.  During Primary NHL season I commend out the lines on MLB, and switch it around for when MLB is in primary season.
ex:
```
[Service]
Environment="SCOREBOARD_ARGS=--led-brightness=40 --led-slowdown-gpio=2 --led-gpio-mapping="adafruit-hat-pwm""
WorkingDirectory=/home/pi/mlb-led-scoreboard
ExecStart=/usr/bin/python main.py $SCOREBOARD_ARGS
#Restart=always
#RestartSec=2
```

Then with my iPhone setup a SSH Shortcut for each action.
Shut down NHL and start MLB:
```
sudo service nhl-led-scoreboard stop
sudo service mlb-led-scoreboard start
```

Shut down MLB and start NHL:
```
sudo service mlb-led-scoreboard stop
sudo service nhl-led-scoreboard start
```

xTrackstar04x, 2021-05-12:
If you use HomeBridge, I have a guide on the discord on how to make each board into a switch. Just let me know and I'll try to post it here.

jwallner0710, 2021-05-24:
Now that my Blues are out of the playoffs......didn't stand a chance against the Avs.... I'm looking to get the MLB repository up and running. I believe I have download the MLB from the github page and ran the necessary code. I'm trying to run through the directions the @xjmsp has outlined above. 

I have created the mlb-led-scoreboard.service file. I have placed the file on raspberry pi, but when I run the code to place it in the correct file, it says file not found. I have also download WinSCP place the file that way, but it saying I don't have permission to place the file in the right spot manually. 

Currently, my scoreboard is running nhl-led-scoreboard with supervisor enabled. Any help is appreciated. Thanks! 

This is where I'm at...
[image]

sflems, 2021-05-24:
@FRiles427 - I've just released v2 of the [`led-board-manager`](https://github.com/sflems/led-board-manager/) app. Its a web interface for managing boards (NHL/NFL/MLB and custom boards) and creating configurations. It's a definite helper over the standard supervisor features. Feel free to check it out!

> Not really an issue, but a question. How would one go about switching from the NHL scoreboard to the MLB scoreboard? By the time I get this thing done the Blackhawks season will be over and my [...]